projecthorus/radiosonde_auto_rx #545: Web interface with reverse proxy

tom2238, 2021-07-18:
Hi. 
Web options in station.cfg is now:
web_host = 0.0.0.0
web_port = 5000
etc..
So. Is possible to add some "web_path = /autorx" option. I would like to use radiosonde_auto_rx behind Nginx reverse proxy like http://192.168.10.251/autorx/. Current possible usage is only with http://192.168.10.251:5000/. Or exists some easy code modification (eg. in index.html) to add /autorx/ prefix into URLs?
Thanks.

darksidelemm, 2021-07-18:
Hmm, unsure about this one! There would also be issues with SocketIO's connections.
It's probably possible but I'd need someone with a bit more experience with flask and flask-socketio to figure it out..

tom2238, 2021-07-22:
Hi, reverse proxy with subdirectory (eg. http://192.168.10.251/autorx/) didn't work, but if I use subdomain (eg. http://autosonde.dt.local/) it works. However this option request static DNS record in PC or in router.

closed 2024-11-16 by darksidelemm