projecthorus/radiosonde_auto_rx #547: Support radiosondy.info

OK1SLM, 2021-07-20:
older versions of this program also stored data on the radiosondy.info server, but I found out that it no longer only saves them on the APRS map. Is that an intention?

darksidelemm, 2021-07-20:
No? 
The default APRS server in the example configuration file is radiosondy.info: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L185

If you enable APRS uploading (and fill out the appropriate settings) then uploading to radiosondy.info should work fine...

Example: https://radiosondy.info/sonde.php?sondenumber=S4610353
All those stations are running recent versions of auto_rx.

OK1SLM, 2021-07-20:
This is my setting
########################
# APRS UPLOAD SETTINGS #
########################
# Settings for uploading to APRS-IS
[aprs]
# Enable APRS upload (you will also need to change some options below!)
aprs_enabled = True

# APRS-IS Login Information
# The aprs_user field can have an SSID on the end if desired, i.e. N0CALL-4
# If you are a licensed amateur radio operator, you may want to change the aprs_port number below 
# to 14580, so that your uploaded telemetry makes its way out to the wider APRS network.
aprs_user = OK1SLM-15
# APRS-IS Passcode. You can generate one for your callsign here: https://apps.magicbug.co.uk/passcode/
aprs_pass = xxxxxx

# APRS Upload Rate - Upload a packet every X seconds.
# This has a lower limit of 30 seconds, to avoid flooding the APRS-IS network.
# Please be respectful of other uses of the APRS-IS network, and do not attempt
# to upload faster than this. 
upload_rate = 30

# APRS-IS server to upload to.
# Default to radiosondy.info for now, to allow stats to show up on http://radiosondy.info
# Packets are forwarded onto the rest of the APRS-IS network from radiosondy.info.
# If you wish to inject packets directly into the APRS-IS network, use rotate.aprs2.net
aprs_server = radiosondy.info

# APRS-IS Port Number to upload to.
# When using radiosondy.info:
#   Port 14590 - Packets stay within radiosondy.info. Non-licensed operators can use this.
#   Port 14580 - Packets are forwarded out to the wider APRS-IS network. Only licensed amateur radio operators should use this!!
# For all other APRS-IS servers (licensed amateur radio operators only!), use port 14580.
aprs_port = 14580

# Custom Object name / Callsign to be used when uploading to APRS-IS (Max 9 chars)
# WARNING: THIS SHOULD BE LEFT AT THE DEFAULT OF <id> UNLESS YOU HAVE A VERY GOOD REASON
# TO CHANGE IT!
#
# Using <id> means the uploads from multiple stations remain consistent, and we don't end up with
# lots of duplicate sondes on APRS-IS. If you enable the station location beaconing (below), maps
# like aprs.fi and radiosondy.info will show your station as the receiver.
#
# If used, this field should be either a callsign with a -11 or -12 suffix (i.e. N0CALL-12),
# or <id>, which will be replaced with the radiosondes serial number.
#
# WARNING - If running multiple RTLSDRs, setting this to a fixed callsign will result in odd behaviour on the online tracker.
# DO NOT SET THIS TO ANYTHING OTHER THAN <id> IF YOU ARE USING MORE THAN ONE SDR!
aprs_object_id = <id>

# Confirmation of the above setting.
# Set to True to confirm use of the custom object ID set above.
# Please read the notes above before setting this to True.
aprs_use_custom_object_id = False

# Select between uploading the radiosonde position as either an APRS Object, or as a position report.
# False = Upload position as an Object, sourced from <aprs_user>
# True = Position Report (with emulated iGating via <aprs_user>)
# Note that if using Position Reports with aprs_object_id set to <id>, then non-AX.25 compatible callsigns will be generated.
# If in doubt, leave this at False.
aprs_position_report = True

# The APRS-IS beacon comment. The following fields can be included:
# <freq> - Sonde Frequency, e.g. 401.520 MHz
# <type> - Sonde Type (e.g. RS41-SG, DFM09, etc....)
# <id> - Sonde Serial Number (i.e. M1234567)
# <vel_v> - Sonde Vertical Velocity (i.e. -5.1m/s)
# <temp> - Sonde reported temperature. If no temp data available, this will report -273 degC. 
# <humidity> - Sonde reported humidity.
# <pressure> - Sonde reported pressure
# <batt> - Battery Voltage (e.g. 3.1V)
# The default comment (below) is compatible with radiosondy.info's parsers, and should only be modified
# if absolutely necessary.
aprs_custom_comment = Clb=<vel_v> t=<temp> h=<humidity> p=<pressure> <freq> Type=<type> Radiosonde

# APRS Station Location Beaconing
# If enabled, you will show up on APRS using the aprs_user callsign set above.
# This also requires your station location to be set in the above [location] section.
station_beacon_enabled = True

# Station beaconing rate, in minutes.
station_beacon_rate = 30

# Station beacon comment.
# <version> will be replaced with the current auto_rx version number
station_beacon_comment = radiosonde_auto_rx SondeGate v<version>

# Station Beacon Icon
# The APRS icon to use, as defined in http://www.aprs.org/symbols.html
# Note that the two characters that define the icon need to be concatenated. Examples:
# Antenna Tower = /r
# House with Yagi = /y
# Satellite Dish = /`    (This is required if you want to show up on radiosondy.info's station list.)
station_beacon_icon = /`

OK1SLM, 2021-07-20:
Any idea what could be wrong?

darksidelemm, 2021-07-20:
Why did you set aprs_position_report to True? The default for this is False.
All you should need to modify is your aprs_user and aprs_pass, and set station_beacon_enabled to True - the rest of the fields should be left at their defaults.

OK1SLM, 2021-07-20:
OK thank you.

darksidelemm, 2021-07-20:
Check that change works on the next launch then close the issue if all OK.

closed 2021-08-15 by darksidelemm