projecthorus/radiosonde_auto_rx #549: DMF decoding failed on 1.5.5-beta2

fkemps, 2021-07-21:
Decoding of DMF failed while running just fine with 1.5.4 GA.

This might be due to back-end movement of APRS code.
Switching back to 1.5.4 GA code immediately result in recognition of the sonde.

I enabled raw output option, but no raw output appeared in ./log/ directory. 
```
Jul 21 10:14:26 spy auto_rx[767459]: 2021-07-21 10:14:26,357 INFO:Task Manager - Detected new DFM sonde on 402.310 MHz!
Jul 21 10:14:26 spy auto_rx[767459]: 2021-07-21 10:14:26,359 INFO:Task Manager - SDR #14 has been allocated to Decoder (DFM, 402.310 MHz).
Jul 21 10:14:27 spy auto_rx[767459]: 2021-07-21 10:14:27,745 INFO:Sondehub Uploader - Uploaded 15 telemetry packets to Sondehub in 0.2 seconds.
Jul 21 10:14:29 spy auto_rx[767459]: 2021-07-21 10:14:29,395 INFO:Decoder #14 DFM 402.310 - Using fsk_demod decoder chain.
Jul 21 10:14:29 spy auto_rx[767459]: 2021-07-21 10:14:29,416 INFO:Decoder #14 DFM 402.310 - Starting decoder subprocess.
Jul 21 10:14:33 spy auto_rx[767459]: Exception in thread Thread-3145:
Jul 21 10:14:33 spy auto_rx[767459]: Traceback (most recent call last):
Jul 21 10:14:33 spy auto_rx[767459]:   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
Jul 21 10:14:33 spy auto_rx[767459]:     self.run()
Jul 21 10:14:33 spy auto_rx[767459]:   File "/usr/lib/python3.8/threading.py", line 870, in run
Jul 21 10:14:33 spy auto_rx[767459]:     self._target(*self._args, **self._kwargs)
Jul 21 10:14:33 spy auto_rx[767459]:   File "/home/fkemps/radiosonde_auto_rx/auto_rx/autorx/decode.py", line 1082, in decoder_thread
Jul 21 10:14:33 spy auto_rx[767459]:     _ok = self.handle_decoder_line(_line)
Jul 21 10:14:33 spy auto_rx[767459]:   File "/home/fkemps/radiosonde_auto_rx/auto_rx/autorx/decode.py", line 1329, in handle_decoder_line
Jul 21 10:14:33 spy auto_rx[767459]:     _telemetry["aprsid"] = generate_aprs_id(_telemetry)
Jul 21 10:14:33 spy auto_rx[767459]:   File "/home/fkemps/radiosonde_auto_rx/auto_rx/autorx/utils.py", line 246, in generate_aprs_id
Jul 21 10:14:33 spy auto_rx[767459]:     _dfm_id = int(sonde_data["id"].split("-")[-1])
Jul 21 10:14:33 spy auto_rx[767459]: ValueError: invalid literal for int() with base 10: 'xxxxxxxx'
```

After switching the regular decoding process works as expected

```
Jul 21 12:30:12 spy auto_rx[793128]: 2021-07-21 12:30:12,207 INFO:Scanner #11 - Detected peaks on 7 frequencies (MHz): [404.31 402.81 402.01 404.5  401.36 405.08 405.4 ]
Jul 21 12:30:15 spy auto_rx[793128]: 2021-07-21 12:30:15,625 INFO:Task Manager - Detected new DFM sonde on 404.309 MHz!
Jul 21 12:30:15 spy auto_rx[793128]: 2021-07-21 12:30:15,626 INFO:Task Manager - SDR #12 has been allocated to Decoder (DFM, 404.309 MHz).
Jul 21 12:30:18 spy auto_rx[793128]: 2021-07-21 12:30:18,587 INFO:Decoder #12 DFM 404.309 - Using fsk_demod decoder chain.
Jul 21 12:30:18 spy auto_rx[793128]: 2021-07-21 12:30:18,597 INFO:Decoder #12 DFM 404.309 - Starting decoder subprocess.
Jul 21 12:30:20 spy auto_rx[793128]: 2021-07-21 12:30:20,593 INFO:Task Manager - Detected new DFM sonde on 402.809 MHz!
Jul 21 12:30:20 spy auto_rx[793128]: 2021-07-21 12:30:20,594 INFO:Task Manager - SDR #13 has been allocated to Decoder (DFM, 402.809 MHz).
Jul 21 12:30:22 spy auto_rx[793128]: 2021-07-21 12:30:22,419 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
Jul 21 12:30:23 spy auto_rx[793128]: 2021-07-21 12:30:23,607 INFO:Decoder #13 DFM 402.809 - Using fsk_demod decoder chain.
Jul 21 12:30:23 spy auto_rx[793128]: 2021-07-21 12:30:23,626 INFO:Decoder #13 DFM 402.809 - Starting decoder subprocess.
Jul 21 12:30:24 spy auto_rx[793128]: 2021-07-21 12:30:24,526 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
Jul 21 12:30:25 spy auto_rx[793128]: 2021-07-21 12:30:25,611 INFO:Task Manager - Detected new DFM sonde on 402.011 MHz!
Jul 21 12:30:25 spy auto_rx[793128]: 2021-07-21 12:30:25,611 INFO:Task Manager - SDR #14 has been allocated to Decoder (DFM, 402.011 MHz).
Jul 21 12:30:26 spy auto_rx[793128]: 2021-07-21 12:30:26,531 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
Jul 21 12:30:26 spy auto_rx[793128]: 2021-07-21 12:30:26,532 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
Jul 21 12:30:27 spy auto_rx[793128]: 2021-07-21 12:30:27,334 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
Jul 21 12:30:27 spy auto_rx[793128]: 2021-07-21 12:30:27,435 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
Jul 21 12:30:28 spy auto_rx[793128]: 2021-07-21 12:30:28,538 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
Jul 21 12:30:28 spy auto_rx[793128]: 2021-07-21 12:30:28,541 INFO:Telemetry Logger - Using existing log file: /home/fkemps/radiosonde_auto_rx/auto_rx/log/20210721-102806_DFM-20067988_DFM09_404309_sonde.log
```
Running on Ubuntu with git clone testing branch.

darksidelemm, 2021-07-21:
Yep, that's a bug... Just pushed a basic fix (wrap the aprs id calculation in try/except) to the testing branch.

fkemps, 2021-07-21:
I can confirm 1.5.5-beata3 is working again.

closed 2021-07-21 by darksidelemm