projecthorus/radiosonde_auto_rx #559: Disable Sondehub uploading by default

MikeTango, 2021-08-15:
Hi,

Public data upload should be disabled unless the user explicitly consents to their data being used for any non-local purpose.

Moll1989, 2021-08-15:
@MikeTango just my 20cents as another user of the project:

Firstly, it's not exactly data that could present a risk to someone's privacy.

Secondly, the installation instructions ask the user to configure the callsign and location details for upload into sondehub which makes it quite clear that data will be uploaded by default.

Finally, I suspect the overwhelming majority of users would want their data uploaded to sondehub, so why make them configure an additional line of the config file? It would make more sense for the few that don't want it uploaded to disable it.

xssfox, 2021-08-15:
Uploading data to community backends is kind of the point of radiosonde_auto_rx. It's within the first three sentences of the README.md. We don't try to encourage competitive behaviour in sonde chasing. We encourage the data to be shared for research.

closed 2021-09-18 by darksidelemm