projecthorus/radiosonde_auto_rx #571: SDR dongle not working?

Prutzz, 2021-09-06:
Hi there!

Got my self a fresh new Rasp. Pi 3+ and SDR RTL V3 dongle. 

I've followed the whole tutorial and trying to get the software working. But it seems something is wrong. I did rtl_test and there is no problem. But when I want to try to use the software I'm getting a fault. 

`Using device 0: Generic RTL2832U OEM
Number of frequency hops: 3
Dongle bandwidth: 2479166Hz
Downsampling by: 1x
Cropping by: 20.00%
Total FFT bins: 12288
Logged FFT bins: 9830
FFT bin size: 605.27Hz
Buffer size: 16384 bytes (3.30ms)
Reporting every 20 seconds
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Failed to open log_power_0.csv

2021-09-06 15:08:47,883 WARNING:Scanner #0 - RTLSDR produced no output... resetting and retrying.
2021-09-06 15:08:48,078 INFO:RTLSDR - Attempting to reset: Bus: 1  Device: 4
2021-09-06 15:08:58,335 INFO:Scanner #0 - Running frequency scan.
2021-09-06 15:08:58,923 CRITICAL:Scanner #0 - rtl_power call failed with return code 1.
2021-09-06 15:08:58,925 CRITICAL:Scanner #0 - rtl_power reported error: Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
`

anoyone got a clue what is wrong?

darksidelemm, 2021-09-06:
The error `Failed to open log_power_0.csv` suggests that perhaps there is a `log_power_0.csv` file present, created by a different user? Did you perhaps try and run up auto_rx as root at one point?
If this file exists, try removing it.

Prutzz, 2021-09-15:
Problem was I think I did all the actions the Wiki said about the SDR dongle.... I flashed a new image for the Rasp and only used the "sudo apt-get install rtl-sdr" command. And now it works like a charm. :)

closed 2021-09-15 by Prutzz