projecthorus/radiosonde_auto_rx #589: autorx does not decrypt.  10 minutes pass, he begins to decipher.

GUGA-TV, 2021-11-29:
Hello.  problem: I see a probe on the Spectrum, I hear it on the radio.  autorx does not decrypt.  10 minutes pass, he begins to decipher.  the signal power is high, the radio can be heard with overload, a very powerful signal is also visible on the autorx spectrum.  what is the problem?  maybe you need to put a checkbox where the detection peak is, you need to put a scan detection?  Tell me please.  you can write to me in telegram, my ID: @idiots_and_robots.  I am Russian.[image]

darksidelemm, 2021-11-29:
That large 'signal' (if it is even a signal and not some other kind of noise), is probably not a radiosonde, but something else. At 1 MHz wide i'd be extremely surprised if that was a radiosonde.

GUGA-TV, 2021-12-03:
It is radiosonde MRZ-N1. It is very wide

darksidelemm, 2021-12-03:
In that case, I suspect support will not be added anytime in the near future - for now we only support the narrowband MRZ-H1 sondes. That is an *extremely* wideband signal for a radiosonde in the 400 MHz band.

darksidelemm, 2021-12-03:
Going to continue to track this here: https://github.com/projecthorus/radiosonde_auto_rx/issues/592#issuecomment-985369181

closed 2021-12-03 by darksidelemm