projecthorus/radiosonde_auto_rx #592: Reception of 1680 MHz plain sondes and new things from Russia

FellowWroclaw, 2021-12-03:
Hi everyone here! In Eastern Europe and parts of Central Asia  commonly used Russian-made radiosondes other than MRZ-N1 400 MHz . They are tracked with radar and computer with SDR radio receives data from on IF output of radar receiver. Meteo-Radiy MRZ-3MK , an Aeropribor AK2-02м series .  New radars are installed because it’s cheaper to operate with cost of 1 plain microwave sonde about $20 and cost of GPS one is about $45. 

Also Aeropribor features  АК2-02мФ АК2-02мН sondes for 403 MHz band using not only plain AFSK or GFSK but LoRa modulation  

As for reception of 1680ish sondes it’s not possible directily with SDR , firstly you need to get IF from eg. satellite tv tuner and then feed it for further AM 800 kHz demodulation

darksidelemm, 2021-12-03:
It's worth noting that many stations in the US are receiving 1680 MHz LMS6 sondes, though these are about 300 kHz in bandwidth. Most stations there are using either dish, or helix antennas with preamplifier to receive.

As for the LoRa sondes, as far as I'm aware there isn't an open-source LoRa demodulator that's actually functional, and it's also not really something I'd want to spend effort on. If someone wants to make a demodulator for these using a LoRa chip then they can have at it, and upload the results to Sondehub.

Either way, without a large amount of IQ samples and some 'truth' data for a few launches, it's not really possible to add any support.

darksidelemm, 2021-12-03:
So, it seems the sondes that may need investigation are:
* Meteo-Radiy MRZ-3M - 1680 MHz, ~800 kHz BW, unknown modulation (AM demod?!)
* Aeropribor AK2-02м - 403 MHz - LoRa (what settings?)
* Meteo-Radiy-MRZ-N1 - 403 MHz - Wide bandwidth? (Refer screenshot here: https://github.com/projecthorus/radiosonde_auto_rx/issues/589#issue-1066045826)

Nothing can proceed on any analysis of these without large amounts of IQ samples.

UE9AAG, 2021-12-05:
It makes no sense to decode the MRZ-3M telemetry, there is no position information in it. The location is calculated by the active radar at the base station.

I have a weather balloon ak2-02m 403 MHz. It uses a non-standard LORA.
I researched it and came to the conclusion that it needs a hardware decoder on the sx1278 chip to decode. You can use a standard module built on such a microcircuit, but it requires improvement.

BatiaMahno, 2021-12-18:
Есть метеозонд Метео-Радий МРЗ-3М - 1680 МГц  двойная модуляция,декодируется только телеметрия,координат нет.
Есть метеозонд  Аэроприбор АК2-02м - 403 МГц - LoRa могу предоставить замеры

BatiaMahno, 2021-12-27:
Hello. You can add a decoder of weather balloons AK-02m with LORA modulation, operating frequency 403 MHz. Here's a friend who wrote a sketch. Also read the firmware from the weather balloon
[AK2-02_Mn.zip](https://github.com/projecthorus/radiosonde_auto_rx/files/7781536/AK2-02_Mn.zip)

darksidelemm, 2021-12-27:
No, I can't add this to auto_rx. What you have supplied is code for an Arduino board.

What I need is software which can take quadrature samples from a software-defined radio, and demodulate it to produce the data packets.

BatiaMahno, 2021-12-28:
SX1278 + PC   ?

darksidelemm, 2021-12-28:
Not something that will be done within the auto_rx project. You'd have better luck asking the rdz_ttgo_sonde project about that, since they are already using a LoRa chipset on the TTGO.

closed 2022-12-22 by darksidelemm