projecthorus/radiosonde_auto_rx #594: Multiple Signals?

K4KDR, 2021-12-05:
Searching the 'Issues', I see that this has been brought up in past years.

I often have more than one radiosonde in-range at the same time.  Tonight I realized that scanning stops when valid decodes are received and does not go back to scanning for signals until/unless the source of the current decodes disappears.

I tried bouncing the service with 

`sudo systemctl restart auto_rx.service`

... but never could 'trick' the code into scanning for the "other" signal.  

Is there a recommended 'best' way to receive more than one radiosonde at the same time?  Or any work-around in the code that some of us could implement to force a re-scan once per minute?  Along that line, is there a way to **reverse** the direction of the frequency scan so that the lowest numerical frequency is not always found first? (I'm just blindly assuming that the scans run from low-to-high frequency)

Or, should restarting the service repeatedly accomplish the same thing and I just didn't luck into catching a signal from the other radiosonde?

Many thanks!

-Scott,  K4KDR

darksidelemm, 2021-12-05:
Currently the only way to decode multiple radiosonde is to... add more RTLSDRs (with a suitable RF splitter). Yes, this isn't particularly efficient, but it's all I've got right now. 
There is some work in progress by KA9Q to develop a multi-channel server system which will allow decoding of many sondes simultaneously (CPU limited), but this is months away, and will have higher CPU requirements than we have currently (Pi 4 or better).

You can 'prod' auto_rx to go back into scan mode using the web interface, though you need web_control enabled. You can halt a currently running decoder, which will cause auto_rx to re-scan: https://github.com/projecthorus/radiosonde_auto_rx/wiki/Web-Interface-Guide#advanced-controls

As for ordering of scanned signals, currently it orders the detected peaks by SNR, and then works through these in the detection stage, starting a decoder on the first signal it finds. If you add entries to the always_scan list, these will get processed first in the detection stage.

K4KDR, 2021-12-05:
All those details are extremely helpful & much appreciated!

K4KDR, 2021-12-06:
Using the web controls worked!  Well, of course we can't track multiple radiosondes at the same time but by **stopping** the decoding of the current device & requesting a decode to **START** on the specific frequency of the 'other' unit, AutoRX switched over to tracking the object that had been orphaned.   Good to know how to do that - thanks!

[image]

[image]

closed 2022-01-30 by darksidelemm