projecthorus/radiosonde_auto_rx #600: Not  SNR

BatiaMahno, 2022-01-05:
Hello. SNR (dB) is not displayed, what's the problem? This is not only me.

darksidelemm, 2022-01-05:
SNR will not be displayed for iMet, LMS6-1680, IMS-100 and MRZ sondes, due to the signal processing chain in use for those sondes, which does not perform signal Eb/N0 estimation.

BatiaMahno, 2022-01-06:
Sudak city sonde station MRZ 102180-124118    (7DB,403 MHZ)     we have the same equipment

darksidelemm, 2022-01-06:
OK, so by default the MRZ is decoded using a 'safe' decode chain which doesn't support SNR reporting.
There is an 'experimental' decode chain which can be enabled here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L481

This would report SNR.

Perhaps try changing that setting to True and see how it works for you. If it decodes OK, please let me know, and I'll make that the default in future updates.

BatiaMahno, 2022-01-06:
Why does a friend work? The software versions and hardware are the same.

darksidelemm, 2022-01-06:
They have probably set the setting I linked above to True.

GUGA-TV, 2022-01-06:
yes, I am the INR_locator station near Moscow.  I have false, but sometimes it shows SNR, and sometimes it doesn't.

darksidelemm, 2022-01-06:
For MRZ sondes, with that setting set to False, there is no SNR information passed into auto_rx from the decoder, so I don't know how you would be getting any SNR information when decoding MRZ sondes. 

With the setting set to True, it uses a different decode chain that does produce SNR estimates. Try that decode chain and see if it performs OK. I have not been able to test it due to insufficient high-SNR recordings (e.g. from a captured sonde).

radiostalker, 2022-01-06:
Ok, I will try. If it will work, can you turn it on for default PLZ?

BatiaMahno, 2022-01-06:
MRZ 101473-125546 Oneline

BatiaMahno, 2022-01-06:
481 On.  No DB.

radiostalker, 2022-01-06:
Same situation as a mahno. My station INR_locator with true mode tracking sonde, but no SNR

darksidelemm, 2022-01-06:
Can you paste the relevant section of your station.cfg file please. Make sure you have restarted auto_rx after changing the configuration.

radiostalker, 2022-01-06:
DEMODULATOR / DECODER TWEAKS #
################################

#
# Alternate Decode Chains
#
# NOTE: This section will be deprecated soon, with the fsk_demod chains used permanently.
#
# These (not so) experimental demod chains use David Rowe's fsk_demod modem.
# They have much better drift handling performance, and so may be better suited for
# 1680 MHz sondes, but use considerably more CPU. Recommend use of a Pi 3 or better.
# The experimental decoders can be enabled/disabled independently for each radiosonde type.
# It is recommended that these are always left enabled. 
rs41_experimental = True
rs92_experimental = True
dfm_experimental = True
m10_experimental = True
lms6-400_experimental = True
imet54_experimental = True
# LMS6-1680 Decoder options:
# False = Less CPU, possibly worse weak-signal performance.
# True = Higher CPU (Pushing a RPi 3 to its limits), possibly better weak-signal performance.
lms6-1680_experimental = False

# MRZ sondes have not yet been tested with the fsk_demod flowgraph in the wild.
# If someone can confirm that this works, I'll set it to True by default!
mrz_experimental = True

# Note: As iMet-4 sondes use AFSK, using fsk_demod does not give any advantage, so there is no experimental decoder for them.

# Optimize 1680 MHz Scanning for RS92-NGP Sondes
# This flag sets the use of IQ detection when scanning for 1680 MHz sondes, which allows RS92-NGP sondes
# to be detected reliably.
# Set this to True if you are sure that only RS92-NGPs are flying in your area.
ngp_tweak = False

Yes, I restarted it.

darksidelemm, 2022-01-06:
Ok... the other way to confirm this particular demod chain is running is to look for a log message "Using fsk_demod decoder chain." when the decoder starts up. You can either wait for a sonde, or if you are running auto_rx 'natively' you can shutdown auto_rx then start it up from the terminal using: 
`python auto_rx.py -f 403.0 -m MRZ -v`

If this *doesn't* appear, then it means it's using the non-experimental chain which won't show SNR.

darksidelemm, 2022-01-06:
So I just ran some checks locally, and yes, with `mrz_experimental = True`, the correct decoder chain is used, which should result in SNR output. Unfortunately I don't have a MRZ sonde to test with, so all I could really check was that the SNR statistics code was operating (which it was).

```
 % python auto_rx.py -f 403.0 -m MRZ -v
2022-01-07 09:03:24,246 INFO:Reading configuration file...
2022-01-07 09:03:24,247 WARNING:Config - Web Password not set, disabling web control
2022-01-07 09:03:24,247 DEBUG:RTLSDR - Not a native Linux system, skipping RTLSDR search.
2022-01-07 09:03:26,047 INFO:Config - Tested SDR #0 OK
2022-01-07 09:03:26,049 WARNING:Overriding RX timeout for manually specified radiosonde type. Decoders will not automatically stop!
2022-01-07 09:03:26,050 INFO:Started Flask server on http://0.0.0.0:5001
2022-01-07 09:03:26,055 INFO:Telemetry Logger - Started Telemetry Logger Thread.
2022-01-07 09:03:26,056 INFO:OziMux - Started OziMux / Payload Summary Exporter
2022-01-07 09:03:26,056 INFO:Sondehub Uploader - Started Sondehub Uploader Thread.
2022-01-07 09:03:26,348 INFO:Version - Local Version: 1.5.8 - Up to date!
2022-01-07 09:03:26,348 INFO:Task Manager - Detected new MRZ sonde on 403.000 MHz!
2022-01-07 09:03:26,348 INFO:Task Manager - SDR #0 has been allocated to Decoder (MRZ, 403.000 MHz).
2022-01-07 09:03:26,349 DEBUG:RTLSDR - Not a native Linux system, skipping RTLSDR search.
2022-01-07 09:03:27,100 INFO:Sondehub Uploader - Uploaded station information to Sondehub.
2022-01-07 09:03:28,157 INFO:Decoder #0 MRZ 403.000 - Using fsk_demod decoder chain.
2022-01-07 09:03:28,158 DEBUG:Decoder #0 MRZ 403.000 - Demodulator Command: rtl_fm -p 0 -d 0 -M raw -F9 -s 48000 -f 402988000 2>/dev/null |./fsk_demod --cs16 -s -b 1200 -u 22800 --stats=5 2 48000 2400 - -
2022-01-07 09:03:28,158 DEBUG:Decoder #0 MRZ 403.000 - Decoder Command: ./mp3h1mod --auto --json --softin --ptu 2>/dev/null
2022-01-07 09:03:28,170 INFO:Decoder #0 MRZ 403.000 - Starting decoder subprocess.
```

The important lines here being `Using fsk_demod decoder chain.`, and also the Demodulator command information, which includes `fsk_demod` with the `--stats=5` argument, showing that demodulator statistics should be being generated.

BatiaMahno, 2022-01-07:
481 > On > reset> 40 db  = Ok

closed 2022-01-07 by BatiaMahno

BatiaMahno, 2022-01-07:
Зонд MRZ-1369-23206 может видеть только 0 SV - положение отбрасывается как плохое.
0 SV what does it mean?

darksidelemm, 2022-01-07:
I can't understand most of that... but the '0 SV' part probably means the sonde in question didn't have GPS lock (no 'space vehicles' tracked).

BatiaMahno, 2022-01-07:
Ок