projecthorus/radiosonde_auto_rx #606: Location Confusion - 2 sondes on same freq

K4KDR, 2022-02-03:
More of an FYI than a request or issue...

tonight I am in-range of two radiosondes transmitting on the same frequency.  As you can see in the images below, either autoRX on my local device, or the sondehub.org website (don't know which) seems to have difficulty with the position plots.    Thanks!

[image]

[image]

darksidelemm, 2022-02-03:
Welcome to the wonders of DFM telemetry... Will have to provide a longer
answer when I'm off my phone, but there is very little we can do here...

TheSkorm, 2022-02-03:
FYI we'll be implementing some more intensive backend data checking specifically for DFMs in the future to hopefully reduce this occurring. 

DFMs send their serial in one frame, and their telemetry in several other frames. So the serial will decode in one frame but then you rx telemetry from a completely different radiosonde in the the proceeding frames.

K4KDR, 2022-02-03:
VERY interesting - thank you for the info!!

TheSkorm, 2022-02-03:
Added to the backend : https://github.com/projecthorus/sondehub-infra/commit/001fc6c5ca6028cafa287a92cd890b39fe6a4a5a

Keep an eye out for this type of thing occurring again. Hopefully its less likely now

closed 2022-12-22 by darksidelemm