projecthorus/radiosonde_auto_rx #607: Computer Sound Card Instead of SDR

Frojoe6969, 2022-02-03:
Hello,

I started doing all the research and seems I have everything set up properly in terms of radio, antenna, hearing the packets from the weather balloon, Auto-RX set up on a virtual Linux machine… The one issue I am having is getting the sound to go to auto RX from the microphone jack instead of using an SDR. 

What am I missing?  When I try to run $ Python3 auto_rx.py I get an error message “No Working SDRs! Cannot Run”
 
I would assume you could connect the output a UHF radio connected to my computer microphone jack and auto_rx would be able to see the audio to decode. Is that not the case? Am I restricted to using and SDR?

darksidelemm, 2022-02-03:
Yes, this software is SDR only. It is intended to be fully automated.

TheSkorm, 2022-02-03:
You might want to consider using this sdr++ plugin https://github.com/dbdexter-dev/sdrpp_radiosonde or sondemonitor

closed 2022-02-03 by TheSkorm