projecthorus/radiosonde_auto_rx #608: Scan plot height in web interface

pavolgaj, 2022-02-03:
How to change the height of the scan plot (make it smaller) in the web interface? If I have opened scan plot, the map is very small - I couldn't see anything on it.
I tried to look in CSS files but I didn't find anything...

TheSkorm, 2022-02-03:
[image]
[image]
[image]

You can use the settings to hide the scan plot entirely and make the map full screen

pavolgaj, 2022-02-04:
Yes, I know about this option. I want to show scan plot but smaller.

closed 2023-03-10 by darksidelemm