projecthorus/radiosonde_auto_rx #616: IMET Decoder doesn't open

HoshenKadosh, 2022-02-21:
Hi, on my RPI when running auto_rx, it detects the sonde (IMET), starts a decoder, and then closes it immediately, without any errors, after that it goes back to scanning and so on..
Tried opening the decode manually from the web UI, and rebuilding the decoders, didn't seem to help.

It is worth mentioning that other sondes, like RS41 - work, and it decodes them successfully.
Any idea what this problem can be related to? Thanks.

Log output:
```
`Feb 21 02:42:20 raspberrypi auto_rx[1968]: 2022-02-21 02:42:20,777 INFO:Task Manager - SDR #00000004 has been allocated to Scanner.
Feb 21 02:42:24 raspberrypi auto_rx[1968]: 2022-02-21 02:42:24,056 INFO:Scanner #00000004 - Starting Scanner Thread
Feb 21 02:42:24 raspberrypi auto_rx[1968]: 2022-02-21 02:42:24,086 INFO:Scanner #00000004 - Running frequency scan.
Feb 21 02:42:39 raspberrypi auto_rx[1968]: 2022-02-21 02:42:39,825 INFO:Scanner #00000004 - Detected peaks on 4 frequencies (MHz): [401.99 402.   400.33 403.66]
Feb 21 02:42:48 raspberrypi auto_rx[1968]: 2022-02-21 02:42:48,118 INFO:Task Manager - Detected new IMET sonde on 401.996 MHz!
Feb 21 02:42:48 raspberrypi auto_rx[1968]: 2022-02-21 02:42:48,134 INFO:Halting Scanner to decode detected radiosonde.
Feb 21 02:42:48 raspberrypi auto_rx[1968]: 2022-02-21 02:42:48,153 INFO:Scanner #00000004 - Waiting for current scan to finish...
Feb 21 02:43:00 raspberrypi auto_rx[1968]: 2022-02-21 02:43:00,990 INFO:Scanner #00000004 - Scanner Thread Closed.
Feb 21 02:43:01 raspberrypi auto_rx[1968]: 2022-02-21 02:43:01,017 INFO:Task Manager - SDR #00000004 has been allocated to Decoder (IMET, 401.996 MHz).
Feb 21 02:43:04 raspberrypi auto_rx[1968]: 2022-02-21 02:43:04,512 INFO:Decoder #00000004 IMET 401.996 - Starting decoder subprocess.
Feb 21 02:43:07 raspberrypi auto_rx[1968]: 2022-02-21 02:43:07,651 INFO:Decoder #00000004 IMET 401.996 - Closed decoder subprocess.
Feb 21 02:43:08 raspberrypi auto_rx[1968]: 2022-02-21 02:43:08,413 INFO:Task Manager - SDR #00000004 has been allocated to Scanner.
Feb 21 02:43:11 raspberrypi auto_rx[1968]: 2022-02-21 02:43:11,731 INFO:Scanner #00000004 - Starting Scanner Thread
Feb 21 02:43:11 raspberrypi auto_rx[1968]: 2022-02-21 02:43:11,784 INFO:Scanner #00000004 - Running frequency scan.
Feb 21 02:43:28 raspberrypi auto_rx[1968]: 2022-02-21 02:43:28,079 INFO:Scanner #00000004 - Detected peaks on 2 frequencies (MHz): [401.99 402.  ]
Feb 21 02:43:33 raspberrypi auto_rx[1968]: 2022-02-21 02:43:33,787 INFO:Task Manager - Detected new IMET sonde on 401.996 MHz!
Feb 21 02:43:33 raspberrypi auto_rx[1968]: 2022-02-21 02:43:33,804 INFO:Halting Scanner to decode detected radiosonde.
Feb 21 02:43:33 raspberrypi auto_rx[1968]: 2022-02-21 02:43:33,822 INFO:Scanner #00000004 - Waiting for current scan to finish...
Feb 21 02:43:46 raspberrypi auto_rx[1968]: 2022-02-21 02:43:46,441 INFO:Scanner #00000004 - Scanner Thread Closed.
Feb 21 02:43:46 raspberrypi auto_rx[1968]: 2022-02-21 02:43:46,466 INFO:Task Manager - SDR #00000004 has been allocated to Decoder (IMET, 401.996 MHz).
Feb 21 02:43:49 raspberrypi auto_rx[1968]: 2022-02-21 02:43:49,931 INFO:Decoder #00000004 IMET 401.996 - Starting decoder subprocess.
Feb 21 02:43:53 raspberrypi auto_rx[1968]: 2022-02-21 02:43:53,183 INFO:Decoder #00000004 IMET 401.996 - Closed decoder subprocess.
Feb 21 02:43:53 raspberrypi auto_rx[1968]: 2022-02-21 02:43:53,803 INFO:Task Manager - SDR #00000004 has been allocated to Scanner.
Feb 21 02:43:57 raspberrypi auto_rx[1968]: 2022-02-21 02:43:57,111 INFO:Scanner #00000004 - Starting Scanner Thread
Feb 21 02:43:57 raspberrypi auto_rx[1968]: 2022-02-21 02:43:57,131 INFO:Scanner #00000004 - Running frequency scan.
`

darksidelemm, 2022-02-21:
Can you run autorx with the -v option, perhaps manually (python3 auto_rx.py
-v -f 402.000 -m IMET ).
This will print out the decode command, which you can then try to run
outside of autorx, which will hopefully expose any errors.

darksidelemm, 2022-02-21:
Some example relevant lines from my terminal:

```
$ python3 auto_rx.py -v -f 402.0 -m IMET
...
2022-02-21 06:52:12,468 INFO:Task Manager - SDR #00000002 has been allocated to Decoder (IMET, 402.000 MHz).
2022-02-21 06:52:13,153 INFO:Sondehub Uploader - Uploaded station information to Sondehub.
2022-02-21 06:52:15,631 DEBUG:Decoder #00000002 IMET 402.000 - Opening /home/pi/radiosonde_auto_rx/auto_rx/log/20220221-065212_IMET_402000000.raw to save decoder raw data.
2022-02-21 06:52:15,633 DEBUG:Decoder #00000002 IMET 402.000 - Decoder Command: /home/pi/rtl-sdr/build/src/rtl_fm -p 0 -d 00000002 -g 32.8 -M fm -F9 -s 15k -f 402000000 2>/dev/null |sox -t raw -r 15k -e s -b 16 -c 1 - -r 48000 -b 8 -t wav - highpass 20 2>/dev/null |./imet1rs_dft --json -r 2>/dev/null
2022-02-21 06:52:15,634 INFO:Task Manager - SDR #00000003 has been allocated to Scanner.
```
The above shows the command in use. I can then run this, removing some of the 2>/dev/null bits, which suppress stderr output

```
$ /home/pi/rtl-sdr/build/src/rtl_fm -p 0 -d 00000002 -g 32.8 -M fm -F9 -s 15k -f 402000000 |sox -t raw -r 15k -e s -b 16 -c 1 - -r 48000 -b 8 -t wav - highpass 20 |./imet1rs_dft --json -r
sox WARN wav: Length in output .wav header will be wrong since can't seek to fix it
Found 2 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000003
  1:  Realtek, RTL2838UHIDIR, SN: 00000002

Using device 1: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 32.80 dB.
Tuned to 402480000 Hz.
Oversampling input by: 128x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Sampling at 1920000 S/s.
Output at 15000 Hz.
sample_rate: 48000
bits       : 8
channels   : 1
```
At this point i get no more output, it just sits there, as it waiting for traffic. If there was a sonde being received, then you should see json (and other) output.

HoshenKadosh, 2022-02-21:
That's the output i was getting from running the command:
[image]

Strangely, still no errors. After more investigation, I found that sox , which is used wasn't installed for some reason. after installing it (`sudo apt install sox`), the decoder stayed open and decoded the sonde.

closed 2022-02-21 by HoshenKadosh