projecthorus/radiosonde_auto_rx #618: Too frequent transmit rate, synchronised to create a DOS attack at APRS-IS services

hessu, 2022-02-28:
Hi,

The sondegate software appears to send packets at a **very high** rate to the APRS-IS. For example:

https://aprs.fi/?c=raw&call=T1310381

Once a  minute on average should do fine, it seems to be alright for mostly everyone else on the APRS-IS and sondes should not receive special high-resolution treatment. If the rate is not reduced, I need to start filtering sondes off from aprs.fi to reduce load at this end.

Furthermore, it appears that the packets are sent every 30 seconds, synchronised to wall clock. This behaviour is explained in 
https://github.com/projecthorus/radiosonde_auto_rx/issues/373#issuecomment-769724610 . When all sondegates do this, there are two large synchronised bursts of packets every minute precisely at 12:00:00, 12:00:30, 12:01:00, 12:01:30, etc. The gateways are typically NTP synchronised, so they all do this blasting in unison. This fills up buffers on APRS-IS and clients receiving the data.

This needs to be be changed to send every 30 or preferably 60 seconds, starting from a **random** wall-clock offset which is **different at each gateway, preferably different offset for each sonde**. For example, every 30 seconds counting from the precise receive time of the first packet from a sonde. This dithering should preferably be done at a subsecond resolution. Only the latest packet must be sent at that interval.

73!

- Hessu, OH7LZB, aprs.fi

jtmnt, 2022-02-28:
> Packets must not be buffered and sent later

As I understand, buffering is there only to send one single latest packet after some time period. RS41 sonde sends different data in every 2 seconds.

TheSkorm, 2022-02-28:
3 seconds for a single station is certainly someone messing with the code. The config will usually clamp upload rate to 30 seconds : https://github.com/projecthorus/radiosonde_auto_rx/blob/a85a8d2fbe64ffb7d31d63bfc339d1d96b3546a8/auto_rx/autorx/config.py#L353

so this user has gone out of their way to have a higher upload rate it seems.

PR I put in moves the min upload rate up to 60 seconds

darksidelemm, 2022-02-28:
Hi @hessu , sorry about all this.

I wrote the synchronised upload part a long time ago, back when radiosonde_auto_rx only had a few users. Now there's ~600 users, it's clearly causing issues. 

I'm incredibly tempted to drop APRS-IS support in auto_rx altogether, or at least lock it down so that it only goes to radiosondy and not out to the wider aprs-is network (using port 14590 on the radiosondy server)... however I can guarantee that will cause complains ("why can't I see my callsign on aprs.fi!").

Also worth noting that it's not just radiosonde_auto_rx that sends data into APRS at this kind of rate. The other software packages you're going to need to try and deal with are:
- RS41Tracker (Closed source)
- dxlAPRS, and the many many forks of it. A bunch of the forks are closed. This is what is running on the rpi image radiosondy.info distributes.

hessu, 2022-02-28:
Hi,

It has been suggested that this doesn't belong on the APRS-IS, since APRS-IS is specifically built for carrying amateur radio traffic, and this clearly is non-ham traffic to start with. I kind of agree, although I do find this sonde stuff cool. So, it's quite OK if it goes to other networks. I can consider pulling the data to aprs.fi if it is available at a suitable data rate and format.

I know there are multiple software packages; they will need to be handled one by one. It'd be great if one would work right, then that one could be used as a standard example of a proper implementation.

jtmnt, 2022-02-28:
There is a lot of AIS traffic on APRS - how is this different and has this been handled?

hessu, 2022-02-28:
@jtmnt Where? I don't see any. :)

Note that aprs.fi is not APRS; aprs.fi behind the curtains supports APRS, AIS and ADS-B and Tetra, and a couple of other formats.

In this ticket we are discussing what goes to the APRS-IS network. I don't currently see much AIS on APRS-IS.

jtmnt, 2022-02-28:
Thanks, that answers it. Maybe weather balloons should not be regular APRS and be more like that. Only reason for it is to have them on the map for easy recovery.

TheSkorm, 2022-02-28:
> It has been suggested that this doesn't belong on the APRS-IS, since APRS-IS is specifically built for carrying amateur radio traffic, and this clearly is non-ham traffic to start with. I kind of [...]

The default config is aprs disabled with the default endpoint radiosonde.info without forwarding to aprs-is.

darksidelemm, 2022-03-05:
I'm going to close this one now, as v1.5.10 removes the synchronous upload option, and also limits the upload rate.

closed 2022-03-05 by darksidelemm