projecthorus/radiosonde_auto_rx #624: Gmail SMTP unsupported from May 30 (Email Notification)

JRus-design, 2022-03-04:
I received a message from Gmail, telling low secure access to a Gmail Account will be not supported from May 30. They recommend to use OAuth 2.0. Radiosonde auto rx can be configured to the new access?
Thanks,

darksidelemm, 2022-03-04:
Can you point to documentation as to how a developer is meant to send e-mail via this kind of system?

JRus-design, 2022-03-04:
[image]

Hi Mark!

Sorry I don´t understand very well your question. I use a dedicated gmail account to launch the notification mails, configured as attached. To use this logging method I need to cofigure the mail account to low secure access. I don't know is the correct method. 

Best regards

darksidelemm, 2022-03-04:
For a long time you've need to use 'device specific passwords' with gmail to be able to use SMTP. Are you saying that support for device-specific passwords is going to be dropped?

What i'm looking for is a worked example of how to send an e-mail with the new authentication system that gmail is requiring, and what users need to do to get the required codes.

JRus-design, 2022-03-04:
I send you the message received from Gmail. Sorry is in Spanish.

Best regards,

trevor229, 2022-06-09:
I am facing the same issue as well. Google sent the following email to users with the less secure apps option enabled, stating the following:

> On May 30, you may lose access to apps that are using less secure sign-in technology.
>
> To help keep your account secure, Google will no longer support the use of third-party apps or devices which ask you to sign in to your Google Account using only your username and password. Instead, you’ll need to sign in using [Sign in with Google] or other more secure technologies, like OAuth 2.0.

And links to [this Google support page](https://support.google.com/accounts/answer/6010255). To fix this, sign in with google or more likely OAuth2 will need to be implemented to fix email notification support.

trevor229, 2022-06-09:
I just tested "App passwords" with my Dell server iDRAC's which do not support OAuth or sign in with Google. They function perfectly for this application.

These essentially seem to function in the exact same way as the depricated "less secure app" setting which allowed account access without any other authentication for third party apps, but now there is per-application password control. You create a randomized password and give it a name in your account settings. You can view/copy the password only once. Using it combined with the email its associated with as a login works just as before. The only caveat is the account you use for email notifications *must* have 2 factor authentication enabled. [Here is Google's support page on creating app passwords](https://support.google.com/accounts/answer/185833)

Ultimately, implementing OAuth2 may be a better option for the future.

darksidelemm, 2022-06-13:
Yep, App passwords are what I'm using. 

As for OAuth support, I'd like to see it supported, but I'm really running short on time to work on this kind of thing at the moment.

rodrigoscatarino, 2025-12-15:
Mine is not working even with app passwords.
I've changed the snmp port to 587 and the auth to TLS. Will wait to see if it works.
Is there a way to trigger a test email using those settings?

darksidelemm, 2025-12-16:
There's info on how to test the email functionality here:
https://github.com/projecthorus/radiosonde_auto_rx/wiki/Configuration-Settings#e-mail-notifications

I'm surprised about app passwords not working. They are still available in gmail, though you appear to have to have 2-step verification enabled.