projecthorus/radiosonde_auto_rx #627: Does someone recognise this error during startup?

edegraaff, 2022-03-12:
Hi, 
i have installed the radiosonce_auto_rx and it seems to work fine but during the log checks i found this eror, should there be any action on this one?
Mar 12 20:28:47 raspberrypi auto_rx[1385]: 2022-03-12 20:28:47,358 ERROR:The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)

TheSkorm, 2022-03-12:
It can be ignored

edegraaff, 2022-03-12:
thanks for your fast reply, we are up ...  Is there a way to extract the data from a station from the sondehub btw? i tried to get some insights in the results of my tracking node but it seens not that easy.  This since once and a while i reinstall my pi with other or newer software. Waiting for a new pi4 for the dockers, now it is manually build on a pi3+ and works like a charm. Thanks for the project.

TheSkorm, 2022-03-12:
For the last month of data we can get it out of elastic search if really needed, but we generally suggest using our opendata platform to download the serials you need https://github.com/projecthorus/pysondehub

edegraaff, 2022-03-12:
Thanks, are there some guides how to start with this, i am no programmer, and it seems to me the example is a bit outdated. I have an running Sagemaker now but here i fail. There is also an option to select a kernel.

TheSkorm, 2022-03-12:
What kind of information are you looking for? Autorx will save csv logs locally which may be easier to work with depending on what you are after

edegraaff, 2022-03-12:
The  PI's have a nasty habbit, sometimes the SD card will fail and then data will be lost. What i was searching for was some sort of statistics collected by sondehub similar as habhub does. Some diagrams regarding the receiving stations, and or insights regarding the data collected. Sonde info is for me quite new, and i am still puzzeled if i will be able to use the sondehub data and my noaa satelite info as some sort of self made weather forecast :-)

TheSkorm, 2022-03-12:
I don't really know what habhub stats your referring - maybe you could share screenshost? If you have `jq` installed you can get some basic stats per sonde. 
```
$ sondehub --download S5110315 | jq -r '.uploader_callsign' | sort | uniq -c
7026 #HAB1_AUTO_RX
7042 Bigears
2491 CJAST
6843 Noddy_._
7069 VK3BQZ_AUTO_RX
6651 VK3GIS
   2 VK3JUG_AUTO_RX
3541 VK3TMC
6447 VK3TWC
6579 VK3ZYC__RX
6660 Yarra_Valley
```

Or you could generate a CSV of SNR values
```
sondehub --download S5110315 | jq -r '"\(.datetime),\(.uploader_callsign),\(.snr)"'
...
2022-03-12T13:20:11.000000Z,Bigears,27.8
2022-03-12T13:20:11.000000Z,VK3TWC,16
2022-03-12T13:20:12.000000Z,Bigears,24.8
2022-03-12T13:20:12.000000Z,VK3TWC,16.2
2022-03-12T13:20:13.000000Z,Bigears,26.5
2022-03-12T13:20:13.000000Z,VK3TWC,16.4
2022-03-12T13:20:14.000000Z,VK3TWC,16.4
2022-03-12T13:20:14.000000Z,Bigears,26.5
2022-03-12T13:20:15.000000Z,VK3TWC,16
2022-03-12T13:20:15.000000Z,Bigears,29.2
2022-03-12T13:20:16.000000Z,Bigears,29.2
2022-03-12T13:20:16.000000Z,VK3TWC,16
2022-03-12T13:20:17.000000Z,Bigears,24.6
2022-03-12T13:20:17.000000Z,VK3TWC,15.6
2022-03-12T13:20:18.000000Z,Bigears,23.1
2022-03-12T13:20:18.000000Z,VK3TWC,13.1
2022-03-12T13:20:19.000000Z,Bigears,28
2022-03-12T13:20:19.000000Z,VK3TWC,12.5
2022-03-12T13:20:20.000000Z,Bigears,28
2022-03-12T13:20:20.000000Z,VK3TWC,12.1
2022-03-12T13:20:21.000000Z,VK3TWC,10.1
2022-03-12T13:20:21.000000Z,Bigears,28.5
2022-03-12T13:20:22.000000Z,Bigears,28.5
2022-03-12T13:20:23.000000Z,Bigears,30.5
2022-03-12T13:20:23.000000Z,VK3TWC,9.1
2022-03-12T13:20:25.000000Z,Bigears,27
2022-03-12T13:20:26.000000Z,Bigears,27
2022-03-12T13:20:27.000000Z,Bigears,26.1
2022-03-12T13:20:30.000000Z,Bigears,14.8
```

edegraaff, 2022-03-12:
Seems we are more or less think the same way. I have a pi with Radiosonde Auto-RX 1.5.10 installed. 
I have an account on AWS and Amazon SageMaker >Notebook instances > nl13974-sonde avalible. 
From there i would create some output based on your examples for the Bilt and for nl13974

darksidelemm, 2022-03-12:
As an aside, if you want to gather data received by your pi and send it to your own database, received telemetry is emitted on UDP port 55673 by default in JSON format. 

Anyway, its sounding like this isn't an auto_rx issue at this point.

closed 2022-03-12 by darksidelemm