projecthorus/radiosonde_auto_rx #634: Error determining version

rgf123, 2022-05-11:
Hi,
From latest update to 1.5.10 I get this error message in log:

ERROR:Version - Error determining version from URL https://raw.githubusercontent.com/projecthorus/radiosonde_auto_rx/master/auto_rx/autorx/__init__.py: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Read timed out. (read timeout=5)

ERROR:Version - Could not determine latest versions.

Is this something could I fix?

Thanks.

darksidelemm, 2022-05-11:
Given that URL ( https://raw.githubusercontent.com/projecthorus/radiosonde_auto_rx/master/auto_rx/autorx/__init__.py ) is working, this suggests a connection issue from where you are running auto_rx to github.

auto_rx should continue to run OK, It just won't be able to display if there's any newer versions available.

rgf123, 2022-05-11:
It runs OK even with that error, you are right.
Running in raspberry CLI:
wget https://raw.githubusercontent.com/projecthorus/radiosonde_auto_rx/master/auto_rx/autorx/__init__.py

--2022-05-12 00:24:44--  https://raw.githubusercontent.com/projecthorus/radiosonde_auto_rx/master/auto_rx/autorx/__init__.py
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response...

So, it connects but can't find the file?

darksidelemm, 2022-05-11:
In the case of your error from auto_rx, the request is timing out: 
```
HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Read timed out. (read timeout=5)
```

rgf123, 2022-05-11:
Yes, so both are failing:
- auto_rx internal process times out
- requesting the file from cli, it also fails (it was 'awaiting response....' minutes), the timeout was a CTRL+C

I think it is related to availability of the server. I have tested several times from CLI and I got the file once from many attempts.

darksidelemm, 2022-05-11:
Yep. In which case there's not a whole lot I can do about this one, sorry!

rgf123, 2022-05-11:
You are right.
Could I configure logging to only show INFO messages?

darksidelemm, 2022-05-11:
No - inhibiting errors like this is generally is a bad idea. It should only appear once on startup anyway...

rgf123, 2022-05-11:
Understood.
Thanks.

closed 2022-05-11 by rgf123