projecthorus/radiosonde_auto_rx #635: no live data

jojoradio, 2022-05-19:
Since the new web interface of radiosonde_auto_rx was introduced I don't get live data anymore. I always have to reload the webpage to see the newest data. Also all sondes also, the one that is received live by the RTL-SDR at this very moment is marked as "old" in the table.
I am running Radiosonde_auto_rx 1.5.10, buuild from source on a raspberry pi 2 running buster.
Has anyone an Idea how to solve the issue?
[image]

jojoradio, 2022-05-19:
That is weird: ntpd always terminates after some minutes and i am not able to enable ntp at timedatectl.

Altosaxophony, 2022-05-19:
Looks like you should remove ntp/ntpd so you will be able to use timedatectl then.
_# sudo apt purge ntp_
(please see "Step 3" @ https://www.digitalocean.com/community/tutorials/how-to-set-up-time-synchronization-on-debian-10 )

My Raspi 3 runs on the "original" Debian 11 Bullseye flashed from the Raspi Pi Imager and it has no ntp/ntp, timedatectl works just fine:
:~ $ timedatectl status
               Local time: Thu 2022-05-19 21:49:09 CEST
           Universal time: Thu 2022-05-19 19:49:09 UTC
                 RTC time: n/a
                Time zone: Europe/Zurich (CEST, +0200)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no

Maybe it is a faster solution to reflash your Raspi with Bullseye OS (i use version 04-04-2022) and then reinstall radiosonde_auto_rx from scratch. So you dont have to mess around with ntp/ntpd. 
Just backup the config file first so you can later restore it to the fresh install.

darksidelemm, 2022-05-19:
This issue is very likely nothing to do with NTP.

This is most likely due to a mismatch between the flask-socketio version and the socketIO.js library included by auto_rx, used in the web interface. You can confirm this by opening your browsers 'Web Console' (or sometimes called a 'Browser Console', or 'Javascript console', often under 'Developer Tools'), and looking for an error which will be shown on startup. You will also see a similar error in the auto_rx logs indicating a version mismatch.

Please try updating flask-socketio using the command: sudo pip install -U flask-socketio

darksidelemm, 2022-05-19:
As another note, to avoid issues with dependencies, I highly recommend using the Docker image method of installing auto_rx now: https://github.com/projecthorus/radiosonde_auto_rx/wiki/Docker

closed 2022-12-22 by darksidelemm