projecthorus/radiosonde_auto_rx #641: no DFM decode with 1.60b1

F1TKE, 2022-07-12:
Today with dfm sonde
Jul 12 09:49:53 F1TKE-RS auto_rx[1133]: 2022-07-12 09:49:53,276 INFO:Task Manager - Detected new DFM sonde on 403.249 MHz!
Jul 12 09:49:53 F1TKE-RS auto_rx[1133]: 2022-07-12 09:49:53,280 INFO:Halting Scanner to decode detected radiosonde.
Jul 12 09:49:53 F1TKE-RS auto_rx[1133]: 2022-07-12 09:49:53,282 INFO:Scanner (RTLSDR 1) - Waiting for current scan to finish...
Jul 12 09:50:08 F1TKE-RS kernel: [ 1423.673758] Voltage normalised (0x00000000)
Jul 12 09:50:13 F1TKE-RS auto_rx[1133]: 2022-07-12 09:50:13,765 INFO:Scanner (RTLSDR 1) - Scanner Thread Closed.
Jul 12 09:50:13 F1TKE-RS auto_rx[1133]: 2022-07-12 09:50:13,767 INFO:Task Manager - SDR #1 has been allocated to Decoder (DFM, 403.249 MHz).
Jul 12 09:50:16 F1TKE-RS auto_rx[1133]: 2022-07-12 09:50:16,958 INFO:Decoder (RTLSDR 1) DFM 403.249 - Using fsk_demod decoder chain.
Jul 12 09:50:17 F1TKE-RS auto_rx[1133]: 2022-07-12 09:50:17,040 INFO:Decoder (RTLSDR 1) DFM 403.249 - Starting decoder subprocess.
Jul 12 09:52:16 F1TKE-RS auto_rx[1133]: 2022-07-12 09:52:16,984 ERROR:Decoder (RTLSDR 1) DFM 403.249 - RX Timed out.
Jul 12 09:52:17 F1TKE-RS auto_rx[1133]: 2022-07-12 09:52:17,991 INFO:Decoder (RTLSDR 1) DFM 403.249 - Closed decoder subprocess.
Jul 12 09:52:19 F1TKE-RS auto_rx[1133]: 2022-07-12 09:52:19,100 INFO:Task Manager - SDR #1 has been allocated to Scanner.
Jul 12 09:52:22 F1TKE-RS auto_rx[1133]: 2022-07-12 09:52:22,408 INFO:Scanner (RTLSDR 1) - Starting Scanner Thread
Jul 12 09:52:22 F1TKE-RS auto_rx[1133]: 2022-07-12 09:52:22,417 INFO:Scanner (RTLSDR 1) - Running frequency scan.

F1TKE, 2022-07-12:
i look with another dfm if decode or not

darksidelemm, 2022-07-12:
Hmm, I did test with a DFM17 locally before release.

F1TKE, 2022-07-12:
with another is ok, probably because too close in frequency to a qrm on 403.200.

darksidelemm, 2022-07-12:
Hmm, 50 kHz off *should* be ok. We've successfully decoded sondes on 403.210 MHz before (with some changes which are in this release)

darksidelemm, 2022-07-12:
The errors you just pasted relate to a RS41 without GPS lock? (based on the serial number).

F1TKE, 2022-07-12:
oups 
I had visitors at the same time who pressed an rs-41. I redid a test with a dfm09

F1TKE, 2022-07-12:
arf all my dfm09 are dead. i watch with another flying dfm09 if i have the same prb

darksidelemm, 2022-07-12:
I'll try out a DFM09 on the bench here too

darksidelemm, 2022-07-12:
Yep, there was an issue with the JSON output on the DFM decoder. There was a '+' added in the front of one of the fields (`"diff_GPS_MSL": +0.71`), which is not JSON compatible, resulting in auto_rx discarding all the data. 

I've just merged in a patch to the testing branch, and a new docker image will come out in an hour or so.

F1TKE, 2022-07-12:
upgrade done , start testing beta2