projecthorus/radiosonde_auto_rx #643: IMET4 decoder subprocess timing out

RavenV-tiff, 2022-07-13:
Hi,
I have a problem with auto_rx when trying to pickup iMet-4RSB sondes. The sondes are being detected but the decoder always times out. I do not have this issue with other sonde types. My hardware is the Nesdr smart SDR dongle with a yagi antenna. I have attached a screen shot of the output of the program.
[image]
While the screenshot shows version 1.5.9 being ran, I have tested with the latest 1.5.10 version with the same results. I don't know if this is a different variant of imet-4 from which auto_rx can decipher.

RavenV-tiff, 2022-07-14:
It may be some time before I get the next opportunity to check the program. Our field study ended, but we do still have some RSBs remaining. So next time we send a payload up I'll try recording audio and provide the output with -v.

darksidelemm, 2022-07-14:
Ahh damn. If you could turn one on on the ground and get a recording that would be really useful!

darksidelemm, 2022-12-22:
Any updates on this? A recording of an iMET4-RSB, in particular with a XDATA payload attached would be handy.

RavenV-tiff, 2022-12-23:
Unfortunately we haven't had any deployments since I last posted. I will discuss with my lab coordinator about opening up a sonde to capture the radio signals with XDATA from an ozonesonde we recovered. They may say not to until next deployment, which I really cannot so for sure when that will be.

RavenV-tiff, 2023-01-27:
@darksidelemm I received permission to open an iMET4-RSB. I will attach an ECC ozonesonde to it, so it will have valid XDATA. In terms of audio recording how would you like me to do it? I can simply record the computer audio with something like Audacity and send you the wave file with the decoded data. Would this work? How long of a sample would you need?

darksidelemm, 2023-01-27:
Maybe a minute should be fine. FM-demodulated audio (like you would use with the iMet software) is all we need for this.

closed 2023-02-03 by darksidelemm