projecthorus/radiosonde_auto_rx #650: MEISEI decoding not work with spyserver

papasven, 2022-08-11:
Small problem with decoding, unfortunately MEISEI does not work with spyserver networking option.
version 1.6.0-beta4

2022-08-11 16:31:43,883 INFO:Web - Got decoder start request: MEISEI, 405602000.000000
2022-08-11 16:31:43,989 DEBUG:Telemetry Logger - Wrote line: 2022-08-11T15:31:37.000Z,DFM-21052258,1344267097,51.46637,12.70697,1678.3,-3.0,6.8,263.4,12.6,-1.0,-1.0,DFM17,402.810,13.3,0,20,5.2,-1,-1
2022-08-11 16:31:43,990 DEBUG:Telemetry Logger - Wrote line: 2022-08-11T15:31:38.000Z,DFM-21052258,1344267098,51.46637,12.70688,1675.5,-2.7,6.3,267.8,12.6,-1.0,-1.0,DFM17,402.810,11.7,0,20,5.2,-1,-1
2022-08-11 16:31:43,991 DEBUG:Telemetry Logger - Wrote line: 2022-08-11T15:31:39.000Z,DFM-21052258,1344267099,51.46637,12.70678,1672.7,-2.8,6.6,266.5,12.7,-1.0,-1.0,DFM17,402.810,11.7,0,20,5.2,-1,-1
2022-08-11 16:31:43,992 DEBUG:Telemetry Logger - Wrote line: 2022-08-11T15:31:40.000Z,DFM-21052258,1344267100,51.46636,12.70669,1669.8,-3.1,6.3,268.5,12.7,-1.0,-1.0,DFM17,402.810,11.7,0,20,5.2,-1,-1
2022-08-11 16:31:43,994 DEBUG:Telemetry Logger - Wrote line: 2022-08-11T15:31:41.000Z,DFM-21052258,1344267101,51.46636,12.70660,1666.6,-3.1,6.4,270.0,12.7,-1.0,-1.0,DFM17,402.810,11.7,0,20,5.2,-1,-1
2022-08-11 16:31:45,322 DEBUG:Sondehub Uploader - Pre-compression: 10676 bytes, post: 1182 bytes. 11.1 % compression ratio, in 0.0 s
2022-08-11 16:31:45,354 INFO:Task Manager - Detected new MEISEI sonde on 405.602 MHz!
2022-08-11 16:31:45,354 INFO:Task Manager - SDR #SPY05 has been allocated to Decoder (MEISEI, 405.602 MHz).
2022-08-11 16:31:45,355 DEBUG:SpyServer - Testing using command: timeout 10 ./ss_iq -f 401500000 -s 48000 -r localhost -q 5555 -n 48000 - > /dev/null 2> /dev/null
2022-08-11 16:31:45,634 DEBUG:Delta time: 3
2022-08-11 16:31:45,738 DEBUG:Delta time: 2
2022-08-11 16:31:45,858 INFO:Sondehub Uploader - Uploaded 17 telemetry packets to Sondehub in 0.5 seconds.
2022-08-11 16:31:45,998 DEBUG:Telemetry Logger - Wrote line: 2022-08-11T15:31:42.000Z,DFM-21052258,1344267102,51.46636,12.70651,1663.5,-3.1,6.5,269.3,12.8,-1.0,-1.0,DFM17,402.810,11.7,0,20,5.2,-1,-1
2022-08-11 16:31:46,000 DEBUG:Telemetry Logger - Wrote line: 2022-08-11T15:31:43.000Z,DFM-21052258,1344267103,51.46637,12.70642,1660.3,-3.3,6.4,272.3,12.8,-1.0,-1.0,DFM17,402.810,11.2,0,20,5.2,-1,-1
2022-08-11 16:31:48,550 DEBUG:Delta time: 3
2022-08-11 16:31:50,007 DEBUG:Telemetry Logger - Wrote line: 2022-08-11T15:31:45.000Z,DFM-21052258,1344267105,51.46637,12.70623,1654.0,-3.1,6.4,273.3,12.9,-1.0,-1.0,DFM17,402.810,11.4,0,20,5.2,-1,-1
2022-08-11 16:31:50,830 DEBUG:Delta time: 3
2022-08-11 16:31:51,011 DEBUG:Telemetry Logger - Wrote line: 2022-08-11T15:31:47.000Z,DFM-21052258,1344267107,51.46636,12.70604,1647.8,-3.1,6.7,270.4,12.9,-1.0,-1.0,DFM17,402.810,12.1,0,20,5.2,-1,-1
2022-08-11 16:31:51,118 CRITICAL:FM Demod Source - Unsupported SDR type SpyServer
2022-08-11 16:31:51,119 DEBUG:Decoder (SpyServer localhost:5555) MEISEI 405.602 - Opening /home/shorty/radiosonde_auto_rx/auto_rx/log/20220811-153145_MEISEI_405602000.raw to save decoder raw data.
2022-08-11 16:31:51,121 DEBUG:Decoder (SpyServer localhost:5555) MEISEI 405.602 - Decoder Command: false |./meisei100mod --json --ptu --ecc 2>/dev/null
2022-08-11 16:31:51,162 INFO:Decoder (SpyServer localhost:5555) MEISEI 405.602 - Starting decoder subprocess.
2022-08-11 16:31:52,267 INFO:Decoder (SpyServer localhost:5555) MEISEI 405.602 - Closed decoder subprocess.

darksidelemm, 2022-08-11:
Yep, this is correct. Currently we are using the Meisei decoder with a FM demodulator, and the spyserver software only outputs IQ.

If you're happy to do some testing for me, I can make some changes this weekend to switch the Meisei decoder to use IQ input, and we can see how well that works.

papasven, 2022-08-11:
Ok thanks I'm ready.
There are a lot of radiosondes in the air for the next few weeks

https://www.gruan.org/community/campaigns/uaii2022

darksidelemm, 2022-08-11:
OK, I wont be able to get this done until Saturday morning at the earliest. 
I'll make a change in the testing branch to switch the Meisei decoder over to using IQ, with some parameters as advised by @rs1729 
This will be released as a new docker image as well, but its worth noting that this may completely break meisei decoding if I get the parameters wrong.

Do you have a good idea of when the launches will be occuring?

papasven, 2022-08-12:
The launches will be over 4 weeks from now, 80 launches in total.

https://community.wmo.int/activity-areas/imop/intercomparisons

darksidelemm, 2022-08-12:
OK, so the testing branch of auto_rx (1.6.0-beta6) now has a new configuration option here: https://github.com/projecthorus/radiosonde_auto_rx/blob/testing/auto_rx/station.cfg.example#L533

If set to `True`, this will enable an IQ decode version of the Meisei demodulator. This is entirely untested.

If this doesn't work, the command to run on the command-line is:
```
rtl_fm -M raw -F9 -p 0 -d 0 -s 48000 -f 401500000 - 2>/dev/null | ./meisei100mod --IQ 0.0 --lpIQ --dc  - 48000 16 --json --ptu --ecc
```

Replacing the frequency in this example (401500000) with the relevant frequency. Let me know if that produces any output.

papasven, 2022-08-12:
Works thank you, now quickly trace the radiosondes. ;)

darksidelemm, 2022-08-12:
Aha, I see them on tracker.sondehub.org :-)

Still, an IQ recording over a flight would be be useful, for assessing the drift, and how the demodulators handle it.

darksidelemm, 2022-08-20:
@papasven  i've just released an update in the testing branch (1.6.0-beta7), where i've switched the 'normal' meisei decoder to be the IQ one that you've been using, and the 'experimental' one to using fsk_demod, like most of the other decoder chains.

Could you please give this a go? If this work, it will provide accurate frequency information, and SNR to SondeHub.

papasven, 2022-08-21:
I have updated.
On Monday the radiosonde spectacle continues at 08:00 UTC, then I also make recordings of meisei with rtl_fm.

darksidelemm, 2022-08-21:
Thanks! I've already gotten a short bit of IQ from Rolf, but this was only from the upper part of a flight. I'm guessing you are not receiving below ~23km or so?

darksidelemm, 2022-08-24:
That recording from the 17th was awesome :-) Great dataset for testing the decoder!

It looks like the fsk_demod decode chain is working fine - I can see you are running multiple rx stations with it now?

papasven, 2022-08-26:
> That recording from the 17th was awesome :-)

2 more files with rtl-fm raw recording are now on the server, Rolf can give you the address.
It took a while now, there were just too many radiosondes. ;)

 > I can see you are running multiple rx stations with it now?

I only have one station running and one rdzTTGOsonde with the same SSID. I will change the SSID.

darksidelemm, 2022-08-26:
Hah its all OK :-)

The first recording was enough for me to characterise the drift of the Meisei iMS100. I think the current settings in auto_rx are sufficient to decode an iMS100 throughout flight - now to get it on to some stations which are close to iMS100 launch sites and see how it goes in the wild!

closed 2022-11-20 by darksidelemm