projecthorus/radiosonde_auto_rx #653: not able to log into browser :5000

vkkanga, 2022-08-25:
Docker build on Raspberry Pi 3. Have rebuilt, same issue.

pi@raspberrypi:~ $ docker logs --tail 50 --follow radiosonde_auto_rx

2022-08-25 06:59:31,743 CRITICAL:Error in configuration file! Exiting...
2022-08-25 07:00:37,085 INFO:Reading configuration file...
2022-08-25 07:00:37,115 WARNING:Config - Web Password not set, disabling web control
2022-08-25 07:00:37,116 WARNING:Please do not use APRS ports which forward data 
out to the wider APRS-IS network and cause network congestion. Switching to defa
ult port of 14590. If you believe this to be in error, please raise an issue at 
https://github.com/projecthorus/radiosonde_auto_rx/issues
2022-08-25 07:00:37,217 ERROR:RTLSDR - RTLSDR with serial #0 is not present!
2022-08-25 07:00:37,219 WARNING:Config - SDR #0 invalid.
2022-08-25 07:00:37,220 ERROR:Config - No working SDRs! Cannot run...
2022-08-25 07:00:37,221 CRITICAL:Error in configuration file! Exiting...
2022-08-25 07:01:42,447 INFO:Reading configuration file...

darksidelemm, 2022-08-25:
The issue is stated in the log:
```
2022-08-25 07:00:37,217 ERROR:RTLSDR - RTLSDR with serial #0 is not present!
2022-08-25 07:00:37,219 WARNING:Config - SDR #0 invalid.
2022-08-25 07:00:37,220 ERROR:Config - No working SDRs! Cannot run..
```

This suggests that the RTLSDR USB device is not accessible from the docker image. 

Please check the RTLSDR exists using `lsusb`, and that you have done the required [module blacklisting](https://github.com/projecthorus/radiosonde_auto_rx/wiki/Docker#12-rtl-sdr-kernel-blacklisting) steps and then rebooted your RPi.

vkkanga, 2022-08-25:
pi@raspberrypi:~ $ lsusb
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Have not touched the SDR (nooelec)

sudo shutdown -r  working again.

darksidelemm, 2022-08-25:
There's no Nooelec SDR showing up in that list! That suggests something is not right with the SDR...

closed 2022-09-24 by darksidelemm