projecthorus/radiosonde_auto_rx #654: raise RuntimeError('The Werkzeug web server is not '

Szetya, 2022-08-29:
I catched message in systemctl status. 
Next message: RuntimeError: The Werkzeug web server is not designed to run in production. Pass allow_unsafe_werkzeug=True to the run() method to disable this error.
The  line of the web.py file has been modified as follows. socketio.run(app, host=host, port=port, allow_unsafe_werkzeug=True)
The web server is run now.

darksidelemm, 2022-08-29:
I suspect this is a change in a recent version of werkzeug, brought in by recent versions of flask / flask-socketio. 

I'll have to add this in, in a way that doesnt break previous versions. Thanks!

darksidelemm, 2022-08-29:
Actually, could you try and check what versions of the various libs you have? I just tried in a fresh venv and am not getting this error. 

When installed the packages into a venv (pip install -r requirements.txt) I ended up with:
```
Successfully installed Jinja2-3.1.2 MarkupSafe-2.1.1 Werkzeug-2.2.2 bidict-0.22.0 certifi-2022.6.15 charset-normalizer-2.1.1 click-8.1.3 crcmod-1.7 flask-2.2.2 flask-socketio-5.3.0 idna-3.3 importlib-metadata-4.12.0 itsdangerous-2.1.2 numpy-1.23.2 python-dateutil-2.8.2 python-engineio-4.3.4 python-socketio-5.7.1 requests-2.28.1 semver-2.13.0 simplekml-1.3.6 six-1.16.0 urllib3-1.26.12 zipp-3.8.1
```

darksidelemm, 2022-08-29:
Oh I think i see where the problem will be here - it will probably only manifest in either a docker container, or running as a systemd service, because there is no TTY available (based on this check: https://github.com/miguelgrinberg/Flask-SocketIO/commit/e35a0f4a69c343412cdfb879e7545707f933934a#diff-d4032ca32e5dce64b1e46683f4bc8544ccefadb6e8f3e09a62152dda9c0a0578R600 )

Szetya, 2022-08-29:
> Valójában meg tudnád nézni, hogy milyen verziói vannak a különböző libeknek? Most próbáltam egy friss venv-ben, és nem kapom ezt a hibát.
> 
> Amikor telepítettem a csomagokat egy venv-be (pip install -r követelmények.txt), a következőt kaptam:
> 
> ```
> Successfully installed Jinja2-3.1.2 MarkupSafe-2.1.1 Werkzeug-2.2.2 bidict-0.22.0 certifi-2022.6.15 charset-normalizer-2.1.1 click-8.1.3 crcmod-1.7 flask-2.2.2 flask-socketio-5.3.0 idna-3.3 importlib-metadata-4.12.0 itsdangerous-2.1.2 numpy-1.23.2 python-dateutil-2.8.2 python-engineio-4.3.4 python-socketio-5.7.1 requests-2.28.1 semver-2.13.0 simplekml-1.3.6 six-1.16.0 urllib3-1.26.12 zipp-3.8.1
> ```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: crcmod in /usr/lib/python3/dist-packages (from -r requirements.txt (line 1)) (1.7)
Requirement already satisfied: python-dateutil in /usr/lib/python3/dist-packages (from -r requirements.txt (line 2)) (2.8.1)
Requirement already satisfied: flask in /usr/lib/python3/dist-packages (from -r requirements.txt (line 3)) (1.1.2)
Requirement already satisfied: flask-socketio in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 4)) (5.3.0)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from -r requirements.txt (line 5)) (1.19.5)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from -r requirements.txt (line 6)) (2.25.1)
Requirement already satisfied: semver in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 7)) (2.13.0)
Requirement already satisfied: simplekml in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 8)) (1.3.6)
Requirement already satisfied: python-socketio>=5.0.2 in /usr/local/lib/python3.9/dist-packages (from flask-socketio->-r requirements.txt (line 4)) (5.7.1)
Requirement already satisfied: python-engineio>=4.3.0 in /usr/local/lib/python3.9/dist-packages (from python-socketio>=5.0.2->flask-socketio->-r requirements.txt (line 4)) (4.3.0)
Requirement already satisfied: bidict>=0.21.0 in /usr/local/lib/python3.9/dist-packages (from python-socketio>=5.0.2->flask-socketio->-r requirements.txt (line 4)) (0.21.4)

Yes! With manual running ($ python auto_rx.py) it works without problems.

darksidelemm, 2022-08-29:
Yep OK, I'll have to add a fix in! For now, pin flask-socketio to 5.2.0 using: 
```
pip3 install flask-socketio==5.2.0
```

darksidelemm, 2022-12-22:
Noting that this is fixed in the testing branch.

closed 2023-03-10 by darksidelemm