projecthorus/radiosonde_auto_rx #658: raspberry pi os native install - web interface can't be reached

gurgeasson, 2022-08-30:
After following instructions 'native' auto_rx install, I run into a problem. I can't reach the web interface once it's in 'automatic operation'.
The testing phase works fine, http://localhost:5000/ - localhost substituted to pi's ip address - shows up, but once it runs as systemd service, the site is unreachable. Everything else seems to be running all right:
● auto_rx.service - auto_rx
     Loaded: loaded (/etc/systemd/system/auto_rx.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-08-29 21:25:26 BST; 23h ago
   Main PID: 370 (python3)
      Tasks: 16 (limit: 4915)
        CPU: 4h 46min 32.881s
     CGroup: /system.slice/auto_rx.service
             ├─  370 /usr/bin/python3 /home/m-flight-a/radiosonde_auto_rx/auto_rx/auto_rx.py -t 0
             ├─17789 /bin/sh -c timeout 10s rtl_fm -p 0 -d 02220702 -g 48.0 -M raw -F9 -s 48000 -f 403900000 2>/dev/null |./dft_detect -t 5 --iq --bw 20>
             ├─17790 timeout 10s rtl_fm -p 0 -d 02220702 -g 48.0 -M raw -F9 -s 48000 -f 403900000
             └─17792 rtl_fm -p 0 -d 02220702 -g 48.0 -M raw -F9 -s 48000 -f 403900000

raspberry pi 4, running headless raspberry pi os 5.15.56-v7l+
python version: 3.9.2-3

What should I check, or where should I start to find out what's causing this? Cheers

Szetya, 2022-08-30:
Try to install 
flask 2.2.2
flask-socketio 5.2.0
Or install requiements from requiements.txt
$ sudo pip3 install -r requirements.txt
Issue #654  is same problem.

darksidelemm, 2022-08-30:
Installing from the requirements.txt as per the instructions should have solved this, but i did only update that file in the master branch a few days ago.

In the testing branch is a more comprehensive fix, which allows the use of the newer version of flask-socketio.

gurgeasson, 2022-08-30:
Yes, you're right, $ sudo pip3 install -r requirements.txt worked.
As issue #658 is same as #654 is suppose I can close it safely. Right?
Thanks for your help, it's much appreciated.

darksidelemm, 2022-08-30:
Yes, this issue can be closed, we can track the problem in the other issue.