projecthorus/radiosonde_auto_rx #659: 1.6.0 Beta 8 no new sondes decoded, running sondes keep working

zanco, 2022-09-10:
Last week auto_rx was decoding sondes from Germany, but after the ozone launch from the Netherlands that one did not appear in the list nor on my map. Also some DFM's got launched which did not appear. 
I switched to the auto_rx terminal and copied some sort of error message.

Restarted auto_rx and all was up and running ok

The error message: after which the upload continued for the running sondes

2022-09-07 13:26:29,194 ERROR:Error on request:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 306, in run_wsgi
    execute(self.server.app)
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 294, in execute
    application_iter = app(environ, start_response)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2463, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/local/lib/python3.8/dist-packages/flask_socketio/__init__.py", line 43, in __call__
    return super(_SocketIOMiddleware, self).__call__(environ,
  File "/usr/local/lib/python3.8/dist-packages/engineio/middleware.py", line 63, in __call__
    return self.engineio_app.handle_request(environ, start_response)
  File "/usr/local/lib/python3.8/dist-packages/socketio/server.py", line 597, in handle_request
    return self.eio.handle_request(environ, start_response)
  File "/usr/local/lib/python3.8/dist-packages/engineio/server.py", line 409, in handle_request
    socket = self._get_socket(sid)
  File "/usr/local/lib/python3.8/dist-packages/engineio/server.py", line 638, in _get_socket
    raise KeyError('Session is disconnected')
KeyError: 'Session is disconnected'
2022-09-07 13:26:30,232 INFO:Sondehub Uploader - Uploaded 41 telemetry packets to Sondehub in 0.5 seconds.

darksidelemm, 2022-09-10:
Unfortunately this error is likely not related. The 'Session is Disconnected' error will show up whenever a web interface client (e.g. a browser window) closes, or otherwise stops communicating with the auto_rx server.

zanco, 2022-11-09:
Hi Mark, unfortunately the same thing happened today, Nicely decoded RS-41 ozone on 403.5 but not the other sondes. Swithced to the desktop of the auto_rx system and the same message was there

2022-11-09 13:30:40,532 INFO:Sondehub Uploader - Uploaded 22 telemetry packets to Sondehub in 0.5 seconds.
2022-11-09 13:30:53,341 ERROR:Error on request:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 306, in run_wsgi
    execute(self.server.app)
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 294, in execute
    application_iter = app(environ, start_response)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2463, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/local/lib/python3.8/dist-packages/flask_socketio/__init__.py", line 43, in __call__
    return super(_SocketIOMiddleware, self).__call__(environ,
  File "/usr/local/lib/python3.8/dist-packages/engineio/middleware.py", line 63, in __call__
    return self.engineio_app.handle_request(environ, start_response)
  File "/usr/local/lib/python3.8/dist-packages/socketio/server.py", line 597, in handle_request
    return self.eio.handle_request(environ, start_response)
  File "/usr/local/lib/python3.8/dist-packages/engineio/server.py", line 429, in handle_request
    socket = self._get_socket(sid)
  File "/usr/local/lib/python3.8/dist-packages/engineio/server.py", line 638, in _get_socket
    raise KeyError('Session is disconnected')

What can I do to help debug this problem ?

darksidelemm, 2022-11-09:
As i've mentioned before, the "Session is Disconnected" error is not the issue. This is just due to a web browser client disconnecting.

I've got some changes I need to make to auto_rx to allow enabling of debug logging from within the configuration file. Once those changes are made I'll get you to turn that on, and maybe that log will show something more useful.

darksidelemm, 2022-11-11:
I've just made these changes to auto_rx testing branch (-beta14).

The new configuration options are here: https://github.com/projecthorus/radiosonde_auto_rx/blob/testing/auto_rx/station.cfg.example#L402

I would enable both the system log, and debug logging. This will show every command run, and the output of each command. This will be a lot of logs!

F1TKE, 2022-11-11:
Hello

I have upgrade to beta14 but 

2022-11-11 11:29:59 UTC – CRITICAL
Decoder (RTLSDR 1) RS41 402.300 - Decoder version (1.6.0-beta13) does not match auto_rx version (1.6.0-beta14). Have you re-built the decoders? (./build.sh)

 

I do two re-build

No warning message

And the one reboot

 

***@***.***:~/radiosonde_auto_rx/auto_rx $ git pull

Already up to date.

***@***.***:~/radiosonde_auto_rx/auto_rx $ ./build.sh

Building for radiosonde_auto_rx version: "1.6.0-beta14"

make: Entering directory '/home/pi/radiosonde_auto_rx'

make -C demod/mod clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/demod/mod'

rm -f rs41mod dfm09mod rs92mod lms6mod lms6Xmod meisei100mod m10mod m20mod imet5                                                                                                                                                             4mod mp3h1mod meteosis_mod rs41mod.c.o dfm09mod.c.o rs92mod.c.o lms6mod.c.o lms6                                                                                                                                                             Xmod.c.o meisei100mod.c.o m10mod.c.o m20mod.c.o imet54mod.c.o mp3h1mod.c.o meteo                                                                                                                                                             sis_mod.c.o bch_ecc_mod.c.o demod_mod.c.o

cc   rs41mod.o demod_mod.o bch_ecc_mod.o  -lm -o rs41mod

cc   dfm09mod.o demod_mod.o  -lm -o dfm09mod

cc   rs92mod.o demod_mod.o bch_ecc_mod.o  -lm -o rs92mod

cc   lms6mod.o demod_mod.o bch_ecc_mod.o  -lm -o lms6mod

cc   lms6Xmod.o demod_mod.o bch_ecc_mod.o  -lm -o lms6Xmod

cc   meisei100mod.o demod_mod.o bch_ecc_mod.o  -lm -o meisei100mod

cc   m10mod.o demod_mod.o  -lm -o m10mod

cc   m20mod.o demod_mod.o  -lm -o m20mod

cc   imet54mod.o demod_mod.o  -lm -o imet54mod

cc   mp3h1mod.o demod_mod.o  -lm -o mp3h1mod

cc   meteosis_mod.o demod_mod.o  -lm -o meteosis_mod

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/demod/mod'

make -C imet clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/imet'

rm -f imet1rs_dft imet4iq imet1rs_dft.o imet4iq.o

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\" -Ofast   -c -o ime                                                                                                                                                             t1rs_dft.o imet1rs_dft.c

cc   imet1rs_dft.o  -lm -o imet1rs_dft

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\" -Ofast   -c -o ime                                                                                                                                                             t4iq.o imet4iq.c

cc   imet4iq.o  -lm -o imet4iq

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/imet'

make -C mk2a clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/mk2a'

rm -f mk2a_lms1680 mk2a1680mod mk2a_lms1680.o mk2a1680mod.o

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o mk2a_lms16                                                                                                                                                             80.o mk2a_lms1680.c

cc   mk2a_lms1680.o  -lm -o mk2a_lms1680

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\" -Ofast   -c -o mk2                                                                                                                                                             a1680mod.o mk2a1680mod.c

cc   mk2a1680mod.o  -lm -o mk2a1680mod

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/mk2a'

make -C scan clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/scan'

rm -f dft_detect dft_detect.o

cc -O3 -w -Wno-unused-variable -DNOC34C50 -Ofast   -c -o dft_detect.o dft_detect.c

cc   dft_detect.o  -lm -o dft_detect

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/scan'

make -C utils clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/utils'

rm -f fsk_demod fsk_demod.o fsk.o modem_stats.o kiss_fftr.o kiss_fft.o

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o fsk_demod.o fsk_demod.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o fsk.o fsk.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o modem_stats.o modem_stats.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o kiss_fftr.o kiss_fftr.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o kiss_fft.o kiss_fft.c

cc   fsk_demod.o fsk.o modem_stats.o kiss_fftr.o kiss_fft.o  -lm -o fsk_demod

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/utils'

make: Nothing to be done for 'all'.

make: Leaving directory '/home/pi/radiosonde_auto_rx'

Copying files into auto_rx directory.

Done!

 

Cet e-mail a été vérifié par le logiciel antivirus d'AVG.
www.avg.com

F1TKE, 2022-11-11:
Me again

 

Deleted the demod repertory and reinstall it from source

Then build and launch decoding with no version error

 

***@***.***:~/radiosonde_auto_rx/auto_rx $ ./build.sh

Building for radiosonde_auto_rx version: "1.6.0-beta14"

make: Entering directory '/home/pi/radiosonde_auto_rx'

make -C demod/mod clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/demod/mod'

rm -f rs41mod dfm09mod rs92mod lms6mod lms6Xmod meisei100mod m10mod m20mod imet54mod mp3h1mod meteosis_mod rs41mod.c.o dfm09mod.c.o rs92mod.c.o lms6mod.c.o lms6Xmod.c.o meisei100mod.c.o m10mod.c.o m20mod.c.o imet54mod.c.o mp3h1mod.c.o meteosis_mod.c.o bch_ecc_mod.c.o demod_mod.c.o

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o rs41mod.o rs41mod.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\" -Ofast   -c -o demod_mod.o demod_mod.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o bch_ecc_mod.o bch_ecc_mod.c

cc   rs41mod.o demod_mod.o bch_ecc_mod.o  -lm -o rs41mod

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o dfm09mod.o dfm09mod.c

cc   dfm09mod.o demod_mod.o  -lm -o dfm09mod

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o rs92mod.o rs92mod.c

cc   rs92mod.o demod_mod.o bch_ecc_mod.o  -lm -o rs92mod

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o lms6mod.o lms6mod.c

cc   lms6mod.o demod_mod.o bch_ecc_mod.o  -lm -o lms6mod

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o lms6Xmod.o lms6Xmod.c

cc   lms6Xmod.o demod_mod.o bch_ecc_mod.o  -lm -o lms6Xmod

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o meisei100mod.o meisei100mod.c

cc   meisei100mod.o demod_mod.o bch_ecc_mod.o  -lm -o meisei100mod

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o m10mod.o m10mod.c

cc   m10mod.o demod_mod.o  -lm -o m10mod

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o m20mod.o m20mod.c

cc   m20mod.o demod_mod.o  -lm -o m20mod

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o imet54mod.o imet54mod.c

cc   imet54mod.o demod_mod.o  -lm -o imet54mod

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o mp3h1mod.o mp3h1mod.c

cc   mp3h1mod.o demod_mod.o  -lm -o mp3h1mod

cc   meteosis_mod.o demod_mod.o  -lm -o meteosis_mod

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/demod/mod'

make -C imet clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/imet'

rm -f imet1rs_dft imet4iq imet1rs_dft.o imet4iq.o

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\" -Ofast   -c -o imet1rs_dft.o imet1rs_dft.c

cc   imet1rs_dft.o  -lm -o imet1rs_dft

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\" -Ofast   -c -o imet4iq.o imet4iq.c

cc   imet4iq.o  -lm -o imet4iq

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/imet'

make -C mk2a clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/mk2a'

rm -f mk2a_lms1680 mk2a1680mod mk2a_lms1680.o mk2a1680mod.o

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o mk2a_lms1680.o mk2a_lms1680.c

cc   mk2a_lms1680.o  -lm -o mk2a_lms1680

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\" -Ofast   -c -o mk2a1680mod.o mk2a1680mod.c

cc   mk2a1680mod.o  -lm -o mk2a1680mod

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/mk2a'

make -C scan clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/scan'

rm -f dft_detect dft_detect.o

cc -O3 -w -Wno-unused-variable -DNOC34C50 -Ofast   -c -o dft_detect.o dft_detect.c

cc   dft_detect.o  -lm -o dft_detect

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/scan'

make -C utils clean all

make[1]: Entering directory '/home/pi/radiosonde_auto_rx/utils'

rm -f fsk_demod fsk_demod.o fsk.o modem_stats.o kiss_fftr.o kiss_fft.o

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o fsk_demod.o fsk_demod.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o fsk.o fsk.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o modem_stats.o modem_stats.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o kiss_fftr.o kiss_fftr.c

cc -O3 -w -Wno-unused-variable -DVER_JSN_STR=\"1.6.0-beta14\"   -c -o kiss_fft.o kiss_fft.c

cc   fsk_demod.o fsk.o modem_stats.o kiss_fftr.o kiss_fft.o  -lm -o fsk_demod

make[1]: Leaving directory '/home/pi/radiosonde_auto_rx/utils'

make: Nothing to be done for 'all'.

make: Leaving directory '/home/pi/radiosonde_auto_rx'

Copying files into auto_rx directory.

mv: replace './dft_detect', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './fsk_demod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './imet4iq', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './mk2a1680mod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './rs41mod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './dfm09mod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './m10mod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './m20mod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './rs92mod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './lms6Xmod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './meisei100mod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './imet54mod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './mp3h1mod', overriding mode 0755 (rwxr-xr-x)? y

mv: replace './meteosis_mod', overriding mode 0755 (rwxr-xr-x)? y

Done!

darksidelemm, 2022-11-11:
Yep, the makefile is broken, sorry... working on it.

darksidelemm, 2022-11-11:
I've reverted the change that broke the builds. The docker image will take a while to be built, but you should be able to pull the changes and re-build now.

zanco, 2022-11-11:
Thanks for the update, now runing Beta 15, logging options both enabled, 170 TB available SSD space ;-)

F1TKE, 2022-11-11:
Upgrade to beta 15 and build ok

All work fine

 

2022-11-11 15:18:57 UTC – INFO
Sondehub Uploader - Uploaded 11 telemetry packets to Sondehub in 0.8 seconds.

2022-11-11 15:18:49 UTC – INFO
E-Mail - E-mail notification sent.

2022-11-11 15:18:46 UTC – INFO
Telemetry Logger - Using existing log file: /home/pi/radiosonde_auto_rx/auto_rx/log/20220610-053248_T2950788_RS41_402700_sonde.log

2022-11-11 15:18:42 UTC – INFO
Decoder (RTLSDR 1) RS41 402.699 - Starting decoder subprocess.

2022-11-11 15:18:42 UTC – INFO
Decoder (RTLSDR 1) RS41 402.699 - Using fsk_demod decoder chain.

2022-11-11 15:18:39 UTC – INFO
Task Manager - SDR #1 has been allocated to Decoder (RS41, 402.699 MHz).

darksidelemm, 2022-11-20:
@zanco can you please enable the new logging options and keep logging until this issue occurs again? I'd like to see the details logs for the detection steps when this is happening.

darksidelemm, 2022-12-22:
Any updates on this @zanco? I know there were some issues due to out-of-bounds frequencies which I think we have fixed?

zanco, 2022-12-22:
Did not happen again for the last month so I think we are ok

closed 2023-03-10 by darksidelemm