projecthorus/radiosonde_auto_rx #685: Scanner stops detecting new radiosondes

Eshco93, 2022-10-16:
Hi,

I think I have a problem with the scanner.
Once my system is up for a few hours, the scanner just seems to stop recognizing new radiosondes. A strong peak in the spectrum is clearly visible on the frequency where the radiosonde is transmitting. But the scanner just doesn't seem to recognize the radiosonde and doesn't start the decoder process.

This screenshot shows an exemplary scan result. The radiosonde is transmitting on 405.3 MHz. But it's not recognized by the scanner.

[image]

I can temporarily solve the issue by just restarting the system. Once I've done that, the scanner recognizes the sonde immediately and starts decoding. But like 12 hours later - when a once again new sondes are launched - the problem occurs again.

Does anyone know what might be the reason for this issue and how to solve it?

Greetings,
Simon

darksidelemm, 2022-10-16:
Can you provide any log information from when it tries to do the detection step and fails?
I'm currently away for work and dont have much time to look at this at the moment.

Eshco93, 2022-10-16:
Here is what the log reads while it's scanning.
Is that what you wanted to see?

`<html><body>

Scanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405.  405.7]2022-10-16T12:00:39Z
--
INFOScanner #00000004 - Running frequency scan.2022-10-16T12:00:18Z
INFOScanner  #00000004 - Detected peaks on 15 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405.  405.7 402. ]2022-10-16T11:57:37Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:57:17Z
INFOScanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.  405.3 405.7]2022-10-16T11:54:46Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:54:25Z
INFOScanner  #00000004 - Detected peaks on 15 frequencies (MHz): [402.5  402.7  403.    403.9  404.1  404.5  404.9  405.1  405.3  405.7  405.9  405.   405.7  405.3  402.63]2022-10-16T11:51:44Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:51:23Z
INFOScanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.  405.3 405.7]2022-10-16T11:48:52Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:48:32Z
INFOScanner  #00000004 - Detected peaks on 13 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.  405.3]2022-10-16T11:46:11Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:45:50Z
INFOScanner  #00000004 - Detected peaks on 13 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.  405.3]2022-10-16T11:43:29Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:43:09Z
INFOScanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405.  405.7]2022-10-16T11:40:38Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:40:17Z
INFOScanner  #00000004 - Detected peaks on 15 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405.  405.7 402. ]2022-10-16T11:37:38Z
INFOAPRS-IS  - Uploaded to APRS-IS:  SIMON2-14>APRARX,SONDEGATE,TCPIP,qAR,SIMON2-14:!5314.25N/00744.65E`radiosonde_auto_rx  SondeGate v1.5.10 !whY!2022-10-16T11:37:26Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:37:17Z
INFOScanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405.  405.7]2022-10-16T11:34:48Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:34:27Z
INFOScanner  #00000004 - Detected peaks on 13 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405. ]2022-10-16T11:32:08Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:31:47Z
INFOScanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405.  405.7]2022-10-16T11:29:18Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:28:57Z
INFOScanner  #00000004 - Detected peaks on 13 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405. ]2022-10-16T11:26:38Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:26:18Z
INFOScanner  #00000004 - Detected peaks on 16 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 402.  404.4 405.3 405.8 405.6]2022-10-16T11:23:27Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:23:06Z
INFOScanner  #00000004 - Detected peaks on 15 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 402.  405.3 404.4 405.8]2022-10-16T11:20:25Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:20:04Z
INFOScanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 402.  405.3 405.8]2022-10-16T11:17:35Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:17:15Z
INFOFlask - New Web Client connected!2022-10-16T11:14:26Z
INFOScanner  #00000004 - Detected peaks on 16 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 402.  404.4 405.3 405.6 405.8]2022-10-16T11:14:24Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:14:03Z
INFOScanner  #00000004 - Detected peaks on 16 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 402.  405.3 404.4 405.8 405.6]2022-10-16T11:11:12Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:10:51Z
INFOScanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405.  402. ]2022-10-16T11:08:22Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:08:02Z
INFOSondehub Uploader - Uploaded station information to Sondehub.2022-10-16T11:07:32Z
INFOAPRS-IS  - Uploaded to APRS-IS:  SIMON2-14>APRARX,SONDEGATE,TCPIP,qAR,SIMON2-14:!5314.25N/00744.65E`radiosonde_auto_rx  SondeGate v1.5.10 !whY!2022-10-16T11:07:26Z
INFOScanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405.  402. ]2022-10-16T11:05:31Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T11:05:10Z
[... cut ...]
INFOScanner  #00000004 - Detected peaks on 14 frequencies (MHz): [402.5 402.7 403.   403.9 404.1 404.5 404.9 405.1 405.3 405.7 405.9 405.3  405.  402. ]2022-10-16T10:56:46Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T10:56:25Z
INFOScanner  #00000004 - Detected peaks on 15 frequencies (MHz): [402.5  402.7  403.    403.9  404.1  404.5  404.9  405.1  405.3  405.7  405.9  405.3  404.75 405.   402.  ]2022-10-16T10:53:44Z
INFOScanner #00000004 - Running frequency scan.2022-10-16T10:53:24Z
INFOFlask - New Web Client connected!2022-10-16T10:52:59Z

</body>
</html>`

darksidelemm, 2022-10-16:
Unfortunately this doesn't tell me much :-(

Issues like this are rather difficult to debug. One of the best ways is to enable full debug logging, but this requires adding a '-v' to the command line when auto_rx is started. This is difficult to do from within a docker container for example... 

I may need to add some configuration file options to enable this kind of logging, but its not something im likely to be able to do for a while.

Eshco93, 2022-10-16:
Yeah, I already thought to myself that this log won't help much. That's why I asked if that's what you wanted to see.
Adding -v to the command line is not a big problem for me. I installed radiosonde_auto_rx natively. So that's done quite easily.
I'll add the -v and get back to you once I have a meaningful log file to show you.

darksidelemm, 2022-10-16:
Thanks for that! What you should see is some log messages each time the detection step occurs (which will occur for each spectrum peak found). Interested to know what is reported when there is an actual signal there (worth confirming this with a separate receiver as well, in case there's some other RF issue going on).

oe3osb, 2022-10-17:
Hi, do you use 4 SDR's ?
Try it with 3 ... ;)

Erich

Eshco93, 2022-10-17:
I'm using only 3 SDRs.

Eshco93, 2022-10-18:
Okay, now I might have a meaningful logfile.

I restarted the system yesterday 7:18 pm. Everything was fine throughout the night and early morning. It looks like the problems began 9:44 am when the scanner on SDR #3 suddenly detected peaks on 18 frequencies (I increased the peak detection limit from 10 to 18). On all the scans before that there were like between 2 to 5 peaks detected. And it also seems like the spectrum plot in general started to look a bit odd. This is what it looks like right now:

[image]

I think that's kind of how it looked like since 9:44 am, when the problems began. Due to work I didn't have time to check the sprectrum plot throughout the day.

I guess due to the 2 MHz bandwidth of those SDR sticks, the whole scan process for the 6 MHz radiosonde band is split into 3 individual scan processes?
Because it seems like something is wrong with the last process from 404 MHz to 406 MHz - since the power level there is down to like -50 to -60 dB. On the other hand it seems like the power level in the two other segments (400 MHz to 402 MHz and 402 MHz to 404 MHz) is raised by a few dB - being the reason why there are suddenly 18 detected peaks in these segments.

With the spectrum looking like this it's quite logical that the radiosondes of today (launched after 9:44 am) were not detected. Today I would have expected to receive radiosondes at 404.5 MHz, 405.3 MHz and 405.7 MHz. So they are all within the last segment of the radiosonde band that's down to -50 to -60 dB in the plot.

On the other hand, I also had the problem with radiosondes not being detected when the spectrum looked perfectly fine. Like in my initial post.

Anyway, here is the logfile. And I also got you the log_power_00000003.csv file.

[20221017-171823_system.log](https://github.com/projecthorus/radiosonde_auto_rx/files/9813860/20221017-171823_system.log)

[log_power_00000003.csv](https://github.com/projecthorus/radiosonde_auto_rx/files/9813861/log_power_00000003.csv)

darksidelemm, 2022-10-18:
The scan data is generated by rtl_power, which automatically steps the RTLSDR in frequency to cover the desired range.
The noise floor stepping makes me wonder if this is an issue related to AGC on the RTLSDRs. You could try setting the gain to a fixed value and see if that makes the scan plot behave.

Eshco93, 2022-10-18:
Yes, I can do so. But since I restartet the service just this evening, I'll need to wait until this happens again. I'll get back to you once I've tested that.

How can I adjust the gain setting while the service is running? Is the gain setting from the station.cfg file read every time a scan process is started? So I can just edit it there and wait for the next scan process in order to check if that helped?

darksidelemm, 2022-10-18:
Unfortunately the gain values are only read one, on startup, sorry.

Eshco93, 2022-10-19:
So basically what I have to do is set the gain to some fixed value, start auto_rx and let it run for some time in order to see if this occures again?
Okay, I'll get back to you once I have some new findings :-)

Eshco93, 2022-10-21:
Okay, so the issue is not related to AGC. I've set the gain to a certain value and after a couple of hours the issue occured again.

darksidelemm, 2022-10-21:
I'm still somewhat unsure whats going on here... if it's putting a peak on the signal, then it should go through to the detection step. What I'm interested in is the output of the detection step when this is a problem. There will be lines along the lines of 'Using detection command ...' which will occur in the log when it's running detection on the peaks that are found.

When you get into the state where it stops detecting anything, its the outputs of those lines i'm interested in. I'm unsure looking through your log where that point is.

darksidelemm, 2022-10-22:
You could also try running the detection command yourself and see if any errors come out
e.g.:
```
rtl_fm -p 0 -d 00000003 -M raw -F9 -s 48000 -f 400880000 2>/dev/null |./dft_detect -t 10 --iq --bw 20 --dc - 48000 16
```
It can also help to remove the `2>/dev/null` parts, as they may hide errors.

Eshco93, 2022-10-22:
It does go through the detection step after the frequency peaks are detected.
Here is one example of detected peaks from the log:

`2022-10-18 12:51:31,319 INFO:Scanner #00000003 - Detected peaks on 18 frequencies (MHz): [401.04 403.02 401.05 400.88 401.2  401.66 401.03 400.58 401.06 401.5
 400.59 400.73 400.87 400.68 401.14 400.61 403.01 401.36]`

After this it will go through all the detected peaks like this:

`2022-10-18 12:51:31,320 DEBUG:Scanner #00000003 - Using detection command: timeout 20s rtl_fm -p 0 -d 00000003 -M raw -F9 -s 48000 -f 401040000 2>/dev/null |./dft_detect -t 10 --iq --bw 20 --dc - 48000 16 2>/dev/null
2022-10-18 12:51:31,320 DEBUG:Scanner #00000003 - Attempting sonde detection on 401.040 MHz
2022-10-18 12:51:51,372 DEBUG:Scanner - dft_detect exited in 20.1 seconds with return code 1.`

This is probably just fine. But the issue is that when the spectrum the detection is running on looks like this:

[image]

Then it's not a big surprise that I don't get any radiosondes that transmit between 404 MHz and 406 MHz.
And all the other peaks between 400 MHz and 404 MHz are just nonsense. There were no radiosondes transmitting in this frequency range when the screenshot was taken. I think the peaks between 400 MHz and 404 MHz are related to the detection algorithm not being able properly work with such a weird spectrum. Which is totally fine. The question is: What's causing the spectrum to look like this? I have no idea. All I can say is that it doesn't seem to be an AGC issue, since I already tried fixed gain values and that didn't solve the problem.

I think after all we are looking at two issues here. The first one is this weird spectrum that I get, like in the screenshot above.
The second one is auto_rx not detecting radiosondes while the spectrum looks totaly fine and shows a peak for the sonde that I want to decode (Like in my initial post).

Eshco93, 2022-10-22:
Maybe the issue with the weird spectrum is somehow related to my SDR sticks. I'm using three of these [NEWGEN RTL2832U SDR Sticks with TXCO](https://de.aliexpress.com/item/32903103541.html?spm=a2g0o.productlist.0.0.5b41276e2B0crC&algo_pvid=d8e412d2-46f2-4840-badb-33019312c13d&algo_exp_id=d8e412d2-46f2-4840-badb-33019312c13d-0&pdp_ext_f=%7B%22sku_id%22%3A%2265887220740%22%7D&pdp_npi=2%40dis%21EUR%2126.34%2117.64%21%21%21%21%21%402100bb4a16664692735838336eb729%2165887220740%21sea&curPageLogUid=xzo6079uXNRN).

I also have one genuine RTL-SDR Blog V3 Stick that I could try.

Is there any way to tell auto_rx to only use one specific stick for the scanning? Or do I have to disconnect my other sticks to make sure that just the RTL-SDR Blog V3 Stick is used?

darksidelemm, 2022-10-22:
Currently no way to specify one stick for scanning unfortunately. auto_rx will just use whatever one is available. 

It might be worth testing the SDRs separately from auto_rx?

darksidelemm, 2022-12-22:
Any updates on this testing?

Eshco93, 2022-12-23:
I'm sorry that I haven't reported back for such a long time.
I was quite busy in the mean time and then I somehow forgot about this issue.
To be honest, I have since replaced the software of my radiosonde receiver station. I'm now using dxlAPRS. The main reason for this decision was that with dxlAPRS, I'm able to cover the whole 6 MHz radiosonde band with 3 SDR Sticks.
But if you are interested in finding the root of my issue(s), I can go back to auto_rx and further investigate the issues like I planned previously.
Otherwise, this matter is no longer relevant to me.

darksidelemm, 2022-12-24:
Worth noting that auto Rx 1.6.0 (testing) now supports connecting to a spy
server, so with one airspy R2 or mini, you can cover the entire radiosonde
band too :-)

closed 2022-12-24 by darksidelemm

Eshco93, 2022-12-24:
I'm aware of that, but thanks for the tip nonetheless.
Since I already own those SDR Sticks, I didn't want to spend additional money on buying an Airspy.