projecthorus/radiosonde_auto_rx #693: Radiosonde Auto-RX 1.6.0-beta13 graphical request

zanco, 2022-10-30:
Hi,
Can I (we) have different colours for the balloons on the map  which are actively received ?

[image]
With some stations doing multiple launches in 2 hours time slot it's sometimes difficult to locate the active flights on the auto_rx map.

Thanks,
Ben

darksidelemm, 2022-10-30:
How about having the green lines only appear if the telemetry is less than X seconds old (say, 10 seconds)?

zanco, 2022-10-31:
Sounds like a good solution !

darksidelemm, 2022-11-14:
I've implemented this in this PR: https://github.com/projecthorus/radiosonde_auto_rx/pull/702
Wait for the docker build to finish and it will be live on testing :-)

closed 2022-11-20 by darksidelemm