projecthorus/radiosonde_auto_rx #700: DFM radiosonde is ignored because 0 GNSS sats.

AstroTaka, 2022-11-13:
I post from Japan.
I use radiosonde_auto_rx on RaspberryPi 3(not b+) with RTL-SDR v3.

And I saw the following log.

```
2022-11-13 16:05:41,402 WARNING:Sonde DFM-605854 can only see 0 GNSS sats - discarding position as bad.
2022-11-13 16:05:43,372 WARNING:Sonde DFM-605854 can only see 0 GNSS sats - discarding position as bad.

```
I found a location where the log is exported.
It is a line 497 of auto_rx.py.

```
    if "sats" in telemetry:
        if telemetry["sats"] < 4:
            logging.warning(
                "Sonde %s can only see %d GNSS sats - discarding position as bad."
                % (telemetry["id"], telemetry["sats"])
            )
            return False
```
I recorded a audio file of the target frequency.
And I made a json data by dfm09mod by the following command.

`./dfm09mod --json --v --ecc -i xxxx.wav
`

And I found that all 'sats' is zero.
But lat and lon seems to be correct...

```
[159] 2022-11-11 03:14:38.0 (0,0,0)   lat: 35.51201 (0)   lon: 140.13434 (0)   alt: 4688.9 (0)   vH:  9.49  D: 119.8  vV:  4.73  (IDx6:603953:DFM06) 
{ "type": "DFM", "frame": 1352171678, "id": "DFM-603953", "datetime": "2022-11-11T03:14:38.000Z", "lat": 35.51201, "lon": 140.13434, "alt": 4688.88000, "vel_h": 9.49000, "heading": 119.79000, "vel_v": 4.73000, "sats": 0, "temp": 146.9, "subtype": "0x6:DFM06" }
[160] 2022-11-11 03:14:39.0 (0,0,0)   lat: 35.51198 (0)   lon: 140.13445 (0)   alt: 4693.1 (0)   vH: 11.97  D: 108.5  vV:  4.14 
{ "type": "DFM", "frame": 1352171679, "id": "DFM-603953", "datetime": "2022-11-11T03:14:39.000Z", "lat": 35.51198, "lon": 140.13445, "alt": 4693.14000, "vel_h": 11.97000, "heading": 108.51000, "vel_v": 4.14000, "sats": 0, "subtype": "0x6:DFM06" }

```
Would you support the above DFM radiosonde?

Please check the following wav file
[https://drive.google.com/file/d/1SNvZ30HqHMaioXcMUDwNyRFmP-ibZzNR/view?usp=sharing](https://drive.google.com/file/d/1SNvZ30HqHMaioXcMUDwNyRFmP-ibZzNR/view?usp=sharing)

darksidelemm, 2022-11-13:
Unsure what's going on here. We should support DFM06's just fine, unless this is some very old DFM06 with possibly different telemetry? @rs1729 might know.

I won't be allowing reporting of positions with sats reported as 0 however... This is far too dangerous and will result in a mess on the map.

rs1729, 2022-11-13:
Don't know if I have seen this before. It looks like a DFM-06, however it has 8 instead of 6 channels, the serial number is actually in channel 8 (not 6 as the decoder says). Could be an additonal pressure sensor. Maybe @bazjo knows if there have been DFM-06 with pressure sensor. Looks like it.
That's why the temperature is calculated based on the wrong values. If I choose the PTU channels that the DFM-09P uses, i.e. 1, 5, 6 (instead of 0,3,4) it is more in line with the [weather data](http://weather.uwyo.edu/cgi-bin/sounding?region=seasia&TYPE=TEXT%3ALIST&YEAR=2022&MONTH=11&FROM=1100&TO=1100&STNM=47646):
```
[ 16] 2022-11-11 03:12:11.0 (0,0,0)   lat: 35.51695 (0)   lon: 140.12116 (0)   alt: 4053.6 (0)   vH:  6.84  D: 104.6  vV:  4.93   T=-6.2C 
[ 18] 2022-11-11 03:12:14.0 (0,0,0)   lat: 35.51684 (0)   lon: 140.12141 (0)   alt: 4066.1 (0)   vH:  8.97  D: 107.1  vV:  4.63   T=-6.3C  (IDx8:603953:DFM06) 
[ 20] 2022-11-11 03:12:16.0 (0,0,0)   lat: 35.51676 (0)   lon: 140.12158 (0)   alt: 4075.1 (0)   vH:  6.75  D: 133.4  vV:  3.91   T=-6.3C 
[ 22] 2022-11-11 03:12:18.0 (0,0,0)   lat: 35.51671 (0)   lon: 140.12176 (0)   alt: 4083.4 (0)   vH: 10.35  D: 120.6  vV:  4.67   T=-6.3C  (IDx8:603953:DFM06) 
[ 23] 2022-11-11 03:12:19.0 (0,0,0)   lat: 35.51666 (0)   lon: 140.12183 (0)   alt: 4088.1 (0)   vH:  7.44  D: 136.2  vV:  4.55   T=-6.4C 
```
```
[144] 2022-11-11 03:14:23.0 (0,0,0)   lat: 35.51253 (0)   lon: 140.13288 (0)   alt: 4624.7 (0)   vH:  8.21  D: 113.9  vV:  4.59   T=-10.9C  (IDx8:603953:DFM06) 
[146] 2022-11-11 03:14:25.0 (0,0,0)   lat: 35.51246 (0)   lon: 140.13307 (0)   alt: 4633.4 (0)   vH:  9.52  D: 115.6  vV:  4.46   T=-11.0C 
[147] 2022-11-11 03:14:26.0 (0,0,0)   lat: 35.51242 (0)   lon: 140.13316 (0)   alt: 4637.8 (0)   vH:  8.43  D: 115.5  vV:  4.67   T=-11.0C 
[148] 2022-11-11 03:14:27.0 (0,0,0)   lat: 35.51239 (0)   lon: 140.13326 (0)   alt: 4641.9 (0)   vH: 11.52  D: 111.8  vV:  3.94   T=-11.0C  (IDx8:603953:DFM06) 
[149] 2022-11-11 03:14:28.0 (0,0,0)   lat: 35.51235 (0)   lon: 140.13336 (0)   alt: 4646.1 (0)   vH:  9.05  D: 116.2  vV:  4.50   T=-11.0C 
```

Not sure if the DFM-06 reported the number of sats, in a recording of 2016 there is also `sats: 0` but you still have the list of satellites. I believe @einergehtnochrein suggested to count the number of satellites since this is more reliable.
Didn't check all the GPS-Sats, but deltaMSL-value (-geoid_height) seems reasonable, well, 5m off compared to [GeoidEval](https://geographiclib.sourceforge.io/cgi-bin/GeoidEval?input=35.51198+140.13445&option=Submit).
```
[157] 2022-11-11 03:14:36.0 (0,0,0)   lat: 35.51208 (0)   lon: 140.13414 (0)   alt: 4680.5 (0)   vH: 11.93  D:  99.5  vV:  3.78   T=-11.3C 
    dMSL: -39.52  sats: 0  ( 02 05 10 12 13 14 15 18 23 24 25 32 )
[159] 2022-11-11 03:14:38.0 (0,0,0)   lat: 35.51201 (0)   lon: 140.13434 (0)   alt: 4688.9 (0)   vH:  9.49  D: 119.8  vV:  4.73   T=-11.3C  (IDx8:603953:DFM06) 
    dMSL: -39.51  sats: 0  ( 02 05 10 12 13 14 15 18 23 24 25 32 )
[160] 2022-11-11 03:14:39.0 (0,0,0)   lat: 35.51198 (0)   lon: 140.13445 (0)   alt: 4693.1 (0)   vH: 11.97  D: 108.5  vV:  4.14   T=-11.3C 
    dMSL: -39.51  sats: 0  ( 02 05 10 12 13 14 15 18 23 24 25 32 )
```
The DFM-ID and temperature seems to be a minor modification, but I have to test if it breaks anything for DFM-09/17.
@darksidelemm with the number of sats, I don't know, JSON does not report the sat-list. Maybe the decoder can count the number of sats in the PRN list, if the NumSat value is 0. Perhaps only for DFM-06, DFM-09 should have NumSat. I believe it is the number of sats used in the solution, so it could be, that there are more PRN in the list than used for the GPS position.

einergehtnochrein, 2022-11-13:
The age-old problem of DFM06 vs, DFM09old vs DFM09new vs....  :-)  My guess here would be that it's an DFM09 based on 8051 hardware instead of the STM32 in later models. Could still be a DFM06 though, who knows.

I made a note somewhere that old DFM's wouldn't send the number of satellites used in the position solution, but I wasn't specific about what 'old' really means. At least I know for sure that only later models of DFM began sending this number.

I agree with @rs1729 that the PRN list might be an alternative, but I guess it will indeed give you the number of visible satellites instead of the number of satellites selected for the navigation solution. However, I guess this hardly matters for our purpose.
You get an SNR value from the sonde along with each satellite PRN, and you might apply some threshold magic, but why would we want to do that?
The only thing I suggest is to simplify satellite counting: The 32-bit field of the second GPS slot is a bitmask for the PRN1 ... PRN32. Just determine the weight of this word to find how many satellites there are.

rs1729, 2022-11-13:
Oh, of course, I forgot about the old DFM09 that transmits like an DFM06.
Maybe the pressure sensor was only available for DFM09-type hardware layout, and for the very early DFM09 that meant it transmitted the data in DMF06 style.

For me the sats are not really important, however auto_rx has a plausibility check built in, I guess.
Since `"Sonde_ %s can only see %d GNSS sats"` could also be the number of SV in sight, I could do the following, @darksidelemm 
If sats==0 and DFM-type is DFM06(P), I count the number of SV in the SV/PRN-list.
Then either (a) I take that numSV for "sats:" in JSON or (b) put it in a new field "numSV:". Would prefer (a) since name "sats" is not so specific, and if both fields exist, it would have to be added to other radiosondes. The sats==0 problem shouldn't be very common anyway.

Result would be something like:
```
[166] 2022-11-11 03:14:45.0   lat: 35.51177  lon: 140.13508  alt: 4719.5 (0)   vH: 13.27  D: 102.3  vV:  3.83   T=-11.5C 
    dMSL: -39.51  sats: 0  numSV: 12 ( 02 05 10 12 13 14 15 18 23 24 25 32 )
{ "type": "DFM", "frame": 1352171685, "id": "DFM-603953", "datetime": "2022-11-11T03:14:45.000Z", "lat": 35.51177, "lon": 140.13508, "alt": 4719.49000, "vel_h": 13.27000, "heading": 102.27000, "vel_v": 3.83000, "sats": 12, "temp": -11.5, "subtype": "0x8:DFM06P", "ref_datetime": "UTC", "ref_position": "GPS", "diff_GPS_MSL": 39.51 }
```

bazjo, 2022-11-13:
this might be very well DFM-06 radisondes, but as this model is no longer produces and in the process of being phased out, I would not put much afford into this. I will check the sensor configuration channel and update accordingly. Not sure about the GPS No. of Sats though. Theoretically, there should be no difference of format used across the different models, but I have not looked into DFM-06 in any detail in particular.

One thing one the old DFM-09 with dual 8051: those have not been manufactured in years and any remaining ones which might occur would have been in storage by the customer for around 10 yrs.

rs1729, 2022-11-13:
If SN is in 8 instead of 6, and then the same channels that DFM-09P and DFM-17P use seem to work also for "DFM-06P".
(except that the old 8051 MCU used float20, STM32 uses float24)
Debug output with channels 0-7:
[20221111_03Z_dbg.txt.gz](https://github.com/projecthorus/radiosonde_auto_rx/files/9997572/20221111_03Z_dbg.txt.gz)

rs1729, 2022-11-13:
Do PS-15 and DFM-17 have GNSS chips that receive both GPS and GLONASS?
Number of sats is usually 16-18.

bazjo, 2022-11-13:
a u-blox 8th generation GNSS solution is used, which is configurable to various GNSS system by the end user (however it is not as flexible as the newer generation, especially with Beidou and Galileo) GPS+Glonass is what most customers will most likely use

rs1729, 2022-11-13:
modified [dfm09mod.c](https://github.com/rs1729/RS/blob/test/demod/mod/dfm09mod.c) for testing

darksidelemm, 2022-11-14:
Have updated the testing branch with this decoder (1.6.0-beta17). Have tested locally with the DFM06 sample provided above, and local DFM09 and DFM17 models. All seem OK.

AstroTaka, 2022-11-14:
Great!

I replaced the test dfm09mod.c.
And I tried the DFM06 sonde again.
It works fine.
I checked a value of 'sats' at that time.
Almost is 12. another value is 10 or 11.

Thank you very much!!

By the way, I found one problem when I check the DFM06 decode.
Probably when a signal condition is not good,  dfm09mod export incorrect serial.
The serial is "DFM-     0".

When DFM-602963 decoding, suddenly "DFM-    0" ("DFM-" + 5 space char (0x20)+ "0") decoded only once.
And then, continued to decode DFM-602963 until signal lost.

`
2022-11-14 18:34:52,578 DEBUG:Telemetry Logger - Wrote line: 2022-11-14T09:34:51.000Z,DFM-     0,1352453691,35.49118,140.77561,10666.5,4.5,73.1,94.2,-102.8,-1.0,-1.0,DFM06P,404.409,11.8,1562,11,-1.0,-1,-1
`

Please check a following log file.
But this is not full log because terminal buffer is full.
https://drive.google.com/file/d/18usKveC4TjPUEllfg-ZlLJ92EaDq-1vM/view?usp=share_link

Please check a following screen picture when "DFM-    0" decoded also.
https://drive.google.com/file/d/1pNm7Ial3g5xrD2-zShwml-5X8oDVWiOf/view?usp=share_link

darksidelemm, 2022-11-14:
Uh oh, that shouldn't have made it through my callsign checks, better look into that... Also I guess the decoder shouldn't be producing it either!

rs1729, 2022-11-14:
Thanks, I guess that happened when the SN channel number was decoded incorrectly (i.e. not to the original code word). Unfortunately the Hamming code is not reliable. Before the DFM06-type radiosondes got sn_ch=6 assigned, now it is the decoded channel number, but if it is not 6 or 8, it took the hex-SN which would be empty.
Should be fixed now.
But the DFM problem remains, if a transmission error is not detected, such a frame might show wrong data.

AstroTaka, 2022-11-15:
I made a patch for checking a serial ID.

Basic thinking of this patch, dfm09mod handle only one serial ID.
So if dfm09mod found same serial ID several times, it must be correct.

At the beginning phase, dfm09mod cannot judge whether a serial ID is correct.
So it is possible that dfm09mod output incorrect serial ID at that time.
But after that, dfm09mod can judge which serial ID is correct.
And it can reject incorrect serial ID.

So this patch increase the reliability of a decode data I believe.

```diff
--- dfm09mod_orig.c     2022-11-13 18:50:56.051671708 +0900
+++ dfm09mod.c  2022-11-15 11:07:12.478608653 +0900
@@ -771,6 +771,9 @@ static int conf_out(gpx_t *gpx, ui8_t *c
     return ret;
 }

+static char last_json_sonde_id[] = "DFM-xxxxxxxx\0\0";
+static int last_json_sonde_id_counter = 0;
+
 static void print_gpx(gpx_t *gpx) {
     int i, j;
     int contgps = 0;
@@ -918,6 +921,28 @@ static void print_gpx(gpx_t *gpx) {
                 default : sprintf(json_sonde_id, "DFM-%6u", gpx->SN);
             }

+            if(strcmp(json_sonde_id,"DFM-xxxxxxxx")!=0){
+             // json_sonde_id is not "DFM-xxxxxxxx"
+             if(strcmp(json_sonde_id,last_json_sonde_id)!=0){
+               // current json_sonde_id is not last json_sonde_id except "DFM-xxxxxxxx"
+               if(last_json_sonde_id_counter<10){
+                 // last json_sonde_id found less than 10 times
+                 // update last json_sonde_id and last json_sonde_id counter
+                 // and accept current json_sonde_id
+                 strcpy(last_json_sonde_id,json_sonde_id);
+                 last_json_sonde_id_counter=0;
+               } else {
+                 // Ignore this data
+                 // different last json_sonde_id found more than 10 times
+                 for (i = 0; i < 9; i++) gpx->pck[i].ec = -1; // copy this from the end of this function.
+                 return;
+               }
+             } else {
+               // increase last json_sonde_id counter
+               last_json_sonde_id_counter++;
+             }
+           }
+
             // JSON frame counter: gpx->sec_gps , seconds since GPS (ignoring leap seconds, DFM=UTC)

             // Print JSON blob     // valid sonde_ID?
```

darksidelemm, 2022-11-15:
I've just updated the testing branch with the absolute latest dfm09mod updates from @rs1729. Please test those.

AstroTaka, 2022-11-15:
@rs1729 
OK, No problem, Thanks!!

AstroTaka, 2022-11-17:
I cannot see this problem at beta 19.
So I close this issue.

closed 2022-11-17 by darksidelemm