projecthorus/radiosonde_auto_rx #704: scan.py of testing branch has error about NameError: name 'device_idx' is not defined

AstroTaka, 2022-11-16:
I am trying to check testing branch.
When I set "save_detection_audio=True" in station.cfg,  I saw the following error

```
Traceback (most recent call last):
  File "/home/AstroTaka/radiosonde_auto_rx_testing/auto_rx/autorx/scan.py", line 841, in scan_loop
    _results = self.sonde_search()
  File "/home/AstroTaka/radiosonde_auto_rx_testing/auto_rx/autorx/scan.py", line 1072, in sonde_search
    (detected, offset_est) = detect_sonde(
  File "/home/AstroTaka/radiosonde_auto_rx_testing/auto_rx/autorx/scan.py", line 348, in detect_sonde
    rx_test_command += "tee detect_%s.raw | " % str(device_idx)
NameError: name 'device_idx' is not defined
```

The code of line 348 of scan.py is the following.
```
        # Saving of Debug audio, if enabled,
        if save_detection_audio:
            rx_test_command += "tee detect_%s.raw | " % str(device_idx)
```
Probably, device_idx should be rtl_device_idx.
The save_decode_iq in station.cfg cannot control this part...

The code of line 402 has the same problem.
```
        # Saving of Debug audio, if enabled,
        if save_detection_audio:
            rx_test_command += "tee detect_%s.wav | " % str(device_idx)
```

darksidelemm, 2022-11-16:
Thanks! That's definitely a bug, will fix...

darksidelemm, 2022-11-16:
fixed in PR https://github.com/projecthorus/radiosonde_auto_rx/pull/705

I haven't bumped the version.

closed 2022-11-16 by darksidelemm