projecthorus/radiosonde_auto_rx #706: rtl_power command does not use '-d' option in sdr_wrappers.py

AstroTaka, 2022-11-16:
I am checking testing branch.
I try to check a difference between IQ and FM of MEISEI sonde.

So I connect two RTL-SDR dongle to one Ubuntu PC and I made two copy of testing branch.
I changed the both station.cfg

test1 station.cfg -> FM
```
sdr_quantity=1
device_idx = 00000003
web_port = 5001
meisei_experimental = False
```

test2 station.cfg -> IQ
```
sdr_quantity=1
device_idx = 00000002
web_port = 5002
meisei_experimental = True
```

When I start the both auto_rx.py, I got the following log.
```
2022-11-16 18:03:59,939 DEBUG:lsusb parsing error: unexpected indentation: "           ** UNAVAILABLE **"
2022-11-16 18:03:59,941 DEBUG:lsusb parsing error: unexpected indentation: "           ** UNAVAILABLE **"
2022-11-16 18:04:01,813 INFO:Scanner (RTLSDR 00000002) - Starting Scanner Thread
2022-11-16 18:04:01,814 INFO:Scanner (RTLSDR 00000002) - Running frequency scan.
2022-11-16 18:04:01,814 DEBUG:Scanner (RTLSDR 00000002) - Running command: timeout -k 30 30 rtl_power -p 0 -f 403900000.0:404100000.0:800.0 -i 20 -1 -c 25% log_power_00000002.csv
2022-11-16 18:04:01,889 CRITICAL:Scanner (RTLSDR 00000002) - rtl_power call failed with return code 1.
2022-11-16 18:04:01,890 CRITICAL:Scanner (RTLSDR 00000002) - rtl_power reported error: Number of frequency hops: 1
Dongle bandwidth: 2666660Hz
Downsampling by: 10x
Cropping by: 25.00%
Total FFT bins: 512
Logged FFT bins: 384
FFT bin size: 520.83Hz
Buffer size: 16384 bytes (3.07ms)
Reporting every 20 seconds
Found 2 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000003
  1:  Realtek, RTL2838UHIDIR, SN: 00000002

Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.

```

According to the log, rtl_power does not use '-d' option.
So rtl_power always to use device #0.
And "Failed to open rtlsdr device #0." occur I think.

And I found the root cause of not use '-d' in line 481 of sdr_wrappers.py.
```
        _rtl_power_cmd = (
            f"{_timeout_cmd} {rtl_power_path} "
            f"{'-T ' if bias else ''}"
            f"-p {int(ppm)} "
            #f"-d {str(rtl_device_idx)} "
            f"{_gain}"
            f"-f {frequency_start}:{frequency_stop}:{step} "
            f"-i {integration_time} -1 -c 25% "
            f"{_log_filename}"
        )
```
The '-d' option is comment out.
Is it correct? or, some debugging purpose?

At the multiple RTL-SDR use case, it is possible occur error as following scenario I think.

1. device # 0 and device # 1 is decoding.
2. device # 1 finish to decode.
3. start rtl_power without '-d' option.
4. rtl_power try to open device # 0 and error occur

darksidelemm, 2022-11-16:
That is very much a bug, and likely leftover from when i was initially testing the network server mode. 

Have fixed it in https://github.com/projecthorus/radiosonde_auto_rx/pull/705 (which i somehow didn't end up merging, but have now)

AstroTaka, 2022-11-16:
OK, Thanks!

closed 2022-11-17 by darksidelemm