projecthorus/radiosonde_auto_rx #712: Two RTL-SDR dongle decode one sonde at the same time

AstroTaka, 2022-11-17:
I changed to 3x RTL-SDR environment.

And two sonde exist.
1. DFM sonde: 404.4MHz
2. MEISEI sonde: 404.5MHz

1st RTL-dongle receive DFM sonde.
2nd RTL-dongle receive MEISEI sonde.
3rd RTL-dongle receive MEISEI sonde too....

I uploaded several log file.
https://drive.google.com/drive/folders/1LjLYOyapdXihRZXFzZM-3eCCwyUAim1s?usp=share_link

I found little bit strange log
```
2022-11-17 12:14:58,145 DEBUG:Scanner (RTLSDR 00000002) - Detected a Meisei Sonde! (Score: 0.76, Offset: 1695.8 Hz)
2022-11-17 12:14:58,146 DEBUG:Scanner (RTLSDR 00000002) - Using detection command: timeout 10 rtl_fm -M raw -F9 -p 0 -d 00000002 -s 48000 -f 405510000 - 2>/dev/null | tee detect_00000002.raw | ./dft_detect -t 5 --iq --bw 20 --dc - 48000 16 2>/dev/null
2022-11-17 12:14:58,147 DEBUG:Scanner (RTLSDR 00000002)- Attempting sonde detection on 405.510 MHz
2022-11-17 12:14:58,432 DEBUG:Delta time: 1
2022-11-17 12:14:58,661 DEBUG:Telemetry Logger - Wrote line: 2022-11-17T03:14:57.000Z,DFM-605803,1352690097,35.53581,140.16698,4100.5,3.3,16.9,99.7,-10.0,-1.0,-1.0,DFM06P,404.408,25.3,1367,11,-1.0,-1,-1
2022-11-17 12:14:59,337 DEBUG:Delta time: 1
2022-11-17 12:14:59,663 DEBUG:Telemetry Logger - Wrote line: 2022-11-17T03:14:58.000Z,DFM-605803,1352690098,35.53578,140.16717,4104.1,3.7,17.0,101.4,-10.0,-1.0,-1.0,DFM06P,404.408,25.3,1367,11,-1.0,-1,-1
2022-11-17 12:14:59,828 INFO:Task Manager - Detected new MEISEI sonde on 404.502 MHz!
2022-11-17 12:14:59,829 INFO:Task Manager - SDR #00000004 has been allocated to Decoder (MEISEI, 404.502 MHz).
```
I expect that dongle of scan and decode is same.
But 00000002 scan and 00000004 decode.

And then, 1st MEISEI decode start at 404.502MHz.
```
2022-11-17 12:15:01,981 INFO:Decoder (RTLSDR 00000004) MEISEI 404.502 - Using fsk_demod decoder chain.
2022-11-17 12:15:01,982 DEBUG:Decoder (RTLSDR 00000004) MEISEI 404.502 - Opening /home/AstroTaka/radiosonde_auto_rx_testing1/auto_rx/log/20221117-031459_MEISEI_404502000.raw to save decoder raw data.
2022-11-17 12:15:01,982 DEBUG:Decoder (RTLSDR 00000004) MEISEI 404.502 - Demodulator Command: rtl_fm -M raw -F9 -p 0 -d 00000004 -s 48000 -f 404502000 - 2>/dev/null |  tee decode_IQ_00000004_MEISEI_20221117_1215.bin |./fsk_demod --cs16 -s -b -15000 -u 15000 --stats=5 2 48000 2400 - -
2022-11-17 12:15:01,983 DEBUG:Decoder (RTLSDR 00000004) MEISEI 404.502 - Decoder Command: ./meisei100mod --softin --json --ptu --ecc 2>/dev/null
2022-11-17 12:15:02,007 INFO:Decoder (RTLSDR 00000004) MEISEI 404.502 - Starting decoder subprocess.
```

And finally, 2nd MEISEI decode start at 404.503MHz
```
2022-11-17 12:19:22,620 INFO:Scanner (RTLSDR 00000002) - Scanner Thread Closed.
2022-11-17 12:19:22,621 INFO:Task Manager - SDR #00000002 has been allocated to Decoder (-MEISEI, 404.503 MHz).
...
2022-11-17 12:19:24,599 INFO:Decoder (RTLSDR 00000002) MEISEI 404.503 - Using fsk_demod decoder chain.
2022-11-17 12:19:24,600 DEBUG:Decoder (RTLSDR 00000002) MEISEI 404.503 - Opening /home/AstroTaka/radiosonde_auto_rx_testing1/auto_rx/log/20221117-031922_-MEISEI_404503000.raw to save decoder raw data.
2022-11-17 12:19:24,600 DEBUG:Decoder (RTLSDR 00000002) MEISEI 404.503 - Demodulator Command: rtl_fm -M raw -F9 -p 0 -d 00000002 -s 48000 -f 404503000 - 2>/dev/null |  tee decode_IQ_00000002_MEISEI_20221117_1219.bin |./fsk_demod --cs16 -s -b -15000 -u 15000 --stats=5 2 48000 2400 - -
2022-11-17 12:19:24,601 DEBUG:Decoder (RTLSDR 00000002) MEISEI 404.503 - Decoder Command: ./meisei100mod --softin --json --ptu --ecc 2>/dev/null
2022-11-17 12:19:24,632 INFO:Decoder (RTLSDR 00000002) MEISEI 404.503 - Starting decoder subprocess.
```

P.S. my setting of decoder_spacing_limit is 15000.

darksidelemm, 2022-11-17:
You are uncovering lots of interesting bugs which I thought I had fixed :-)  This is good!

I think I can see why this is happening, and it's due to the checking of the type field - note that the second one has it detected as an 'inverted' MEISEI sonde. I'll have to modify this code to remove that bit before checking.

AstroTaka, 2022-11-17:
OK, I will check this after you release the next beta.

darksidelemm, 2022-11-17:
I haven't bumped the version, but the change is in. Give that a go.

AstroTaka, 2022-11-17:
After git pull, the problem still occur

https://drive.google.com/file/d/17dniugN_ZOTix2SwgGjOlZOp9q_f7bbn/view?usp=share_link

Would you let me know which file and line you modified.?
I would like to check whether my file updated.

darksidelemm, 2022-11-17:
You can see the changes here: https://github.com/projecthorus/radiosonde_auto_rx/pull/713/files

AstroTaka, 2022-11-17:
my auto_rx.py is updated.
I try to get a log and upload.

AstroTaka, 2022-11-19:
I do not see the problem.
So I close this issue.

closed 2022-11-19 by AstroTaka