projecthorus/radiosonde_auto_rx #714: Log file not uploading to sondehub.org database after being offline during parts of the chase because of terrain/cell coverage.

dalewegner, 2022-11-17:
Seems like positions only post to sondehub.org while connected to internet.  Noticed that log file has many more records near landing site, from during the chase, that do not post to internet because of loss of cell phone coverage.  Is the software designed to only post live records?  I did not see any utility to run that would post these records from the log file chase once cell phone service has been reacquired.

darksidelemm, 2022-11-17:
Yes, we only upload 'near-live' data. If we exceed upload timeouts then that data is discarded. 

We'd rather not end up in a situation where a station uploads a backlog of data from hours ago, this could result in some interesting behaviour on the tracker.

dalewegner, 2022-11-17:
Thanks for the quick response.  I just want to make sure what I saw in the
code and utility folder was correct.  And I was not missing something.

closed 2022-11-17 by darksidelemm