projecthorus/radiosonde_auto_rx #715: Download Logs in Sonde List WebUI does not work

AstroTaka, 2022-11-18:
When I  do not select anything and push "Download Logs", there is no response.
When I choose one or two logs and push "Download Logs", "Bad Request" error occurred.

Should I have to do something?

darksidelemm, 2022-11-18:
Yep, will fix: https://github.com/pallets/flask/issues/4753

darksidelemm, 2022-11-18:
OK, merged in a fix. -beta20

darksidelemm, 2022-11-18:
Keep finding these bugs for me, this is great :-)

AstroTaka, 2022-11-18:
Probably, the reason is that I am a beginner of auto_rx. push every button and try to do anything.
So I found several bugs.

Anyway, I updated beta 20.
When I choose one or two logs and push "Download Logs", zip file download.
It is all right.

But when I do not select anything and push "Download Logs", there is no response.
Is it correct? 
I found a code which download all logs when nothing to choose.
So I ask you whether "no response" is no problem or not.

darksidelemm, 2022-11-18:
Yeah, i think with nothing selected it won't do anything. The download all logs function was a bit dangerous, especially with stations with a lot of logs, so we decided to make it so you have to explicitly select all then click download :-)

AstroTaka, 2022-11-18:
Yes, I think so dangerous to easy to download all log files.
Thanks!

closed 2022-11-19 by AstroTaka