projecthorus/radiosonde_auto_rx #718: Date format of SondeList is for Australia.

AstroTaka, 2022-11-19:
Currently, the date format of Sonde List is for Australia like this.
`temp_time.toLocaleString("en-AU")`

Order of month and date is not common for me.
And I like 24H format but currently 12H format.

About a order of month and date issue, how about the local get from user browser like this.
`temp_time.toLocaleString(window.navigator.language)`

But I have a problem about it does not have '0 padding like this
```
00:00 -> 0:0
12:03 -> 12:3
```

And this is a final solution for me.
`temp_time.toLocaleString(window.navigator.language,{year:"numeric", month:"2-digit", day:'2-digit', hour:'2-digit',minute:'2-digit', second:'2-digit'})`

In Japan, the above code return 24H format.
But 'en-US' and 'en-AU' return 12H format
('en-GB' return 24H format)

The following code is force 24H format for en-US' and 'en-AU' also.
`temp_time.toLocaleString(window.navigator.language,{hourCycle:'h23', year:"numeric", month:"2-digit", day:'2-digit', hour:'2-digit',minute:'2-digit', second:'2-digit'})`

darksidelemm, 2022-11-19:
OK interesting, I usually have the UTC option enabled (I think this might be the default? unsure)

I think the 24 hour format you provided at the end is probably a good approach. I'll have a fiddle around with how to implement this in both the historical and main page.

darksidelemm, 2022-11-20:
OK this change has been made, take a look.

AstroTaka, 2022-11-20:
Yes, many people use UTC I think...
I am light user and I would like to use local timezone...

I checked the latest beta 20.
"Current Scan", "Time" of receiving sonde and "Date" of Sonde List works fine for UTC on/off.

Thank you very much!.

closed 2022-11-20 by AstroTaka