projecthorus/radiosonde_auto_rx #722: strip_sonde_serial does not support MEISEI iMS-100

AstroTaka, 2022-11-22:
When I receive IMS-100, ID of main screen overflow(...) because it try to display "IMS100-xxxxxxx".
strip_sonde_serial does not support IMS100 yet.
Please modify it as following.
`_re = re.compile("^(DFM|M10|M20|IMET|IMET5|IMET54|MRZ|LMS6|IMS100)-")`

Serial of Sonde List has the same problem.
But a log file data already passed strip_sonde_serial.
So please remove "IMS100-" same way as main screen.
```
                    {title:"Serial", field:"serial", width:90, resizable:false, // 105
                        formatter:function(cell, formatterParams, onRendered){
                            return cell.getValue().replace(/^(IMS100)-/,"");
                        }},
```

darksidelemm, 2022-11-22:
I've made the fix to strip_sonde_serial (no version bump).

I won't make the change to the web interface, since it's only for a single log file.

AstroTaka, 2022-11-23:
I checked the latest beta20

Sonde List does not have a issue.
https://drive.google.com/file/d/1j6IW-96HEdtXn4WNEnOCnMDXiPsaMIMN/view?usp=share_link

This is a current receiving sonde list of main screen.
https://drive.google.com/file/d/1CS4bMxnhGn6EjSf02_iTI8yYppzOkN4Z/view?usp=share_link

This is a after change a width of ID column,
https://drive.google.com/file/d/1HEXkOsSYAjCVa0isvkI5go92Ypvc7AWZ/view?usp=share_link

Type of IMS-100 is different from Sonde List.
Sonde List: IMS-100
Main screen: MEISEI

rs1729, 2022-11-23:
Don't know if this matters, but the JSON decoder output for Meisei radiosondes ist
"type": "MEISEI",
and "subtype" either "IMS100" or "RS11G" (depending on radiosonde is iMS-100 oder RS-11G), and that is also the SN prefix of "id".

AstroTaka, 2022-11-24:
Yes, it need to add a code to handle_decoder_line in decode.py to support MEISEI device same as DFM/RS41 subtype support I think.
But, probably the modification makes influence to another part of code...
```
            # Add in the sonde type field.
            if "subtype" in _telemetry:
                if self.sonde_type == "RS41":
                    # For RS41 sondes, we are provided with a more specific subtype string (RS41-SG, RS41-SGP, RS41-SGM)
                    # in the subtype field, so we can use this directly.
                    _telemetry["type"] = _telemetry["subtype"]
                elif self.sonde_type == "DFM":
                    # As of 2021-2, the decoder provides a guess of the DFM subtype, provided as
                    # a subtype field of "0xX:GUESS", e.g. "0xD:DFM17P"
                    if ":" in _telemetry["subtype"]:
                        _subtype = _telemetry["subtype"].split(":")[1]
                        _telemetry["dfmcode"] = _telemetry["subtype"].split(":")[0]
                        _telemetry["type"] = _subtype
                        _telemetry["subtype"] = _subtype
                    else:
                        _telemetry["type"] = "DFM"
                        _telemetry["subtype"] = "DFM"

                    # Check frame ID here to ensure we are on dfm09mod version with the frame number fixes (2020-12).
                    if _telemetry["frame"] < 256:
                        self.log_error(
                            "DFM Frame ID is <256, have you run build.sh recently?"
                        )
                        return False

                // example ----------------------------------------
                elif self.sonde_type == "MEISEI":
                    _telemetry["type"] = _telemetry["subtype"]
                // example ----------------------------------------

                else:
                    # For other sonde types, we leave the type field as it is, even if we are provided
                    # a subtype field. (This shouldn't happen)
                    _telemetry["type"] = self.sonde_type

```

darksidelemm, 2022-11-24:
Yes, that bit of code is a mess. Will fix...

darksidelemm, 2022-11-24:
Looking into this a bit more, the subtype is already used when uploading to SondeHub. It's only locally where it's not handled right. I can at least add it to the strip_sonde_serial section to handle RS11G's better.

AstroTaka, 2022-11-26:
I start to use beta21.
strip_sonde_serial is well.
But the type of iMS-100 of Main screen still MEISEI.

So I applied the following patch.

This is a screen capture of before/after applied patch.
https://drive.google.com/file/d/1iBanxjSXPAhft4a2EVQXtpe3UVuP7Tx0/view?usp=share_link
https://drive.google.com/file/d/1qpos0CgS6MpcXUaBNrCfnNjiVtvt563O/view?usp=share_link

```patch
diff --git a/auto_rx/autorx/templates/index.html b/auto_rx/autorx/templates/index.html
index 8a36ff4..635d97e 100644
--- a/auto_rx/autorx/templates/index.html
+++ b/auto_rx/autorx/templates/index.html
@@ -479,11 +479,22 @@
                 columns:[ //Define Table Columns
                     {title:"SDR", field:"sdr_device_idx", headerSort:true},
                     {title:"Age", field:"age", headerSort:true},
-                    {title:"Type", field:"type", headerSort:true},
+                    {title:"Type", field:"type", headerSort:true, formatter:function(cell, formatterParams, onRendered){
+                        _type = cell.getValue();
+                        if(_type=="MEISEI"){
+                            _subtype = cell.getData().id.split("-")[0];
+                            if(_subtype=="IMS100"){
+                                return "iMS-100";
+                            } else if(_subtype=="RS11G"){
+                                return "RS-11G";
+                            }
+                        }
+                        return _type;
+                    }},
                     {title:'Freq (MHz)', field:"freq", headerSort:true},
                     {title:"ID", field:"id", width:125, headerSort:true, formatter:function(cell, formatterParams, onRendered){
                         _cell_data = cell.getData();
-                        _id = _cell_data.id.replace(/^(DFM|M10|M20|IMET|IMET5|IMET54|MRZ)-/,"");
+                        _id = _cell_data.id.replace(/^(DFM|M10|M20|IMET|IMET5|IMET54|MRZ|IMS100|RS11G)-/,"");
                         _sondehub_id = _cell_data.id.replace(/^(DFM|M10|M20|IMET|IMET5|IMET54|MRZ)-/,"");

                         // Add Sondehub Link
```

closed 2022-12-22 by darksidelemm