projecthorus/radiosonde_auto_rx #726: DFM17 decoded as DFM09P sometimes

AstroTaka, 2022-11-25:
DFM17 decoded as DFM09P sometimes

When I receive DFM17, I saw DFM09P in Main screen sometimes.
But the Sonde List display still DFM17.
So I downloaded the log it and check it.
Usual case, subtype is DFM17.
But DFM09P exists sometimes.

I uploaded sonde.log, debug systen.log and IQ decode.bin.
https://drive.google.com/drive/folders/1rt4RhzZZlHYzlIvQ5RHB5a8qOSfxPbzI?usp=share_link

Please check it.

darksidelemm, 2022-11-25:
It's using this list of 'guesses', in how to interpret the DFM code IDs:
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/sonde_specific.py#L92

I'm wondering if that code might be changing during decoding?

AstroTaka, 2022-11-25:
I searched "decode_dfm_subtype" from whole source code.
It seems to be nobody call it but the similar code exists in the following source.
https://github.com/projecthorus/radiosonde_auto_rx/blob/testing/demod/mod/dfm09mod.c#L880

darksidelemm, 2022-11-25:
Sorry yep, you are right. I take the subtype guess from the decoder itself. 
```
               elif self.sonde_type == "DFM":
                    # As of 2021-2, the decoder provides a guess of the DFM subtype, provided as
                    # a subtype field of "0xX:GUESS", e.g. "0xD:DFM17P"
                    if ":" in _telemetry["subtype"]:
                        _subtype = _telemetry["subtype"].split(":")[1]
                        _telemetry["dfmcode"] = _telemetry["subtype"].split(":")[0]
                        _telemetry["type"] = _subtype
                        _telemetry["subtype"] = _subtype
                    else:
                        _telemetry["type"] = "DFM"
                        _telemetry["subtype"] = "DFM"
```

rs1729, 2022-11-25:
I observed this, too. First I thought it is because of different versions of the decoder.
But sometimes the decoder makes a wrong guess. In principle these two subtypes have different polarity, but it can depend on the receiver setup, which one is which. Thus the decoder looks also in the raw PTU data to determine what channel is raw temperature data etc. Sometimes it can be wrong, in particular for weak signals if there are decoding errors. In the comments is an additional condition that could reduce false detection, but I don't have enough test data, so I don't know if this could produce false positives. If the subtype is wrong, the temperature will be wrong as well. But the majority of frames should be ok.

Another questions is which subtype remains on sondehub as the "final" subtype.

@AstroTaka 
`sensortyp == 'P'`means there are additional channels for pressure data such that the temperature data is not at the "normal" place.

It's always the same for DFM, if you would have an additional CRC over the whole frame, you would know how reliable the decoding is, or you have to use the Hamming code as strict CRC, but then you could not use the error correction. Using the Hamming code for error correction is not reliable for radiosonde transmissions.

AstroTaka, 2022-11-25:
@rs1729  I do not launch a sonde :-) I only receive a sonde which somebody launched.

If you hope to get another kind of log, please let me know.
My 1st tnterest is that it is BUG or unsolvable some kind of limitation.
If you can solve it, I hope to help you as debugger.
Or I understand that it is a limitation.

rs1729, 2022-11-25:
@AstroTaka 
;) Most of us don't.
But the manufacture and the weather stations don't necessarily have the problems we have.

Sure there could be still bugs (most likely), but the main problem is somewhere else, the old backwards compatible frame structure is problematic together with insufficient error detection code. Compare to RS41, much easier to get reliable results.
The DFM decoder makes many compromises. And not all specifications are known to me, so probably it is possible to get better results, though I don't know if it is worth the effort.
Often it depends on the situation, which solution works better. And if you make more restrictions (higher SNR, receiver type), you can also get better results. But e.g. if there are two DFM signals at the same frequency, the packets can be mixed up and it might be difficult sometimes to separate them, see the remarks above. Maybe if you count the channels, though the DFMs don't have all the same channel order. I think the older DFM-06 repeated channel 0 every other subframe, not all do that.
The decoder tries to keep track if there are gaps though (option --dist).

You can try modifying the following and test if it gives better results:
https://github.com/projecthorus/radiosonde_auto_rx/blob/e063903b43300c8dfc7c3ada1f2cad94e84516c4/demod/mod/dfm09mod.c#L718
try
```
        if (gpx->ptu_out >= 0xD || (gpx->ptu_out >= 0xC && gpx->meas24[0] > 1e6)) {
            gpx->sensortyp0xC = 'P';
        }
```
or
```
        if (gpx->ptu_out >= 0xD || (gpx->ptu_out >= 0xC && gpx->meas24[6] < 220e3 && gpx->meas24[0] > 1e6)) {
            gpx->sensortyp0xC = 'P';
        }
```
or maybe adjustig the thresholds for these channels.
But in the end, if a false decoding gets through, the decoded values might suggest, it is a different type.

Perhaps I could add that if the PTU type changes (similar to SN differences), hold back the PTU/SN/subtype output and confirm the old or new state next round, reset everything if necessary. Would require last state and maybe last confirmed state. If it is a different DFM on the same frequency (or someone changed the receiver frequency), the conf-data would have to be reset anyway.

The decoder outputs data if the last GPS packet 8 is received. The PTU packets (channels) run in parallel with a different cycle length (depending on DFM type), that's not making the situation better...

When I run the decoder, it does not matter to me if there are suspicious frames from time to time, if I see the error count for the packets and previous frame. For JSON output it is different, a reliable frame would be desirable, so different receivers send the same data.
The Meisei radiosondes are actually not so different to DFM, not only with regard to the serial number.

rs1729, 2022-11-25:
https://github.com/rs1729/RS/commit/cf91b3966337c3f3c7f26e0b2f0f7af16dec9934
Added a restriction to read only error-free sensor data.
Data acquisition might take a little longer, and sometimes the temperature might be calculated on the basis of the previous frame, but that's better than potentially having completely wrong data.

Will look into more plausibility checks later, but for now this should mitigate false temperature and subtype output.
At least it cleaned up the output when testing the data above.

darksidelemm, 2022-11-25:
This change now in testing, along with the mts01 battery voltage addition

AstroTaka, 2022-11-26:
Thank you very much!
I started to use the latest beta21 which dfm09mod.c modified.

According to NOTAM RJTD 2734/22, I can receive DFM17 until Dec. 5th for every 3 hours.
https://www.skyinfo.jp/

AstroTaka, 2022-11-26:
This is a current result of beta21.
It seems to be OK.
I continue to check it.

```
$ ls  *DFM*.log | xargs grep -c DFM09
----------------------------------------------------
beta20
----------------------------------------------------
20221122-180426_DFM-21063336_DFM17_404405_sonde.log:0
20221122-211403_DFM-21063320_DFM17_404403_sonde.log:1
20221123-000839_DFM-21062767_DFM17_404404_sonde.log:7
20221123-002146_DFM-21062767_DFM17_404404_sonde.log:7
20221123-030754_DFM-21062799_DFM17_404404_sonde.log:2
20221123-061007_DFM-21063293_DFM17_404404_sonde.log:9
20221123-090942_DFM-21062581_DFM17_404404_sonde.log:2
20221123-120932_DFM-21062667_DFM17_404404_sonde.log:3
20221123-151051_DFM-21063310_DFM17_404403_sonde.log:1
20221123-180927_DFM-21062654_DFM17_404404_sonde.log:3
20221123-210358_DFM-21062755_DFM17_404405_sonde.log:5
20221124-000932_DFM-21062768_DFM17_404405_sonde.log:1
20221124-030234_DFM-21063260_DFM17_404404_sonde.log:15
20221124-060335_DFM-21063297_DFM17_404405_sonde.log:2
20221124-090628_DFM-21062781_DFM17_404404_sonde.log:0
20221124-120308_DFM-21063159_DFM17_404404_sonde.log:13
20221124-150401_DFM-21062814_DFM17_404404_sonde.log:12
20221124-180302_DFM-21065745_DFM17_404404_sonde.log:1
20221124-210308_DFM-21062811_DFM17_404403_sonde.log:5
20221125-000304_DFM-21062739_DFM17_404405_sonde.log:0
20221125-030826_DFM-21062551_DFM17_404405_sonde.log:5
20221125-060538_DFM-21065757_DFM17_404402_sonde.log:9
20221125-090357_DFM-21065738_DFM17_404405_sonde.log:3
20221125-123110_DFM-21070040_DFM17_404406_sonde.log:1
20221125-150846_DFM-21062759_DFM17_404404_sonde.log:0
20221125-180241_DFM-21070598_DFM17_404404_sonde.log:3
20221125-210744_DFM-21062740_DFM17_404404_sonde.log:0
20221126-000641_DFM-21063148_DFM17_404404_sonde.log:3
----------------------------------------------------
beta21
----------------------------------------------------
20221126-031035_DFM-21069994_DFM17_404404_sonde.log:0
20221126-060755_DFM-21070919_DFM17_404404_sonde.log:0
20221126-090458_DFM-21069970_DFM17_404404_sonde.log:0
20221126-120449_DFM-21070036_DFM17_404402_sonde.log:0
20221126-150254_DFM-21070144_DFM17_404404_sonde.log:0
```

AstroTaka, 2022-12-11:
This is a additional result.
This is big better than beta20.

beta21
----------------------------------------------------
20221126-180259_DFM-21070633_DFM17_404403_sonde.log:0
20221126-210310_DFM-21070545_DFM17_404404_sonde.log:0
20221127-000253_DFM-21069961_DFM17_404405_sonde.log:0
20221127-030238_DFM-21063815_DFM17_404403_sonde.log:0
20221127-060238_DFM-21069925_DFM17_404404_sonde.log:0
20221127-090752_DFM-21070620_DFM17_404404_sonde.log:0
20221127-120340_DFM-21070896_DFM17_404404_sonde.log:0
20221127-150650_DFM-21070122_DFM17_404404_sonde.log:0
20221127-180329_DFM-21070170_DFM17_404404_sonde.log:0
20221127-211800_DFM-21070604_DFM17_404405_sonde.log:0
20221128-000422_DFM-21069977_DFM17_404404_sonde.log:0
20221128-030358_DFM-21070565_DFM17_404404_sonde.log:0
20221128-060426_DFM-21070617_DFM17_404404_sonde.log:0
20221128-090244_DFM-21070110_DFM17_404404_sonde.log:0
20221128-120813_DFM-21070117_DFM17_404405_sonde.log:0
20221128-150615_DFM-21063155_DFM17_404405_sonde.log:0
20221128-180906_DFM-21065555_DFM17_404404_sonde.log:0
20221128-210706_DFM-21070907_DFM17_404405_sonde.log:0
20221129-000653_DFM-21070566_DFM17_404403_sonde.log:0
20221129-030749_DFM-21062791_DFM17_404403_sonde.log:0
20221129-060613_DFM-21063191_DFM17_404403_sonde.log:0
20221129-090740_DFM-21063241_DFM17_404405_sonde.log:3
20221129-120738_DFM-21062709_DFM17_404404_sonde.log:4
20221129-150513_DFM-21063183_DFM17_404404_sonde.log:0

I attached 20221129-090740_DFM-21063241_DFM17_404405_sonde.log
and 20221129-120738_DFM-21062709_DFM17_404404_sonde.log for your reference.

https://drive.google.com/file/d/1huF4wYFKEItRTPV_Vqfvmzh6al7iIbmL/view?usp=share_link

I think that we can close this issue...

rs1729, 2022-12-11:
As long as it shows temperature -273C (invalid) for misidentified "DFM09P" its ok. And it does not seem to happen that often.

darksidelemm, 2022-12-11:
On a related note, the ID that we're getting (e.g. 21062709) is a truncated form of what's on the DFM17 label. (e.g. it's 22xxx-062709). 
@rs1729 do you know if any more of the ID is transmitted? It would be nice to get the complete ID out of it.

rs1729, 2022-12-17:
Added a check, if temperature T is invalid, reset DFM09P/DFM17 subtype,
i.e. if `"subtype": "0xC:DFM17"`or `"subtype": "0xC:DFM09P"`,  it will be set to
`"subtype": "0xC:DFMxX"` until the temperature is valid again.

Didn't find any test data where the wrong subtype was set with the last version, it should not happened often anyway.
Don't know @AstroTaka , if you also have recordings for two
```
20221129-090740_DFM-21063241_DFM17_404405_sonde.log:3
20221129-120738_DFM-21062709_DFM17_404404_sonde.log:4
```

AstroTaka, 2022-12-27:
Thank yo very much.

Recently, there is no DFM receive.
I will continue to check DFM sonde.

closed 2023-02-03 by darksidelemm