projecthorus/radiosonde_auto_rx #733: Question: why ss_power does not use '-e' (search_step) option?

AstroTaka, 2022-12-20:
I compared a option between ss_power and rtl_power in sdr_wrappers.py.
I found that ss_power does not use '-e' option.
https://github.com/projecthorus/radiosonde_auto_rx/blob/5f56e07861d88c6e641d00e6f49b90801d523d36/auto_rx/autorx/sdr_wrappers.py#L562

Do you have some reason?

I changed it to use step as '-e 800'.
`f"-e {step} "`
But I could not find anything change about scan time and scan plot.

darksidelemm, 2022-12-20:
I don't think that option does anything, as the spyserver client doesnt have any control over the FFT length on the spyserver side. It just presents whatever it gets from the spyserver.

AstroTaka, 2022-12-20:
I see...
Thank you very much!

closed 2022-12-20 by AstroTaka