projecthorus/radiosonde_auto_rx #738: missing auto_rx/ file

Modern-Memories, 2022-12-26:
I am trying to run the test email function and I seem to be mixing the auto_rx/ file for the test 
pi@ARX01:~ $ cd /home/pi/radiosonde_auto_rx/
pi@ARX01:~/radiosonde_auto_rx $ ls
log  station.cfg

What can I get the template for this so I can ass it ?

darksidelemm, 2022-12-26:
Ahh sorry, the test email example is intended for use with a 'native' installation. I'm assuming you are using docker.

With Docker, you can probably get a shell in the docker container with this command: https://github.com/projecthorus/radiosonde_auto_rx/wiki/Docker#opening-a-shell-within-an-existing-running-container

Then run the test command:
```
python3 -m autorx.email_notification
```

Modern-Memories, 2022-12-26:
Thanks that worked like a charm

closed 2022-12-27 by darksidelemm