projecthorus/radiosonde_auto_rx #743: No able to pick up local balloons

Modern-Memories, 2023-01-04:
### I have been trying to trouble shoot this for a few day and I can not seem to pick up balloons that are being a few km launched south of me twice a say. The one in the air right now is DFM17 22045109 is bing picked up by station name KE4GLA-8  at more that 100km further away than me. As far as I can tell tell everything seems to be working correctly however I will note that I am having issues with running most of the testing applications relating to cmake and not being able to find modules example from attempting to install :

> sudo apt-get update
> sudo apt-get install -y cmake gnuradio-dev gr-osmosdr qt5-default libqt5svg5-dev libpulse-dev
> git clone https://github.com/gqrx-sdr/gqrx.git
> cd gqrx
> mkdir build
> cd build
> cmake ..
> make -j4
> sudo make install
> Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
> Get:2 https://download.docker.com/linux/debian bullseye InRelease [43.3 kB]   
> Hit:3 http://deb.debian.org/debian bullseye InRelease                          
> Hit:4 http://deb.debian.org/debian bullseye-updates InRelease                  
> Hit:5 http://archive.raspberrypi.org/debian bullseye InRelease                 
> Fetched 43.3 kB in 1s (36.8 kB/s)
> Reading package lists... Done
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> Package qt5-default is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> 
> E: Package 'qt5-default' has no installation candidate
> fatal: destination path 'gqrx' already exists and is not an empty directory.
> mkdir: cannot create directory ‘build’: File exists
> CMake Error at CMakeLists.txt:126 (find_package):
>   By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by "Qt5", but
>   CMake did not find one.
> 
>   Could not find a package configuration file provided by "Qt5" with any of
>   the following names:
> 
>     Qt5Config.cmake
>     qt5-config.cmake
> 
>   Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
>   to a directory containing one of the above files.  If "Qt5" provides a
>   separate development package or SDK, be sure it has been installed.
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/pi/gqrx/build/CMakeFiles/CMakeOutput.log".
> make: *** No targets specified and no makefile found.  Stop.
> make: *** No rule to make target 'install'.  Stop.

### I have attempted to reinstall the auto_rx serval times to correct it .

Station Name: AMGRUNT
Location: Valrico FL USA

Version: Auto_RX 1.5.10 running in docker

Computer: Raspberry Pi 4
Dongle: RTL-SDR V3 with NOOELEC Luna
Antena: [UHF Fiberglass Mobile Radio Antenna,70CM 400-470mhz GMRS Base Antenna 17inch So239 Connector ]([url]https://www.amazon.com/dp/B08TWN78MM?psc=1&ref=ppx_yo2ov_dt_b_product_details(url))
** There is a 10ft SMA cable running from the Luna to the antenna 

PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
[... 97 lines cut ...]
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
No E4000 tuner found, aborting.

- **_Results of rtl_test -s 3.2e6_**

Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.

- _**Results from rtl_biast**_

Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
root@ARX01:/home/pi/rtl-sdr/build# 

[image]

darksidelemm, 2023-01-05:
This is sounding like a RF issue. 

Was that scan plot produced while there was a radiosonde in the air? (that you could see on sondehub.org, for example)

Modern-Memories, 2023-01-05:
> This is sounding like a RF issue. 
> 
> 
> 
> Was that scan plot produced while there was a radiosonde in the air? (that you could see on sondehub.org, for example)

Yes... sorry o forgot to attach that. You can see how close it's path is the me here. 

[image]

darksidelemm, 2023-01-05:
Sure, but unless its showing up on the scan plot, it'll never get decoded. Your scan plot doesn't show a signal at 403.01 MHz.

Modern-Memories, 2023-01-05:
> Sure, but unless its showing up on the scan plot, it'll never get decoded. Your scan plot doesn't show a signal at 403.01 MHz. 

Oh yea... that scan plot was done while the ballon referenced in the original post was in the air.

Modern-Memories, 2023-01-06:
I swiped to a different RTL-SDR and NOOELEC Luna and also what made the main different was that I swapped to a [NOOELEC RattleSnake antenna](https://www.nooelec.com/store/ratlsnake-m6.html) (3 blue rings). we will see later today if it works. I have it in my attic right now, so it it doesn't work ill move it up to the roof. I am guess that this fiberglass antenna I got off Amazon is no the correct one?
[image]
[image]

darksidelemm, 2023-01-06:
That antenna should have been fine, but perhaps it has a fault, or there is some other connection issue.

Modern-Memories, 2023-01-07:
Problem solved!! I could not get 2 SDRs working, but for now we are doing what is needed. Now I just need to put the antenna in the roof. Along with my Comet DS-150S and start working on SDR setup for HF, UHF/VHF, and WinLink. If someone knows a good amp to use for all that please let me know. 

[image]

closed 2023-01-07 by Modern-Memories