projecthorus/radiosonde_auto_rx #744: IQ data of RS-11G/iMS-100/RS41 which is on a same balloon

AstroTaka, 2023-01-13:
I received RS-11G/iMS-100/RS41 which is on a same balloon.

https://sondehub.org/7135152
https://sondehub.org/2104700
https://sondehub.org/T5050463

I recorded IQ data for every sonde.
Currently, RS-11G support only position.
I hope that the IQ data help to improve a RS-11G support.

https://drive.google.com/drive/folders/1qSYb8o0Q2UuNKscty3200J5HrSuXFqVT?usp=share_link

[image]

einergehtnochrein, 2023-01-23:
Thank you for these recordings, very helpful!
It enabled me to create a first version of temperature/humidity support for the RS-11G. It's similar to the iMS-100, the main difference is in the calibration array layout. Temperature seems pretty ok, but humidity probably requires a bit more work to increase accuracy.

I believe the best way would be for me to create a pull request to the RS repository of @rs1729, from where @darksidelemm can pull the changes and include them in radiosonde_auto_rx.

Was this launch of multiple sondes on one balloon a one-time event or will there be more launches in the future?
Do you have more RS-11G recordings that you can share?

rs1729, 2023-01-23:
@einergehtnochrein 
Since you made the PTU for iMS-100, you are welcome to add it for RS-11G as well. Although I don't know how long the RS-11G will be in service.
The GRUAN TD-5 document suggests there are similar, as you probably know.
The limited amount of data might be a problem though. Don't believe the regular weather stations will launch RS-11G, but maybe some university.

einergehtnochrein, 2023-01-24:
@rs1729: I hope to have the patches ready by tomorrow. You are right, the RS-11G is probably obsolete, but it wasn't a huge effort to implement the changes.

I made a comparison of temperature and humidity values for the three sondes, and I believe it's good enough for our hobbyist demands...
[image]
[image]

rs1729, 2023-01-24:
Looks good!

darksidelemm, 2023-02-03:
@AstroTaka i've merged these updates into the testing branch.
Could you update and let us know what you see on the next RS11G launch?

closed 2023-02-03 by darksidelemm

AstroTaka, 2023-02-18:
@einergehtnochrein 
Last night, I received three sonde with the same situation.
I uploaded IQ files and log files.

https://drive.google.com/drive/folders/1TQlpKN5d0Bc3Gg87IRsh0FXaZFuN2UsA?usp=share_link

It is just for your reference.

Since yesterday, It is in my 1st time to receive a RS11G which is launched by some private company for environment assessment of waste incineration plant.
They are launching it for every two hours.
I only uploaded log files.
If you want to get IQ also, please let me know.

@darksidelemm
I have no issue to receive RS11G now.

einergehtnochrein, 2023-02-18:
@AstroTaka
Thanks for the files!
I can decode RS41 and iMS-100, but unfortunately the RS-11G signal is a bit too noisy, and I cannot collect enough calibration data to decode temperature and humidity. :-(

It would be great if you could create recordings for the standalone RS-11G flights! Very few recordings of RS-11G exist, and it might be helpful to have them for future reference.

AstroTaka, 2023-02-18:
@einergehtnochrein I am recording MEISEI sonde which use limited frequency range and all frequency range RS41 too.
I did not record another RS11G today.
I found one RS11G record for 20230217-060341_RS11G-7136073_RS11G_405399_sonde.log.
I uploaded decode_IQ_SPY04_MEISEI_2023-02-17T060247Z.bin.

Please check it.

AstroTaka, 2023-02-21:
@einergehtnochrein 
The RS11G sonde launch stopped suddenly and change to iMS-100...

Today, I received another new RS11G.
I checked a position and a frequency of it.
I think that it is launched by H.Q of MEISEI.
https://goo.gl/maps/nXacv6oGt1fwYdEV7
https://www.tele.soumu.go.jp/musen/SearchServlet?pageID=4&IT=A&DFCD=0004254220&DD=1&styleNumber=21

I uploaded log and IQ file.
I succeed to decode it by fsk_demod and meisei100mod.
Please check it whether it decodable.
https://drive.google.com/drive/folders/1PMteAWQWY-oWDXIIsQGOHRmDQum03Lgc?usp=share_link

einergehtnochrein, 2023-02-21:
@AstroTaka
Works for me. Thanks you!