projecthorus/radiosonde_auto_rx #749: `read_auto_rx_config` ignores `no_sdr_test=True`

pavolgaj, 2023-02-05:
In the latest version (1.6.0), the parameter `no_sdr_test=True` in `config.read_auto_rx_config` is ignored - lines #753 and #754 are commented. So, testing without SDR is not possible. After uncommenting them, the copy of the config to the global config is missing. I changed these lines to:
`if no_sdr_test: `
`    global_config = copy.deepcopy(auto_rx_config)`
`    return auto_rx_config`

darksidelemm, 2023-02-05:
These lines don't appear to be commented in the repository?
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/config.py#L753

Are these the ones you are referring too?

pavolgaj, 2023-02-06:
Yes, these ones. Sorry, I didn't see that they aren't commented on the repository. I don't know why I had them commented on my Raspberry...

closed 2023-02-06 by pavolgaj