projecthorus/radiosonde_auto_rx #750: Support for AK2 type sondes

ly5p, 2023-02-07:
Please add support for AK2 sondes decoding

darksidelemm, 2023-02-07:
Support for a radiosonde type can only be added with information on:
- Where the radiosonde is used.
- What frequency the radiosonde is used

We also would need good recordings (preferably IQ recordings) of the radiosonde signal, and ideally 'truth' data for a flight (e.g. from a meteorological agency) to compare against.

darksidelemm, 2023-05-16:
Any further updates on this? Without further information I can't do anything.

darksidelemm, 2023-08-15:
Hi @ly5p, i see some of these sondes are being decoded with some custom rdz_ttgo_sonde fork. Do you know anything about this?

ly5p, 2023-08-15:
I have got fw from author, also hw mod is simple to replace rare quartz to
30.720MHz here is some info
https://github.com/UE9AAG/Decoder-meteosonde-AK2-02mN

darksidelemm, 2023-08-15:
OK, good to know what's being used.

Anyway, since this is using LoRa modulation, it's not something we can support in auto_rx. That'll have to be left to the TTGO-type devices, so I'll close this issue.

closed 2023-08-15 by darksidelemm