projecthorus/radiosonde_auto_rx #752: V1.6.0 - Not updating Sondehub for IMET54

raasblaar, 2023-02-20:
I have the below from syslog. It was working fine in V5.10.
Standard install on RPI 3 B - not docker.

I deduct that below error is confusing the Sondehub upload. I see no upload messages in syslog as per previous version.

I upgraded by deleting the ~/radiosonde_auto_rx directory and installed V1.6.0 from scratch as per the instructions.

Feb 20 13:37:17 rpi3 auto_rx[411]: 2023-02-20 13:37:17,662 INFO:Task Manager - Detected new IMET5 sonde on 405.000 MHz!
Feb 20 13:37:17 rpi3 auto_rx[411]: 2023-02-20 13:37:17,664 INFO:Halting Scanner to decode detected radiosonde.
Feb 20 13:37:17 rpi3 auto_rx[411]: 2023-02-20 13:37:17,665 INFO:Scanner (RTLSDR 0) - Waiting for current scan to finish...
Feb 20 13:37:34 rpi3 auto_rx[411]: 2023-02-20 13:37:34,877 INFO:Scanner (RTLSDR 0) - Scanner Thread Closed.
Feb 20 13:37:34 rpi3 auto_rx[411]: 2023-02-20 13:37:34,879 INFO:Task Manager - SDR #0 has been allocated to Decoder (IMET5, 405.000 MHz).
Feb 20 13:37:37 rpi3 auto_rx[411]: 2023-02-20 13:37:37,886 INFO:Decoder (RTLSDR 0) IMET5 405.000 - Using fsk_demod decoder chain.
Feb 20 13:37:37 rpi3 auto_rx[411]: 2023-02-20 13:37:37,933 INFO:Decoder (RTLSDR 0) IMET5 405.000 - Starting decoder subprocess.
Feb 20 13:37:40 rpi3 auto_rx[411]: 2023-02-20 13:37:40,201 INFO:Telemetry Logger - Opening new log file: /home/pi/radiosonde_auto_rx/auto_rx/log/20230220-113740_IMET5-55078855_IMET5_405000_sonde.log
Feb 20 13:37:41 rpi3 auto_rx[411]: /home/pi/radiosonde_auto_rx/auto_rx/autorx/log_files.py:425: RuntimeWarning: divide by zero encountered in log
Feb 20 13:37:41 rpi3 auto_rx[411]:   * (np.log(_rh / 100) + ((17.625 * _temp) / (243.04 + _temp)))
Feb 20 13:37:41 rpi3 auto_rx[411]: /home/pi/radiosonde_auto_rx/auto_rx/autorx/log_files.py:428: RuntimeWarning: divide by zero encountered in log
Feb 20 13:37:41 rpi3 auto_rx[411]:   - np.log(_rh / 100)
Feb 20 13:37:41 rpi3 auto_rx[411]: /home/pi/radiosonde_auto_rx/auto_rx/autorx/log_files.py:424: RuntimeWarning: invalid value encountered in double_scalars
Feb 20 13:37:41 rpi3 auto_rx[411]:   243.04
Feb 20 13:37:45 rpi3 auto_rx[411]: 2023-02-20 13:37:45,517 INFO:E-Mail - E-mail notification sent.

In the meanwhile I changed back to the previous version and all ok.

Thanks

HoshenKadosh, 2023-02-20:
Same issue.

darksidelemm, 2023-02-20:
That's... odd. The function with the issue should only be called when a log file is requested for display on the web interface.
I don't see anywhere else in the code where this could be called. 

Did any data make it into /home/pi/radiosonde_auto_rx/auto_rx/log/20230220-113740_IMET5-55078855_IMET5_405000_sonde.log ?

darksidelemm, 2023-02-20:
OK, I have found another bug that will prevent iMet-54/50 data from showing up on SondeHub in v1.6.0. I've got a fix in the works...

darksidelemm, 2023-02-20:
OK, there is now a fix in the testing branch (1.6.1-beta1). Please try this and see if it works.

The docker images for the testing branch are still being built (check here for status: https://github.com/projecthorus/radiosonde_auto_rx/actions )

raasblaar, 2023-02-22:
Running 1.6.1-beta1
Sondehub is now being updated for IMET54

Thanks

darksidelemm, 2023-02-23:
Re-opening this until 1.6.1 has been released properly.

closed 2023-05-16 by darksidelemm