projecthorus/radiosonde_auto_rx #755: Wideband detection in SpyServer implemention

SteveBorensteinNOAA, 2023-02-24:
This may be an issue in the ss_client, but starting here. Trying to test the SpyServer implementation for multiple sonde detection of iMet 1 sondes (or iMet 4 wideband). Currently using RTLSDR module. Sonde is detected using rtl_power, but not using ss_client. Is it possible ss_power is only looking for narrowband FM?

darksidelemm, 2023-02-24:
The issue will not be in the peak detection stage, but in the next stage, where we run dft_detect over the frequency where the peak was detected and try and determine what the radiosonde type is.

The bandwidth used in this step is only 48 kHz, so I suspect it will not detect wide-band iMet sondes correctly. Also any further decoding steps all assume narrowband sondes.

Are these wideband iMet sondes still in common use? I'm surprised there's even a wideband version of the iMet-4 available!

SteveBorensteinNOAA, 2023-02-24:
Yeah, there are still wideband sondes in use. One of my jobs is hopefully to get our group to move away from them. ;) 

I reset the iMet 4 I'm using for testing to narrow band (12 khz vs 50), but it's still not being picked up with the SpyServer configuration.  The direct to RTLSDR picks up the NB version very quickly. 

I'm still familiarizing myself with the back end of these utilities. I would agree that it seems that there is something differently happening in dft_detect, but I'm not sure how to troubleshoot it.

darksidelemm, 2023-02-24:
Hmm, given i now have abstraction layers in place for the SDRs, I would have thought that detection would work fine for the narrowband iMets. It's sending the same bandwidth IQ in from either the RTLSDR or the SpyServer.
I'm fairly sure there's a station in europe that has been receiving iMet sondes with a spyserver with success. 

Unfortunately I don't have iMet sondes flying near me, so the only testing I can do is on the bench, with a probably not-realistic signal level. It's also possible that too much signal will result in the detection step not working correctly.

I'm guessing the peak from the radiosonde signal still shows up in the scan plot?

darksidelemm, 2023-02-24:
I just had a test with a local iMet-4, and it was picked up in the scan and detected correctly, with decoding then starting. This did have a pretty high SNR however.

SteveBorensteinNOAA, 2023-02-24:
Are you using RTLSDR or Airspy on SpyServer? Seeing it's working fine when not using SpyServer, but not when I am. I'm curious if there's a difference between the way that the data is being encoded from SpyServer vs straight from rtl_fm.

SteveBorensteinNOAA, 2023-02-24:
BTW - This is what I get when I run the detector alone on the command line. 

`./ss_iq -f 402000000.0 -s 48000 -r 0.0.0.0 -q 5555 - 2>/dev/null|./dft_detect -v -t 5 --iq --bw 60 - 48000 16 2
error: open 2
`

SteveBorensteinNOAA, 2023-02-24:
Running the RTL_FM version: 

`
 rtl_fm -M raw -F9 -p 0 -d 0 -s 48000 -f 402000000 - 2>/dev/null | ./dft_detect -v -t 5 --iq --bw 20 --dc - 48000 16
sample: 4580
IMET4: 0.9834 , -2742.8Hz   [ fq-ofs: -0.057142 = -2742.8Hz ]
`

darksidelemm, 2023-02-24:
The detection stage command for ss_iq should be:
```
./ss_iq -f 402000000.0 -s 48000 -r 0.0.0.0 -q 5555 - 2>/dev/null|./dft_detect -t 5 --iq --bw 20 --dc - 48000 16
```
(e.g. remove the last 2 you have in your line)

The bandwidth (--bw 20) will need to be less than 48 kHz.

darksidelemm, 2023-02-24:
I'm using an Airspy Mini.

There shouldn't be any difference - it should just be 48 kHz bandwidth IQ data. 

Something that might be worth testing is to see if decoding works at all. You can test this either by running up auto_rx directly with:

python3 auto_rx.py -f 402.000 -m IMET -v

(This assumes you're running auto_rx 'natively', not in docker),

Alternatively, you could add an entry into the 'always_decode' section of the configuration file:
```
always_decode = [[402.0,"IMET"]]
```

SteveBorensteinNOAA, 2023-02-24:
Yes, when running on the RTLSDR, decoding works and I get data in auto_rx. I believe that RTLSDR is 8 bit, and Airspy is 16. I am curious if there is something happening in the client stage? That said, the PSD seems to work on scan.

darksidelemm, 2023-02-24:
So the RTLSDR has an ADC size of 8-bit (and the Airspy is actually 12-bit), but the sample width that comes out of both rtl_fm and ss_iq is signed 16-bit, as this is after tuning and decimation. We never pass around the 'raw' IQ data from the SDR.

SteveBorensteinNOAA, 2023-02-24:
BTW - forcing the decoder as suggested works on RTLSDR direction, but not RTLSDR through SpyServer. I think for now, I'm going to focus my evaluation on the direct RTLSDR route. The goal with this, however, is to work towards autotuning and multiple simultaneous receive.

darksidelemm, 2023-02-24:
OK. I haven't tested SpyServer with the RTLSDR. Given the iMet's rather large frequency steps, I would suggest an Airspy Mini is probably a better option, which would provide 4.8 MHz of coverage, covering all the iMet steps (402 through 405 MHz). 

Just be aware that as the iMet radiosondes don't transmit a serial number (I *really* wish Intermet Systems would do something about this...), a serial number is automatically generated by auto_rx based off the transmit frequency and the power-on time. 
This works ok for the iMet-4, but the iMet-1 does weird things in the telemetry, with occasional skips of sequence numbers, which breaks things.

SteveBorensteinNOAA, 2023-02-24:
Our standard method of receive is to use SkySonde server, developed in our group (before me, but I've inherited it). All the receive math is being done in C#, and is rather CPU inefficient. While I've got dual receive and autotune working in that setup, I'd like to work towards something like an RPi based, automatic full band receiver. 

We have lots of SDRPlay RSPdx, and I see that the KA9Q server now supports it. Maybe I'll give it a shot. Or, maybe I'll buy an Airspy to add to the mix. 

Thanks for the help!

darksidelemm, 2023-02-24:
Unfortunately I don't yet support the KA9Q server in auto_rx. There's still interface software that needs to be developed on that end before I can hook into it like I do with the SpyServer. Unsure when that will happen.

closed 2023-05-16 by darksidelemm