projecthorus/radiosonde_auto_rx #757: UDP mode functionality broken in 1.6.0

buttim, 2023-03-03:
As stated in https://github.com/projecthorus/radiosonde_auto_rx/wiki/RS----Auto_RX-JSON-Formats the special id TCP001 used as the device_id value in station.cfg should activate the UDP mode of auto_rx.
auto_rx does not recognize the special value and tries to open the SDR device with a key value of TCP001, resulting in errors:

RTLSDR - Could not find RTLSDR with serial TCP001!
WARNINGScanner (RTLSDR TCP001) - SDR produced no output... resetting and retrying.

darksidelemm, 2023-03-03:
Can you confirm you are starting auto_rx with:
```
python3 auto_rx.py -m UDP -f 403.000
```
...as specified in the page you linked.

This works for me.

buttim, 2023-03-03:
I read the line in the doc and then completely forgot it, i'm really sorry.
Now I don't get an error and the UDP port 50000 is open.
Sadly it doesn't seem to react to my UDP packets, do I need to specify a specific origin port?
Should the log say something if the packet is malformed?

buttim, 2023-03-03:
The packet I'm sending:
`{"id": "F0000001", "type": "rs41", "frame": 999, "datetime": "2023-03-04T00:03:28.730490Z", "lat": 45.5, "lon": 7.65, "alt": 8848}`

darksidelemm, 2023-03-03:
I suspect that type field would probably get discarded due to not being what auto_rx expects (RS41).
Still, this should result in an error. 

Try running auto_rx with `-v` to see if anything else is showing up.

darksidelemm, 2023-03-03:
I should also ask - what are you trying to do here?

buttim, 2023-03-03:
No output when I send the packet even with the -v option.
I corrected the type to RS41 but nothing changed

buttim, 2023-03-03:
Just experimenting for now, but the idea could be attach a TTGO to the raspberry USB and use the auto_rx framework

darksidelemm, 2023-03-03:
Honestly, I'd prefer that didn't happen. 
The UDP input was only ever intended to allow the RS decoders to be run independently from auto_rx for testing purposes, not as a way for other decoders to feed in data. There's too much linkage between the RS decoders and auto_rx for that to work reliably, without significant work to make other decoders 'look' the same as the RS decoders.

buttim, 2023-03-03:
I see. Maybe it should be worth it to quickly check if the functionality is indeed broken and update the docs accordingly.
Thanks for helping and feel free to close this issue

closed 2023-03-03 by darksidelemm

Szetya, 2023-04-30:
I use the chasemapper program for this. There everything runs together (AutoRX and TTGO data). TTGO settings:
Chasemapper active: 1
Chasemapper UDP host: host machine IP or address 
Chasemapper UDP port: 55673
Only one profile with port 55673 is created in the chasemapper. 
RDZsonde is slightly modified, it also sends the station name in the json. 
It works fine. 
[image]
[image]