projecthorus/radiosonde_auto_rx #760: Question regarding telemetry output for use with automatic directional antenna.

lelun, 2023-03-13:
So i have a project to upgrade my raspberry pi rtlsdr reciver to have a tracking yagi antenna and a antenna switch so it can switch between a omidirectional antenna and the servo controled yagi antenna. 

i am wondering where would be the best place to get the lat long alt cordinates and active decoding status from auto_rx. 
my intent is to have another progam on the pi use the infomation to help a arduino guide the antenna on to the radio sonde. and to switch from the omni directional antenna to the directional antennas once decoding has started.

so the data i need is decoding status, and lat long / alt of the sonde. what would be the best way to get it out of my radiosonde auto rx?

 current sett up is a RP3b+ with  auto_rx running in docker with a single RTLSDR dongle attached to it via a USB extention calbe to cut down on interferance. and i am planing to add a arduino to controll the antenna servos and a gps/compass and a relay antenna switch.

darksidelemm, 2023-03-13:
The UDP broadcast output is going to be the best option, which is emitted on port 55673 by default.
There's an example on how to receive this here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/utils/horus_udp_receiver_example.py

auto_rx can also control a rotctld instance already: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L412
.. though of course this doesn't help with your antenna switching.

closed 2023-03-13 by darksidelemm

lelun, 2023-03-14:
Thanks for the help much apreceated :)