projecthorus/radiosonde_auto_rx #761: I found a RS11G sonde which cannot decode a serial

AstroTaka, 2023-03-16:
Today, I found RS11G sonde which cannot decode a serial as "WARNING:Payload ID RS11G-xxxxxx is invalid".
Usually, the "WARNING:Payload ID RS11G-xxxxxx is invalid" is only a few time in the beginning phase.
But in this case, it continue about 5 min and finally, stop decoder subprocess as "Closed decoder subprocess".
And repeat detect MEISEI sonde and stop decode subprocess.

I hope to know why a serial of the RS11G cannot decode.

I uploaded IQ files of the RS11G and log of two IMS100 and system.log.
https://drive.google.com/drive/folders/1OsekprWasqU70GNarbd2_ZD_1g8fx3-G?usp=sharing

I have one information for you.
When I received the RS11G, I received two IMS100 which is seems to be launched  at HQ of MEISEI office also.
https://sondehub.org/1102438
https://sondehub.org/2102814
And the both seems to be mounted in a same balloon.
So it is possible that the RS11 is mounted in a same balloon also.

I have another question.
I find "Detected a Meteo-Radiy MRZ Sonde!" in system.log sometimes.
I do not know whether it is normal or not.
You can find it in the uploaded system.log also.
Please check.

rs1729, 2023-03-20:
There are more than a few bit errors, not easy to get a clear picture.
Most of the config values are FFFFFFFF. Frequency was 404.4 MHz? One of the few valid config values [15]: freq. Normally the SN would be at [0], [16], [32], [48]. Perhaps there was something wrong with the sensor boom.
GPS data seems to be fine.
```
[10112]  # [00] ffffffff : -nan #   lat: 36.28291  lon: 139.95721  alt: 10246.97   vH: 37.11m/s  D: 103.8  vV: 2.33m/s   2023-03-16 
[10113]  # [01] ffffffff : -nan #   04:16:23.000 
[10114]  # [02] ffffffff : -nan #   lat: 36.28283  lon: 139.95762  alt: 10249.23   vH: 37.25m/s  D: 103.7  vV: 2.33m/s   2023-03-16 
[10115]  # [03] ffffffff : -nan #   04:16:24.000 
[10116]  # [04] ffffffff : -nan #   lat: 36.28275  lon: 139.95802  alt: 10251.64   vH: 37.21m/s  D: 103.9  vV: 2.35m/s   2023-03-16 
[10117]  # [05] ffffffff : -nan #   04:16:25.000 
[10118]  # [06] 02000000 : 0.0 #   lat: 36.28267  lon: 139.95842  alt: 10254.11   vH: 38.07m/s  D: 103.0  vV: 2.36m/s   2023-03-16 
[10119]  # [07] ffffffff : -nan #   04:16:26.000 
[10120]  # [08] ffffffff : -nan #   lat: 36.28260  lon: 139.95885  alt: 10256.31   vH: 39.27m/s  D: 101.8  vV: 2.33m/s   2023-03-16 
[10121]  # [09] ffffffff : -nan #   04:16:27.000 
[10122]  # [10] ffffffff : -nan #   lat: 36.28252  lon: 139.95927  alt: 10258.62   vH: 39.15m/s  D: 102.5  vV: 2.34m/s   2023-03-16 
[10123]  # [11] ffffffff : -nan #   04:16:28.000 
[10124]  # [12] ffffffff : -nan #   lat: 36.28245  lon: 139.95970  alt: 10260.99   vH: 39.18m/s  D: 102.2  vV: 2.35m/s   2023-03-16 
[10125]  # [13] ffffffff : -nan #   04:16:29.000 
[10126]  # [14] ffffffff : -nan #   lat: 36.28237  lon: 139.96011  alt: 10263.37   vH: 37.94m/s  D: 104.0  vV: 2.35m/s   2023-03-16 
[10127]  # [15] 83600000 : 7.0 #   04:16:30.000 
[10128]  # [16] ffffffff : -nan #   lat: 36.28228  lon: 139.96052  alt: 10265.70   vH: 37.86m/s  D: 104.6  vV: 2.35m/s   2023-03-16 
[10129]  # [17] ffffffff : -nan #   04:16:31.000 
[10130]  # [18] ffffffff : -nan #   lat: 36.28220  lon: 139.96093  alt: 10268.02   vH: 37.63m/s  D: 104.1  vV: 2.35m/s   2023-03-16  : fq 404400
[10131]  # [19] ffffffff : -nan #   04:16:32.000 
[10132]  # [20] ffffffff : -nan #   lat: 36.28212  lon: 139.96133  alt: 10270.43   vH: 37.62m/s  D: 104.0  vV: 2.36m/s   2023-03-16 
[10133]  # [21] ffffffff : -nan #   04:16:33.000 
[10134]  # [22] ffffffff : -nan #   lat: 36.28204  lon: 139.96175  alt: 10272.84   vH: 38.50m/s  D: 102.3  vV: 2.36m/s   2023-03-16 
[10135]  # [23] ffffffff : -nan #   04:16:34.000 
[10136]  # [24] ffffffff : -nan #   lat: 36.28197  lon: 139.96217  alt: 10275.13   vH: 38.91m/s  D: 101.0  vV: 2.35m/s   2023-03-16 
[10137]  # [25] ffffffff : -nan #   04:16:35.000 
[10138]  # [26] ffffffff : -nan #   lat: 36.28190  lon: 139.96259  alt: 10277.55   vH: 39.09m/s  D: 101.8  vV: 2.36m/s   2023-03-16 
[10139]  # [27] ffffffff : -nan #   04:16:36.000 
[10140]  # [28] ffffffff : -nan #   lat: 36.28183  lon: 139.96302  alt: 10280.04   vH: 38.58m/s  D: 101.4  vV: 2.38m/s   2023-03-16 
[10141]  # [29] ffffffff : -nan #   04:16:37.000 
[10142]  # [30] ffffffff : -nan #   lat: 36.28175  lon: 139.96343  alt: 10282.45   vH: 37.72m/s  D: 103.3  vV: 2.39m/s  
[10143]  # [31] ffffffff : -nan #   04:16:38.000 
[10144]  # [32] ffffffff : -nan #   lat: 39.64367  lon: 139.97695  alt: 10284.86   vH: 37.94m/s  D: 104.4  vV: 15.19m/s   2023-03-16 
[10145]  # [33] ffffffff : -nan #   04:16:39.000 
[10146]  # [34] ffffffff : -nan #   lat: 36.39955  lon: 139.96424  alt: 10287.27   vH: 37.51m/s  D: 103.4  vV: 2.40m/s   2023-03-18 
[10147]  # [35] ffffffff : -nan #   04:16:40.000 
[10148]  # [36] ffffffff : -nan #   lat: 36.28151  lon: 139.96465  alt: 10289.75   vH: 37.92m/s  D: 103.6  vV: 2.40m/s   2023-03-16 
[10149]  # [37] ffffffff : -nan #   04:16:41.000 
[10150]  # [38] ffffffff : -nan #   lat: 36.28143  lon: 139.96507  alt: 10292.18   vH: 38.62m/s  D: 101.6  vV: 2.40m/s   2023-03-16 
[10151]  # [39] ffffffff : -nan #   04:16:42.000 
[10152]  # [40] ffffffff : -nan #   lat: 36.28136  lon: 139.96549  alt: 10294.50   vH: 39.24m/s  D: 101.2  vV: 2.39m/s   2023-03-16 
[10153]  # [41] ffffffff : -nan #   04:16:43.000 
[10154]  # [42] ffffffff : -nan #   lat: 36.28129  lon: 139.96592  alt: 10297.04   vH: 39.27m/s  D: 101.4  vV: 2.41m/s   2023-03-16 
[10155]  # [43] ffffffff : -nan #   04:16:44.000 
[10156]  # [44] ffffffff : -nan #   lat: 36.28122  lon: 139.96634  alt: 10299.54   vH: 38.37m/s  D: 102.1  vV: 2.42m/s   2023-03-16 
[10157]  # [45] ffffffff : -nan #   04:16:45.000 
[10158]  # [46] ffffffff : -nan #   lat: 36.28114  lon: 139.96676  alt: 10302.05   vH: 38.13m/s  D: 103.9  vV: 2.43m/s   2023-03-16 
[10159]  # [47] ffffffff : -nan #   04:16:46.000 
[10160]  # [48] ffffffff : -nan #   lat: 36.28106  lon: 139.96717  alt: 10304.43   vH: 38.21m/s  D: 104.2  vV: 2.42m/s   2023-03-16 
[10161]  # [49] ffffffff : -nan #   04:16:47.000 
[10162]  # [50] ffffffff : -nan #   lat: 36.28097  lon: 139.96758  alt: 10306.95   vH: 38.16m/s  D: 104.4  vV: 2.43m/s   2023-03-16 
[10163]  # [51] ffffffff : -nan #   04:16:48.000 
[10164]  # [52] ffffffff : -nan #   lat: 36.28089  lon: 139.96800  alt: 10309.47   vH: 39.15m/s  D: 103.1  vV: 2.44m/s   2023-03-16 
[10165]  # [53] ffffffff : -nan #   04:16:49.000 
[10166]  # [54] ffffffff : -nan #   lat: 36.28081  lon: 139.96843  alt: 10311.91   vH: 39.64m/s  D: 102.2  vV: 2.43m/s   2023-03-16 
[10167]  # [55] ffffffff : -nan #   04:16:50.000 
[... cut ...]
[10174]  # [62] 82400000 : 3.0 #   lat: 36.28051  lon: 139.97015  alt: 10323.97   vH: 38.59m/s  D: 102.7  vV: 2.53m/s   2023-03-16 
[10175]  # [63] 81000000 : 1.0 #   04:16:54.000 
```
Looks like the positions match the IMS100 positions of your log file, probably same balloon.

Since many of the radiosondes have similar preambles, there can be false positives, in particular if the header/sync word is not very distinctive. If the threshold is set low, it can happen that dft_detect will suggest a false positive with low score. The decoder would have to tell if it really was that radiosonde type.

AstroTaka, 2023-03-21:
Thank you very much for your analisis.
Yes, the frequency is 404.4MHz.
I have two questions.

(1) How can I judge weather a signal of IQ file has more than a few bit errors?

(2) How can I get the following log?

`[10112]  # [00] ffffffff : -nan #   lat: 36.28291  lon: 139.95721  alt: 10246.97   vH: 37.11m/s  D: 103.8  vV: 2.33m/s   2023-03-16 
`

I tried to use the following command. But I cannot get a part of "# [00] ffffffff : -nan #"

`cat xxx.bin |./fsk_demod --cs16 -s -b -15000 -u 15000 --stats=5 2 48000 2400 - - 2>/dev/null | ./meisei100mod --softin --ptu --ecc -v 2>/dev/null`

About "MRZ Sonde" issue, I ignore it without any worries.

rs1729, 2023-03-21:
(1) I believe 2023-03-16T041325Z is one of the better parts. If you output raw frames together with option "-v", you see the number of errors between `#......#` for each 16 bit word. "F" or "E" means there was a parity check error or it could not correct bit errors. It is some time ago, don't remember exactly, but I think the BCH code here is not so much better than Hamming. The following section is quite ok.
```
./meisei100mod23dbg -v -r --ecc --IQ 0.0 --dc - 48000 16 decode_IQ_SPY05_RS11G_2023-03-16T041325Z.bin 
IF: 48000
dec: 1
049DCE 2622 7FDF FFFF FFFF 00FC 5538 7607 30A2 0000 0000 6004 0610 #000001#  FB6230 8080 15A2 B932 5360 B3E1 000E E862 0F12 2908 00FC 14E7 0310 #000000#  
049DCE 2623 008C FFFF FFFF 0774 5572 7603 31A2 FFFF 00E6 606D 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 BCE4 #000000#  
049DCE 2624 B2A6 FFFF FFFF 00FD 5572 75F8 30A2 0000 0000 4804 0610 #000000#  FB6230 8080 15A2 B586 5360 C435 000E E969 0F5F 2939 00FD 14E7 0310 #000000#  
049DCE 2625 6077 FFFF FFFF 0774 5545 75F9 31A2 FFFF 00E6 4871 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 9B16 #000000#  
049DCE 2626 7FDE FFFF FFFF 00FD 551C 75F9 30A2 0000 0000 3004 0610 #000002#  FB6230 8080 15A2 B1B1 5360 D47D 000E EA6F 0F32 2991 00FD 14E7 0310 #000000#  
049DCE 2627 008C FFFF FFFF 0775 550B 7606 31A2 FFFF 00E6 3075 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 78BE #000000#  
049DCE 2628 B2A6 FFFF FFFF 00FE 5599 7608 30A2 0000 0000 1804 0610 #000000#  FB6230 8080 15A2 ADD3 5360 E4AC 000E EB78 0F2E 2977 00FE 13E7 0310 #000000#  
049DCE 2629 607B FFFF FFFF 0775 558F 75FE 31A2 FFFF 00E6 1879 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 54FF #100000#  
049DCE 262A 7FDF FFFF FFFF 0100 5503 75FD 30A2 0000 0000 0004 0610 #000000#  FB6230 8080 15A2 AA38 5360 F4BD 000E EC89 0EEA 28EC 0100 13E7 0310 #000000#  
049DCE 262B 008C FFFF FFFF 0776 54DD 75FE 31A2 FFFF 00E6 007D 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 31BD #000000#  
049DCE 262C B2A6 FFFF FFFF 0101 54C3 7602 30A2 0000 0000 E804 0610 #000000#  FB6230 8080 15A2 A6F0 5361 049C 000E ED96 0EAF 2897 0101 13E7 0310 #000000#  
049DCE 262D 606D FFFF FFFF 0776 54AC 7609 31A2 FFFF 00E6 E880 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0ED6 #000000#  
049DCE 262E 7FDE FFFF FFFF 0101 54D3 760C 30A2 0408 1100 D004 0610 #000022#  FB623A 8080 15A2 A3D7 5361 147E 000E EEA8 0EC4 284E 0101 13E7 0310 #000000#  
049DCE 262F 008C FFFF FFFF 0777 5502 7609 31A2 FFFF 00E6 D084 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 EC81 #000000#  
049DCE 2630 B2A6 FFFF FFFF 0102 5503 7607 30A2 0000 0000 B804 0610 #000021#  FB6334 8080 15A2 A0BB 5361 245B 000E EFB6 0EAC 288B 0102 13E7 0310 #000000#  
049DCE 2631 607C FFFF FFFF 0777 550A 7609 31A2 FFFF 00E6 B888 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 CA7A #000000#  
049DCE 2632 7FDE FFFF FFFF 0102 550E 7611 30A2 0000 0000 A004 0610 #000000#  FB6230 8080 15A2 9D76 5361 3430 000E F0C4 0EC1 28D3 0102 13E7 0310 #000000#  
049DCE 2633 008C FFFF FFFF 0778 5514 7612 31A2 FFFF 00E6 A08C 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 A879 #000000#  
049DCE 2634 B2A6 FFFF FFFF 0103 5518 7610 30A2 0000 0000 8804 0610 #000000#  FB6230 8080 15A2 99D1 5361 441D 000E F1D7 0EFC 2989 0103 13E7 0310 #000000#  
049DCE 2635 6075 FFFF FFFF 0778 5519 760D 31A2 FFFF 00E6 8890 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 86CA #000000#  
049DCE 2636 7FDF FFFF FFFF 0103 5513 7612 30A2 0000 0000 7004 0610 #000000#  FB6230 8080 15A2 95E7 5361 5429 000E F2EA 0F23 29AD 0103 13E7 0310 #000000#  
049DCE 2637 008C FFFF FFFF 0779 550F 7615 31A2 FFFF 00E6 7094 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 644E #000000#  
049DCE 2638 B2A6 FFFF FFFF 0105 54FE 7618 30A2 0000 0000 5804 0610 #000000#  FB6230 8080 15A2 91F5 5361 6449 000E F3FE 0F27 299F 0105 13E7 0310 #000000#  
049DCE 2639 6072 FFFF FFFF 0779 5526 7616 31A2 FFFF 00E6 5898 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 418C #000000#  
049DCE 263A 7FDE FFFF FFFF 0106 5517 7615 30A2 0000 0000 409C 0610 #00002E#  FB6230 8080 15A2 8E25 5361 7474 000E F51A 0F23 2949 0106 13E7 0310 #000100#  
049DCE 263B 008C FFFF FFFF 077A 54F0 760E 31A2 FFFF 00E6 409C 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 1EAE #000000#  
049DCE 263C B2A6 FFFF FFFF 0108 54F6 760C 30A2 0000 2000 2804 0610 #0000F2#  FB627E 8080 15A2 8AB1 5361 84A7 000E F633 0F0A 289C 0108 13E7 0310 #000000#  
049DCE 263D 606E FFFF FFFF 077B 5509 7612 31A2 FFFF 00E6 28A0 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 FBC6 #000000#  
049DCE 263E 7FDF 0000 4082 0107 54F4 7614 30A2 0000 0000 1004 0610 #000000#  FB6230 8080 15A2 8788 5361 94DD 000E F740 0EFF 285A 0107 13E7 0310 #012000#  
049DCE 263F 008C 0000 0081 077B 54B6 7615 31A2 FFFF 00E6 10A4 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 D996 #E00000#  
049DCE 2640 B2A6 FFFF FFFF 0107 5496 7615 30A2 0000 0000 F804 0610 #000000#  FB6230 8080 15A2 8463 5361 A519 000E F850 0F0C 2872 0107 13E7 0310 #000000#  
049DCE 2641 6076 FFFF FFFF 077C 54D8 7615 31A2 FFFF 00E6 F8A7 040D #000000#  FB6230 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 B7E5 #100000#  
[... cut ...]
                                  | ( (w16[0]&0xFF00)>>8 | (w16[0]&0xFF)<<8 );
                             fw32 = f32e2(w32);
 
+                            if (option_dbg) {
+                                printf(" # [%02d] %08x : %.1f # ", counter % 64, w32, fw32);
+                            }
+
                             if (err_blks == 0) // err_frm zu schwach
                             {
                                 gpx.cfg[counter%64] = fw32;
```
I can include it in the next version if you are interested.

rs1729, 2023-03-22:
About MRZ,
in your case, 404.4 MHz, it appears to be a false positive, since it's the frequency of the RS11-G.

In principle, I don't know if MRZ are launched at east coast Russian stations, from northern Japan it might be possible to receive such signals. Strong winds could bring radiosondes from Vladivostok in receiving range of Tokyo area, I guess.

AstroTaka, 2023-03-25:
@rs1729 Thank you very much!
I downloaded it from your repository and build it and tried it.
I hope you to add the option in the next version.

I received RS11G three days ago.
https://sondehub.org/7110235
I checked IQ log file with "--dbg".
./meisei100mod exported about 7000 lines log.
And there is no fffffff.

Thank you very much.

AstroTaka, 2023-03-25:
@rs1729 About MRZ, thank you very much for your information about a sonde which launched at Vladivostok.

closed 2023-03-25 by AstroTaka

rs1729, 2023-03-25:
I don't really know if and what kind of radiosondes Vladivostok is currently launching.
http://weather.uwyo.edu/upperair/sounding.html
shows activity for station 31977, but shows only data up to 100hPa/16km altitude.
Raw data shows radiosonde type 62, should be MRZ-3MK. Not sure if this is the 1680MHz version.
The MRZ-N1 on 400MHz has type 19 I believe.
L band 1680 MHz would be more difficult to receiver at that distance I guess.

The "--dbg" would be available next time @darksidelemm takes it over to auto_rx, but I don't know if it can be used in auto_rx, i.e. if there is an option in auto_rx to log that output, since it is not in JSON. I believe there is an raw log option? In the raw output you can see the config data as well, but not the corresponding float values though. If you record the signal and have suspicious output, maybe you can go back to the recording and look at it manually with the decoder and "--dbg".

AstroTaka, 2023-03-28:
@rs1729 All right about russian sonde.

And about "--dbg" option, I have no issue even if auto_rx does not support it because I will use it for recorded  IQ file manually.