projecthorus/radiosonde_auto_rx #766: Scan plot on the last version 1.6.1

TREBOR81400, 2023-04-01:
the reception signal does not appear when you activate the Scan Plot

darksidelemm, 2023-04-01:
Explain this a bit more please. Does nothing appear at all, even with the page left open for a while? 
If your station is already decoding a radiosonde, then there may be no scan plot data to show.

TREBOR81400, 2023-04-01:
[image]

TREBOR81400, 2023-04-01:
[image]

darksidelemm, 2023-04-01:
A few things to check:
- Add /get_scan_data to the end of your URL and see if you get anything (If you do, there will be a *lot* of data, so don't paste it here please)
- Open Developer Tools and look in the console for any errors.

ly5p, 2023-04-02:
the problem is with chrome browser
[image]
[image]

darksidelemm, 2023-04-02:
Is it? I've got scan plots working fine on Chrome on both OSX and Windows.

TREBOR81400, 2023-04-02:
Hello, thank you for your help and advice, the situation is back to normal.

closed 2023-04-02 by TREBOR81400