projecthorus/radiosonde_auto_rx #767: parallel reception of many radiosondes

gmke, 2023-04-20:
at noon, I could easily record 7-8 radiosondes in my area (Stuttgart, Germany). I tried to connect 7 RTLSDRs (three separate devices and a 4 channel Kerberos) to the reception pc. This kinda worked for some time but proved to be unstable, loosing devices, frequent usb resets, etc. (with a powered hub) I achieved stable operation with <= 4 devices (Intel Celeron J4105). But using so much hardware seems to be unnecessary. We could cover the used part of the spectrum with just two RTLSDRs or one Airspy mini, if we would quickly switch the frequencies or grab the whole spectrum and analyze for all detected peaks. Any work or thoughts done along this way of decoding?

Markus

darksidelemm, 2023-04-21:
Yes, that many RTLSDRs is unlikely to work reliably.

There is already support to receive many sondes in auto_rx v1.6 using either a Airspy Mini or R2, see here: https://github.com/projecthorus/radiosonde_auto_rx/wiki/Network-SDR-Decoding-Instructions

gmke, 2023-04-21:
got it working!!!! absolutely brilliant! Many, many thanks.

closed 2023-04-21 by gmke

darksidelemm, 2023-04-23:
Good to hear this is working - I'm interested to know how many radiosonde you end up receiving simultaneously?

gmke, 2023-04-24:
it worked for three radiosondes automatically. In the spectrum I saw the peak of a fourth sonde and it did not pick it up. I had to manually add the frequency to the always list. After the restart, all four sondes were found. Theoretically it should be able to detect up to 8 sondes. Will check frequently. Btw, I'm GALIL2-1 and I receive with an Airspy Mini and a Maas D-30 antenna with LNA4ALL.

gmke, 2023-05-04:
can confirm the simultaneous reception of 5 sondes now. In theory, it could be 7 sondes, but unfortunately they transmit on the same frequency. Germany needs a better frequency plan....

gmke, 2023-05-05:
ok, this moment the count is up to 6 simultaneous sondes now. !!! Great work!

gmke, 2023-05-09:
thanks to our armed forces, starting many sondes today, I can confirm the simultaneous reception of 7 sondes and one scanning. Maybe I have to increase to 10 sondes...

Current Task: SDR #SPY01: Scanning SDR #SPY02: Decoding (402.501 MHz) SDR #SPY03: Decoding (404.500 MHz) SDR #SPY04: Decoding (402.900 MHz) SDR #SPY05: Decoding (402.701 MHz) SDR #SPY06: Decoding (404.350 MHz) SDR #SPY07: Decoding (402.300 MHz) SDR #SPY08: Decoding (405.301 MHz)