projecthorus/radiosonde_auto_rx #769: Radiosonde_auto_rx 1.6.1 update

Mopar44084, 2023-05-07:
There seems to be missing columns in the history after the 1.6.1 update. It seems to be missing distance, and packets received etc.
[image]

darksidelemm, 2023-05-07:
Click the configuration button at the top right of the historical view page. you should be able to select what columns you want from there.

Mopar44084, 2023-05-07:
Thanks, that fixed it! How do you log in into controls?

darksidelemm, 2023-05-08:
Not sure what you mean by 'log in' in this case?

Mopar44084, 2023-05-08:
its under settings and then Controls, it asks for a password

darksidelemm, 2023-05-08:
Ahh, this is the 'web control' stuff. Refer here: https://github.com/projecthorus/radiosonde_auto_rx/wiki/Web-Interface-Guide#advanced-controls

Going to close this issue now

closed 2023-05-08 by darksidelemm